chipweinberger/flutter_blue_plus #751: [Bug]: Why is the subcontracting sending device receiving incomplete

GoodbyeV, 2024-01-04:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.4

### Flutter Version

3.13.5

### What OS?

Android

### OS Version

Android 10.0

### Bluetooth Module

esp32

### What is your problem?

I want to send a large set of data, but the mtu on the device side can only be set to 23, so I subcontract to send, and the data size of no packet is 20, as follows:  
void sendDataToDeviceSubPackage(Uint8List data) {
    int offset = 0;
    int packetCount = 0;
    void sendNextChunk() async {
      if (offset < data.length) {
        // 数据发送完成
      var dataLength = _currentBleDevice!.mtuNow - 3;
      print("mtuNow : ${_currentBleDevice!.mtuNow}");
      int chunkSize = (offset + 20 <= data.length)
          ? 20
          : data.length - offset;
      Uint8List chunk = data.sublist(offset, offset + chunkSize);
      await _writeCharacteristic?.write(chunk,withoutResponse: false,timeout: 15);
      packetCount++;
      print(TAG +
            "发送数据成功: , 数据长度: ${chunk.length},包序号:${packetCount}，末尾字节：${chunk.last}");
      offset += chunkSize;
      print(chunk);
      Future.delayed(Duration(milliseconds: DELAY_SEND_TIME), sendNextChunk);
      }else{
        print("数据发完");
      }
    }
    Future.delayed(Duration(milliseconds: DELAY_SEND_TIME), sendNextChunk);
  }
 I also set a delay here, but the device has not been able to get the full data，Does anyone know why？

### Logs

```shell
I/flutter (24739): [FBP] <writeCharacteristic> args: {remote_id: 50:54:7B:7D:71:F3, service_uuid: fff0, secondary_service_uuid: null, characteristic_uuid: fff2, write_type: 1, allow_long_write: 0, value: aaaa0019030102030405060708090a0b0c0d0e0f}
D/[FBP-Android](24739): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](24739): [FBP] onCharacteristicWrite:
D/[FBP-Android](24739): [FBP]   chr: fff2
D/[FBP-Android](24739): [FBP]   status: GATT_SUCCESS (0)
I/flutter (24739): [FBP] <writeCharacteristic> result: true
I/flutter (24739): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 50:54:7B:7D:71:F3, error_code: 0, characteristic_uuid: fff2, value: aaaa0019030102030405060708090a0b0c0d0e0f}
I/flutter (24739): BleManagerPluse test: 发送数据成功: , 数据长度: 20,包序号:1，末尾字节：15
I/flutter (24739): [170, 170, 0, 25, 3, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15]
I/flutter (24739): mtuNow-- : 23
I/flutter (24739): mtuNow : 23
I/flutter (24739): [FBP] <writeCharacteristic> args: {remote_id: 50:54:7B:7D:71:F3, service_uuid: fff0, secondary_service_uuid: null, characteristic_uuid: fff2, write_type: 1, allow_long_write: 0, value: 1011121314151617181902b555}
D/[FBP-Android](24739): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](24739): [FBP] onCharacteristicWrite:
I/flutter (24739): [FBP] <writeCharacteristic> result: true
D/[FBP-Android](24739): [FBP]   chr: fff2
D/[FBP-Android](24739): [FBP]   status: GATT_SUCCESS (0)
I/flutter (24739): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 50:54:7B:7D:71:F3, error_code: 0, characteristic_uuid: fff2, value: 1011121314151617181902b555}
I/flutter (24739): BleManagerPluse test: 发送数据成功: , 数据长度: 13,包序号:2，末尾字节：85
I/flutter (24739): [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 2, 181, 85]
I/flutter (24739): mtuNow-- : 23
I/flutter (24739): 数据发完
```

chipweinberger, 2024-01-04:
why not use `splitWrite`?

```
import 'dart:math';
// split write should be used with caution.
//    1. due to splitting, `characteristic.read()` will return partial data.
//    2. it can only be used *with* response to avoid data loss
//    3. The characteristic must be designed to support split data
extension splitWrite on BluetoothCharacteristic {
  Future<void> splitWrite(List<int> value, {int timeout = 15}) async {
    int chunk = device.mtuNow - 3; // 3 bytes ble overhead
    for (int i = 0; i < value.length; i += chunk) {
      List<int> subvalue = value.sublist(i, min(i + chunk, value.length));
      await write(subvalue, withoutResponse:false, timeout: timeout);
    }
  }
}
```

It is not a FBP problem. Probably device problem.

GoodbyeV, 2024-01-04:
> splitWrite
如何在业务代码中使用这个方法呢？

GoodbyeV, 2024-01-04:
> why not use `splitWrite`? i.e. `chr.splitWrite(data);`
> 
> ```
> import 'dart:math';
> // split write should be used with caution.
> //    1. due to splitting, `characteristic.read()` will return partial data.
> //    2. it can only be used *with* response to avoid data loss
> //    3. The characteristic must be designed to support split data
> extension splitWrite on BluetoothCharacteristic {
>   Future<void> splitWrite(List<int> value, {int timeout = 15}) async {
>     int chunk = device.mtuNow - 3; // 3 bytes ble overhead
>     for (int i = 0; i < value.length; i += chunk) {
>       List<int> subvalue = value.sublist(i, min(i + chunk, value.length));
>       await write(subvalue, withoutResponse:false, timeout: timeout);
>     }
>   }
> }
> ```
> 
> It is not a FBP problem. Probably device problem.

> why not use `splitWrite`? i.e. `chr.splitWrite(data);`
> 
> ```
> import 'dart:math';
> // split write should be used with caution.
> //    1. due to splitting, `characteristic.read()` will return partial data.
> //    2. it can only be used *with* response to avoid data loss
> //    3. The characteristic must be designed to support split data
> extension splitWrite on BluetoothCharacteristic {
>   Future<void> splitWrite(List<int> value, {int timeout = 15}) async {
>     int chunk = device.mtuNow - 3; // 3 bytes ble overhead
>     for (int i = 0; i < value.length; i += chunk) {
>       List<int> subvalue = value.sublist(i, min(i + chunk, value.length));
>       await write(subvalue, withoutResponse:false, timeout: timeout);
>     }
>   }
> }
> ```
> 
> It is not a FBP problem. Probably device problem.

OK，i will try

GoodbyeV, 2024-01-05:
> why not use `splitWrite`? i.e. `chr.splitWrite(data);`
>
> ```
> import 'dart:math';
> // split write should be used with caution.
> //    1. due to splitting, `characteristic.read()` will return partial [...]

> why not use `splitWrite`? i.e. `chr.splitWrite(data);`
>
> ```
> import 'dart:math';
> // split write should be used with caution.
> //    1. due to splitting, `characteristic.read()` will return partial [...]

> why not use `splitWrite`? i.e. `chr.splitWrite(data);`
>
> ```
> import 'dart:math';
> // split write should be used with caution.
> //    1. due to splitting, `characteristic.read()` will return partial [...]

Thank you. It's settled

closed 2024-01-05 by chipweinberger