chipweinberger/flutter_blue_plus #752: Connectable Not working properly

SahilShirsat17, 2024-01-05:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.15.9

### Flutter Version

 3.10.4

### What OS?

Android

### OS Version

android 13

### Bluetooth Module

Xioami 11 i   - bluetooth 5.2

### What is your problem?

I am using the library for scan and connect as I am displaying the if the advertisement data device is connectable or not if connectable then show the button but it is showing connectable for every device available  

Here below is my code snippet 

trailing: result.advertisementData.connectable ? StreamBuilder<BluetoothConnectionState>(
                stream: result.device.connectionState,
                builder: (context, snapshot) {
                  if (snapshot.hasData) {
                    if (snapshot.data == BluetoothConnectionState.disconnected) {
                      return MaterialButton(
                          minWidth: 0,
                          height: 25,
                          color:Theme.of(context).primaryColor,
                          child: Text("Connect" ,style: GoogleFonts.workSans(fontSize: 12, color: Colors.white)),
                          onPressed:() async {
                            try {
                              await result.device.connect();
                            }catch (e){
                              ScaffoldMessenger.of(context).showSnackBar(SnackBar(
                                content: Text('Unable to connect to device: ${result.device.remoteId.toString()}'),
                              ));
                            }
                          });

can you please go through this

### Logs

```shell
NO logs
```

chipweinberger, 2024-01-05:
this bug is already fixed I think. update to latest version

chipweinberger, 2024-01-05:
update to 1.17.1 or later

closed 2024-01-05 by chipweinberger