chipweinberger/flutter_blue_plus #754: [Help]: Scanning a lot for RSSI leads to progressively slower comms on iOS

LeoBound, 2024-01-09:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.7

### Flutter Version

3.16.0

### What OS?

iOS

### OS Version

15.7.9

### Bluetooth Module

RN4870

### What is your problem?

I fetch the connected device's RSSI every second to update a signal strength indicator in my application.
It appears that on iOS this slows down the sending/receiving of data and it's not really clear why.
I am sending with BluetoothCharacteristic.write and receiving by listening to onValueReceived.
This still works well on Android, and used to work on iOS, but I recently updated to 1.28.7 and in this it didn't. (I'm not sure when it broke, I don't test too frequently on iOS).
It appears to still be broken as of 1.31.7.

I don't think the logs below will be that helpful, but the second time reading took a lot longer.

### Logs

```shell
[FBP-iOS] handleMethodCall: flutterHotRestart [FBP-iOS] initializing CBCentralManager [FBP-iOS] initializing checkForMtuChangesTimer [FBP-iOS] disconnectAllDevices(flutterHotRestart)
[FBP-iOS] connectedPeripherals: 0
[FBP-iOS] handleMethodCall: setLogLevel
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn [FBP-iOS] handleMethodCall: startScan [FBP-iOS] handleMethodCall: connect [FBP-iOS] didConnectPeripheral [FBP-iOS] handleMethodCall: discoverServices [FBP-iOS] didDiscoverServices:
[FBP-iOS]   svc: 180a
[FBP-iOS]   svc: 49535343-fe7d-4ae5-8fa9-9fafd205e455
[FBP-iOS] didDiscoverCharacteristicsForService:
[FBP-iOS]   svc: 180a
[FBP-iOS]     chr: 2a29
[FBP-iOS]     chr: 2a24
[FBP-iOS]     chr: 2a25
[FBP-iOS]     chr: 2a27
[FBP-iOS]     chr: 2a26
[FBP-iOS]     chr: 2a28
[FBP-iOS]     chr: 2a23
[FBP-iOS]     chr: 2a2a
[FBP-iOS] didDiscoverCharacteristicsForService:
[FBP-iOS]   svc: 49535343-fe7d-4ae5-8fa9-9fafd205e455
[FBP-iOS]     chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS]     chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS]     chr: 49535343-4c8a-39b3-2f49-511cff073b7e
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 2a29
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 2a24
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 2a25
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 2a27
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 2a26
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 2a28
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 2a23
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 2a2a
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS]     desc: 2902
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: 49535343-4c8a-39b3-2f49-511cff073b7e
[FBP-iOS]     desc: 2902
[FBP-iOS] handleMethodCall: setNotifyValue [FBP-iOS] didUpdateNotificationStateForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] handleMethodCall: stopScan
[FBP-iOS] handleMethodCall: setNotifyValue [FBP-iOS] didUpdateNotificationStateForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -45
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
Can't find keyplane that supports type 4 for keyboard iPhone-Portrait-NumberPad; using 27043_Portrait_iPhone-Simple-Pad_Default
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -44
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -40
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -40
[FBP-iOS] handleMethodCall: writeCharacteristic [FBP-iOS] didWriteValueForCharacteristic:
[FBP-iOS]   chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] handleMethodCall: stopScan
[FBP-iOS] handleMethodCall: stopScan
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -41
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -40
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: writeCharacteristic [FBP-iOS] didWriteValueForCharacteristic:
[FBP-iOS]   chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: writeCharacteristic [FBP-iOS] didWriteValueForCharacteristic:
[FBP-iOS]   chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] handleMethodCall: writeCharacteristic [FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didWriteValueForCharacteristic:
[FBP-iOS]   chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -44
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -44
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -44
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -44
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -44
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -44
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -44
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -43
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -45
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didReadRSSI: -40
[FBP-iOS] handleMethodCall: writeCharacteristic [FBP-iOS] didWriteValueForCharacteristic:
[FBP-iOS]   chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didReadRSSI: -41
[FBP-iOS] handleMethodCall: writeCharacteristic [FBP-iOS] didWriteValueForCharacteristic:
[FBP-iOS]   chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didReadRSSI: -41
[FBP-iOS] handleMethodCall: writeCharacteristic [FBP-iOS] didWriteValueForCharacteristic:
[FBP-iOS]   chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS] handleMethodCall: writeCharacteristic [FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didWriteValueForCharacteristic:
[FBP-iOS]   chr: 49535343-8841-43f4-a8d4-ecbe34729bb3
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] handleMethodCall: readRssi
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]   chr: 49535343-1e4d-4bd9-ba61-23c647249616
[FBP-iOS] didReadRSSI: -44
```

LeoBound, 2024-01-09:
It looks like disconnecting and reconnecting resets the 'delay', but after a few tens of calls to get the RSSI, it is slow again

chipweinberger, 2024-01-09:
In FBP, all comms are one at a time. This is for best reliability on android, but we use the same behavior for iOS too.

This means when rssi is requested, writes are paused until the rssi request is done. 

You could try to implement a new feature for FBP, a `device.allowSimultaneousComms` function that affects the mutex keys (see `_Mutex` in the code).

> progressively slower

I'm not sure why it would get `progressively` slower though

chipweinberger, 2024-01-09:
also, are you using `await`?

fbp uses a queue. if you call FBP without `await`, the request will be added to the end of the queue.

it is recommended to use `await`, or if you do not use `await` your should have a mechanism to check if the last request has completed before you schedule more work

chipweinberger, 2024-01-14:
same issue: https://github.com/boskokg/flutter_blue_plus/issues/505

LeoBound, 2024-01-15:
Yes I'm using await to get the rssi,
I'll have a look into it today and see if I make any progress.

chipweinberger, 2024-01-19:
ideally, you would tell me the fbp version that this first "broke".

the most likely culprit are the mutex changes. look at the changelog.

LeoBound, 2024-01-22:
Apologies for the delay,
I can confirm it seems to work as expected in 1.20.8 and prior, and as of 1.21.0 I start getting this issue.

chipweinberger, 2024-01-22:
the changes made in 1.21.0 are for reliability, so I wont be reverting it.

But you can fork FBP and change the mutex key from "global" to "RSSI", and it should fix the issue for you.

LeoBound, 2024-01-22:
Changing the mutex key does appear to fix the issue, thanks for your help!

closed 2024-01-22 by LeoBound