chipweinberger/flutter_blue_plus #755: [Help]: Does re-connect work in flutter blue plus ?

ringonotts, 2024-01-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.37.7

### Flutter Version

3.10.1

### What OS?

Android

### OS Version

13

### Bluetooth Module

STM32

### What is your problem?

In flutter_blue i implemented the connectionState.listen stream and when i walked out of range of my bluetooth device i would get a indication that the BLE connection state was disconnected. This still works in flutter blue plus obviously.
When i walked back in range of my device I would get a indciation on the stream that I am "connected". This was pretty handy.
However, in flutter blue plus this has stopped working. The stream is never called when i get back in range.
Is this supposed to happen in flutter blue plus or is there another way to do it ? Or has Android changed ?

Thanks, Ringo.

### Logs

```shell
W/ProviderInstaller(24097): Failed to load providerinstaller module: No acceptable module com.google.android.gms.providerinstaller.dynamite found. Local version is 0 and remote version is 0.
D/nativeloader(24097): Configuring clns-5 for other apk /system/framework/org.apache.http.legacy.jar. target_sdk_version=34, uses_libraries=ALL, library_path=/data/app/~~pgc8IechCwUv3TauLhI0uA==/com.google.android.gms-jsGGmNNrEbyESPwiJAHHXg==/lib/arm64:/data/app/~~pgc8IechCwUv3TauLhI0uA==/com.google.android.gms-jsGGmNNrEbyESPwiJAHHXg==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/com.google.android.gms
D/nativeloader(24097): Extending system_exposed_libraries: libedgetpu_dba.google.so
D/nativeloader(24097): Configuring clns-6 for other apk /system/framework/com.android.media.remotedisplay.jar. target_sdk_version=34, uses_libraries=ALL, library_path=/data/app/~~pgc8IechCwUv3TauLhI0uA==/com.google.android.gms-jsGGmNNrEbyESPwiJAHHXg==/lib/arm64:/data/app/~~pgc8IechCwUv3TauLhI0uA==/com.google.android.gms-jsGGmNNrEbyESPwiJAHHXg==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/com.google.android.gms
D/nativeloader(24097): Extending system_exposed_libraries: libedgetpu_dba.google.so
W/orthings.batmon(24097): Loading /data/misc/apexdata/com.android.art/dalvik-cache/arm64/system@framework@com.android.location.provider.jar@classes.odex non-executable as it requires an image which we failed to load
D/nativeloader(24097): Configuring clns-7 for other apk /system/framework/com.android.location.provider.jar. target_sdk_version=34, uses_libraries=ALL, library_path=/data/app/~~pgc8IechCwUv3TauLhI0uA==/com.google.android.gms-jsGGmNNrEbyESPwiJAHHXg==/lib/arm64:/data/app/~~pgc8IechCwUv3TauLhI0uA==/com.google.android.gms-jsGGmNNrEbyESPwiJAHHXg==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/com.google.android.gms
D/nativeloader(24097): Extending system_exposed_libraries: libedgetpu_dba.google.so
D/nativeloader(24097): Configuring clns-8 for other apk /data/app/~~pgc8IechCwUv3TauLhI0uA==/com.google.android.gms-jsGGmNNrEbyESPwiJAHHXg==/base.apk. target_sdk_version=34, uses_libraries=, library_path=/data/app/~~pgc8IechCwUv3TauLhI0uA==/com.google.android.gms-jsGGmNNrEbyESPwiJAHHXg==/lib/arm64:/data/app/~~pgc8IechCwUv3TauLhI0uA==/com.google.android.gms-jsGGmNNrEbyESPwiJAHHXg==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/com.google.android.gms
V/NativeCrypto(24097): Registering com/google/android/gms/org/conscrypt/NativeCrypto's 302 native methods...
W/orthings.batmon(24097): Accessing hidden method Ljava/security/spec/ECParameterSpec;->getCurveName()Ljava/lang/String; (unsupported, reflection, allowed)
I/ProviderInstaller(24097): Installed default security provider GmsCore_OpenSSL
D/BluetoothGatt(24097): onConnectionUpdated() - Device=00:80:E1:27:AF:81 interval=6 latency=0 timeout=500 status=0
D/TrafficStats(24097): tagSocket(166) with statsTag=0xffffffff, statsUid=-1
W/orthings.batmon(24097): Accessing hidden field Ljava/net/Socket;->impl:Ljava/net/SocketImpl; (unsupported, reflection, allowed)
W/orthings.batmon(24097): Accessing hidden method Ljava/security/spec/ECParameterSpec;->setCurveName(Ljava/lang/String;)V (unsupported, reflection, allowed)
D/BluetoothGatt(24097): onSearchComplete() = Device=00:80:E1:27:AF:81 Status=0
D/[FBP-Android](24097): [FBP] onServicesDiscovered:
D/[FBP-Android](24097): [FBP]   count: 5
D/[FBP-Android](24097): [FBP]   status: 0GATT_SUCCESS
D/BluetoothGatt(24097): onPhyUpdate() - status=0 address=00:80:E1:27:AF:81 txPhy=2 rxPhy=2
D/BluetoothGatt(24097): onConnectionUpdated() - Device=00:80:E1:27:AF:81 interval=24 latency=0 timeout=500 status=0
I/flutter (24097): [FBP] [[ OnDiscoveredServices ]] result: {error_string: GATT_SUCCESS, success: 1, remote_id: 00:80:E1:27:AF:81, error_code: 0, services: [{included_services: [], characteristics: [{descriptors: [{service_uuid: 1801, remote_id: 00:80:E1:27:AF:81, descriptor_uuid: 2902, characteristic_uuid: 2a05}], service_uuid: 1801, remote_id: 00:80:E1:27:AF:81, characteristic_uuid: 2a05, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 0, authenticated_signed_writes: 0, extended_properties: 0, indicate: 1, indicate_encryption_required: 0, write: 0, notify: 0}}], service_uuid: 1801, is_primary: 1, remote_id: 00:80:E1:27:AF:81}, {included_services: [], characteristics: [{descriptors: [], service_uuid: 1800, remote_id: 00:80:E1:27:AF:81, characteristic_uuid: 2a00, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 1, n
I/flutter (24097): [FBP] <requestMtu> args: {remote_id: 00:80:E1:27:AF:81, mtu: 512}
```

chipweinberger, 2024-01-12:
yes it's supposed to work.

make sure you set `autoconnect ` to true when calling `connect`

ringonotts, 2024-01-12:
@chipweinberger Thanks - yes it does work! Thanks loads - you're a legend.
Ringo

closed 2024-01-12 by ringonotts