chipweinberger/flutter_blue_plus #756: [Help]: Scan function migration guide, possible regresion for iOS from 1.30.3 to 1.31.8

4WayMitja, 2024-01-15:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

flutter_blue_plus: 1.31.8

### Flutter Version

Flutter 3.16.3 • channel stable • https://github.com/flutter/flutter.git Framework • revision b0366e0a3f (6 weeks ago) • 2023-12-05 19:46:39 -0800 Engine • revision 54a7145303 Tools • Dart 3.2.3 • DevTools 2.28.4

### What OS?

iOS, All

### OS Version

16.7.2

### Bluetooth Module

Kinetis® KW41Z

### What is your problem?

Please check your migration guide for `scan` function.

In section subscription you have:
```
var subscription = FlutterBluePlus.scanResults.listen(
      (r) => if(r.isNotEmpty) {controller.add(r.first);},
      onError: (e, stackTrace) => controller.addError(e, stackTrace),
    );
```
Which should probably be:
```
var subscription = FlutterBluePlus.scanResults.listen(
      (r) {if(r.isNotEmpty) controller.add(r.first);},
      onError: (e, stackTrace) => controller.addError(e, stackTrace),
    );
```

In any case I can confirm you that above extension somehow works for iOS in version "flutter_blue_plus: 1.30.3" but it is broken in "flutter_blue_plus: 1.31.8" (regression maybe?). Please re-check migration guide for scan function.

### Logs

```shell
No logs.
```

4WayMitja, 2024-01-15:
Relevant link:
https://github.com/boskokg/flutter_blue_plus/blob/2cba28a242823f0123ae8914d3866263d6f1526f/MIGRATION.md#flutterblueplusscan-was-removed

chipweinberger, 2024-01-15:
can you open a PR? thanks

chipweinberger, 2024-01-15:
fixed the guide

closed 2024-01-15 by chipweinberger