chipweinberger/flutter_blue_plus #757: How can we calculate the advertisement interval to display in the scan result list

SahilShirsat17, 2024-01-17:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.8

### Flutter Version

3.16.5

### What OS?

Android

### OS Version

android 13

### Bluetooth Module

Xioami 11 i   - bluetooth 5.2

### What is your problem?

How can we display advertisement interval  in scan result  

### Logs

```shell
No logs
```

chipweinberger, 2024-01-17:
you cant. there are all the possible keys

https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/advertisement_data_retrieval_keys

SahilShirsat17, 2024-01-17:
Can't we use the package and display the advertising interval in the scan result rather going for native android

chipweinberger, 2024-01-17:
not currently possible. you will need to open a PR.

closed 2024-01-18 by chipweinberger