chipweinberger/flutter_blue_plus #760: [Bug]: Extension scan is not always working on iOS from version 1.30.3

4WayMitja, 2024-01-18:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

flutter_blue_plus: 1.31.8

### Flutter Version

Flutter 3.16.3 • channel stable • https://github.com/flutter/flutter.git Framework • revision b0366e0a3f (6 weeks ago) • 2023-12-05 19:46:39 -0800 Engine • revision 54a7145303 Tools • Dart 3.2.3 • DevTools 2.28.4

### What OS?

iOS

### OS Version

16.7.2

### Bluetooth Module

Kinetis® KW41Z

### What is your problem?

Extension scan is not working with newer versions of flutter_blue_plus.

Here is a relevant list of flutter_blue_plus versions:
-> 1.31.8 not working
-> 1.31.4 sometimes works
-> 1.31.0; 1.30.9; 1.30.8; 1.30.7 breaks our app
-> 1.30.6; 1.30.5 works
-> 1.30.4 does not build
-> 1.30.3 works in production (with 1000+ active users)

We can point to a commit that caused the "breaks our app" issue:

In pubspec.yaml file write and you will get the first breaking library example:
```
flutter_blue_plus:
    git:
      url: https://github.com/boskokg/flutter_blue_plus.git
      ref: 22d7e4a84323823a41d796268fc86fdbe94b38fd
```
      
Repository is commit:
https://github.com/boskokg/flutter_blue_plus/commit/22d7e4a84323823a41d796268fc86fdbe94b38fd

In short:
Extension scan is finding devices in 1.30.3, never finds device in 1.31.8 and sometimes find device in 1.31.4.

Please do not just close the issue and tell me if you need additional information or debug from our site.
  

### Logs

```shell
No logs.
```

4WayMitja, 2024-01-18:
Related issue:
https://github.com/boskokg/flutter_blue_plus/issues/756

chipweinberger, 2024-01-18:
Can you post logs from 1.30.8?

this might be due to us no longer using `pushToStream:false` https://github.com/boskokg/flutter_blue_plus/commit/d1a9292bb1b3fd44e7df31334b4ba26bcf6cc6d0

4WayMitja, 2024-01-22:
Here are the logs form 1.31.8 (not finding device, PROBLEM) and 1.30.3 (finding device, WORKING) with cleaned up code sample - which indicates our code usage.

If you would need we could implement you a full example app with reproducible issue.

[flutter_ble_1_30_3.txt](https://github.com/boskokg/flutter_blue_plus/files/14009440/flutter_ble_1_30_3.txt)
[scan_mimimal_example_code.txt](https://github.com/boskokg/flutter_blue_plus/files/14009441/scan_mimimal_example_code.txt)
[flutter_ble_1_31_8.txt](https://github.com/boskokg/flutter_blue_plus/files/14009442/flutter_ble_1_31_8.txt)

chipweinberger, 2024-01-22:
thanks. 

You should put print statements everywhere in your code and debug what code is getting called in your extension,  etc.

4WayMitja, 2024-01-24:
When using extension scan from (scan_code_example_20240124):
https://github.com/boskokg/flutter_blue_plus/blob/1.31.8/MIGRATION.md#flutterblueplusscan-was-removed
With usage:
```
// Example code usage
    FlutterBluePlus.setLogLevel(LogLevel.verbose, color: false);
    final scanDuration = const Duration(milliseconds: 2500);
    final waitDuration = const Duration(milliseconds: 4500);

    debugPrint('BLE start scan...');
    Scan.scan(timeout: scanDuration).listen((scanResult) {
      debugPrint('BLE scanResult: $scanResult');
      debugPrint('BLE scanResult advertisementData: ${scanResult.advertisementData}');
      debugPrint('BLE scanResult advertisementData.advName: ${scanResult.advertisementData.advName}');
    });
    debugPrint('BLE scanning...');
    await Future<void>.delayed(waitDuration);
    debugPrint('BLE scanning DONE');
```

for version 1.30.3 we get results which can be seen in the logs:
```
flutter: BLE scanResult: ScanResult{device: BluetoothDevice{remoteId: D1DABE3C-7D73-8BF4-00BF-DE79A1BE477A, platformName: timmy, services: null}, advertisementData: AdvertisementData{advName: , txPowerLevel: 12, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: []}, rssi: -41, timeStamp: 2024-01-24 10:52:44.002288}
flutter: BLE scanResult advertisementData: AdvertisementData{advName: , txPowerLevel: 12, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: []}
flutter: BLE scanResult advertisementData.advName:
```
but for version 1.31.8 the code inside
```
(scanResult) {
      debugPrint('BLE scanResult: $scanResult');
      ...
    }
```
is never executed and so no logs produced.
It can be seen that devices are still visible like in previous version but the returned `controller.stream` is empty.

Same code is returning data with compiled library version 1.30.3 but does not return data when compiled with library version 1.31.8.

[flutter_ble_1_30_3_20240124.txt](https://github.com/boskokg/flutter_blue_plus/files/14037304/flutter_ble_1_30_3_20240124.txt)
[scan_code_example_20240124.txt](https://github.com/boskokg/flutter_blue_plus/files/14037305/scan_code_example_20240124.txt)
[flutter_ble_1_31_8_20230124.txt](https://github.com/boskokg/flutter_blue_plus/files/14037306/flutter_ble_1_31_8_20230124.txt)

chipweinberger, 2024-01-24:
please put logs in the extension code

chipweinberger, 2024-01-24:
I've fixed the extension.

needed to add `skip(1)`

closed 2024-01-24 by chipweinberger

4WayMitja, 2024-01-25:
@chipweinberger it is working now. Tested for "flutter_blue_plus: 1.30.3", "flutter_blue_plus: 1.31.8" and "flutter_blue_plus: 1.31.10". Thanks.

We will now use your updated extension.

Do you think that in the future we should migrate to startScan usage without extension? (or will support for scan remain and we do not need to migrate)

chipweinberger, 2024-01-25:
no need to migrate