chipweinberger/flutter_blue_plus #761: [Feature]: Mark library with "Please be fully prepared to deal with breaking changes. This package must be tested on a real device."

4WayMitja, 2024-01-18:
### FlutterBluePlus Version

flutter_blue_plus: 1.31.8

### Flutter Version

Flutter 3.16.3 • channel stable • https://github.com/flutter/flutter.git Framework • revision b0366e0a3f (6 weeks ago) • 2023-12-05 19:46:39 -0800 Engine • revision 54a7145303 Tools • Dart 3.2.3 • DevTools 2.28.4

### What OS?

All

### OS Version

16.7.2

### Bluetooth Module

Kinetis® KW41Z

### What is your feature request?

In original flutter_blue library we had "Please be fully prepared to deal with breaking changes. This package must be tested on a real device." in the Readme.md section:
https://github.com/boskokg/flutter_blue#alpha-version

This is still true for this active library as we have more than 100+ regression in the past year.

Please consider adding this text or changing install instructions on
https://pub.dev/packages/flutter_blue_plus/install
from:
```
dependencies:
  flutter_blue_plus: ^1.31.8
```
to:
```
dependencies:
  flutter_blue_plus: 1.31.8
```

This will not auto break apps if people have build process independent of strict developer interaction (git pipeline deploy etc.).

### Logs

```shell
No logs.
```

chipweinberger, 2024-01-18:
This text is no longer needer. 

As of this month, every BLE feature has been implemented, and every open issue fixed.  

This package will be very stable, and breaking changes are not expected.

As for testing on a real device, neither iOS or Android simulator supports BLE. But if they did, it should work fine!

closed 2024-01-18 by chipweinberger

chipweinberger, 2024-01-18:
thanks for filing this issue!

4WayMitja, 2024-01-22:
This was proposed based on:
https://github.com/boskokg/flutter_blue_plus/issues/730

Maybe we should have implemented some check that example app builds on Android and iOS before this package is released in the wild. I do not know if working with iOS git CI/CD is possible on public repositories but it would be great if we could come up with the solutions which tests iOS/Android build for the example app.

Maybe do it with community supported pre-release so that we can check if library version is still working as expected for all devices.
Something like:
https://pub.dev/packages/firebase_core/versions (Prerelease versions of firebase_core)

Probably you (@chipweinberger ) should decide in which direction you would like to go, but ideally this problem should be addressed.