chipweinberger/flutter_blue_plus #762: [Feature]: In the iOS system, some devices establish a Bluetooth connection in the app, the connection is not released after the app is closed.

stengfei, 2024-01-19:
### FlutterBluePlus Version

1.29.12

### Flutter Version

3.16.5

### What OS?

iOS

### OS Version

16.7.4

### Bluetooth Module

unknow

### What is your feature request?

In the iOS system, some devices establish a Bluetooth connection in the app, the connection is not released after the app is closed. As a result, the device cannot be scanned after the app is launched again. Android phones do not have this problem.

### Logs

```shell
In the iOS system, some devices establish a Bluetooth connection in the app, the connection is not released after the app is closed. As a result, the device cannot be scanned after the app is launched again. Android phones do not have this problem.
```

stengfei, 2024-01-19:
When the Bluetooth connection is not released, when the app is started again, the system's Bluetooth connection cannot be obtained by using the FlutterBluePlus.system Devices method.

stengfei, 2024-01-19:
It's not a problem with the FBP component. This problem also occurs when using the iPhone's Bluetooth debugging tool. I don't know how to solve it.

chipweinberger, 2024-01-19:
yes, this does not sound like a FBP problem.

it is a problem with your device firmware.

you need to make sure your device advertises again after the connection is closed.

closed 2024-01-19 by chipweinberger