chipweinberger/flutter_blue_plus #766: [Feature]: Support Appearance values

MrCsabaToth, 2024-01-22:
### FlutterBluePlus Version

1.31.8

### Flutter Version

3.19.0-0.1.pre

### What OS?

Android

### OS Version

12, 13

### Bluetooth Module

.

### What is your feature request?

Appearance type is a Bluetooth LE standard feature.

Many applications have to deal with various types of BLE devices. One way to identify them is to look at the service data in the advertisement. However that is often vague, and it's even advised to first look at the Appearance data, especially when multiple services mentioned in the services list (see https://stackoverflow.com/a/44292698/292502 where it's a Herta Rate Sensor + a Running Cadence and Speed Sensor).

It's extremely costly (and kind of infeasible) to connect to devices just to categorize them. This is why somewhat surprising that iOS is not supporting this standard (#202), however there are already multiple features which iOS holds back but the Android side of FBP supports. This would be another one, and the PR IMHO is not an invasive change.

### Logs

```shell
.
```

MrCsabaToth, 2024-01-22:
Somehow the fork is behind although GitHub doesn't indicate it. There should be only 3 file changes, I'll fix it.

MrCsabaToth, 2024-01-22:
Also ref https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/413

MrCsabaToth, 2024-01-22:
Ok, somehow there was a wrong merge while syncing the fork? I'll probably nuke the fork and recreate the PR. The other thing is that I used an API level 33 call (`ScanRrecord.getAdvertosimgDataMap`
), so I'll add support for level 21+ `ScanRrecord.getBytes`.

chipweinberger, 2024-01-23:
merged. will be in next release 1.31.10

closed 2024-01-23 by chipweinberger

chipweinberger, 2024-01-23:
this code was reverted. see comments in https://github.com/boskokg/flutter_blue_plus/pull/772

MrCsabaToth, 2024-02-12:
New PR: https://github.com/boskokg/flutter_blue_plus/pull/791/

chipweinberger, 2024-02-13:
tested working?

MrCsabaToth, 2024-02-15:
New virgin PR: https://github.com/boskokg/flutter_blue_plus/pull/794

PR #772 (2024-01-23): [Feature] Adding Appearance support (part of advertisement data)
Fixes #766 and Fixes #202