chipweinberger/flutter_blue_plus #768: [Help]: Nothing returned

salex996, 2024-01-22:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

flutter_blue_plus: ^1.31.8

### Flutter Version

3.16.7

### What OS?

Android

### OS Version

14

### Bluetooth Module

Google Tensor G2

### What is your problem?

I followed the example but I still don't have anything in return

The code I have is the following: 

Future<List> findDevices() async {

  List<ScanResult> devices = [];

  var subscription = FlutterBluePlus.scanResults.listen((results) {
    if (results.isNotEmpty) {
      ScanResult r = results.last; // the most recently found device
      devices.add(r);
      print('${r.device.remoteId}: "${r.advertisementData.advName}" found!');
    }
  },
    onError: (e) => print(e),
  );
  await FlutterBluePlus.startScan(timeout: const Duration(seconds: 20), continuousUpdates: true);

  return devices;

Can you please help me? I do not understand why If I use the example of scanning that you have in the main page, I have nothing in devices

Thank you very much for your support 

### Logs

```shell
[GoRouter] setting initial location /
D/permissions_handler( 3688): Bluetooth permission missing in manifest
D/[FBP-Android]( 3688): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android]( 3688): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android]( 3688): [FBP] connectedPeripherals: 0
D/[FBP-Android]( 3688): [FBP] onMethodCall: getAdapterState
D/[FBP-Android]( 3688): [FBP] onMethodCall: turnOn
D/[FBP-Android]( 3688): [FBP] onMethodCall: startScan
D/BluetoothAdapter( 3688): isLeEnabled(): ON
D/BluetoothLeScanner( 3688): onScannerRegistered() - status=0 scannerId=3 mScannerId=0
I/ompany.terrazap( 3688): Background concurrent mark compact GC freed 928148(23MB) AllocSpace objects, 0(0B) LOS objects, 88% free, 3215KB/27MB, paused 201us,5.022ms total 63.416ms
I/ompany.terrazap( 3688): Background concurrent mark compact GC freed 955931(24MB) AllocSpace objects, 0(0B) LOS objects, 88% free, 3243KB/27MB, paused 237us,6.262ms total 59.098ms
I/ompany.terrazap( 3688): Background concurrent mark compact GC freed 950135(23MB) AllocSpace objects, 0(0B) LOS objects, 87% free, 3351KB/27MB, paused 1.426ms,9.181ms total 66.606ms
Application finished.
the Dart compiler exited unexpectedly.
```

chipweinberger, 2024-01-22:
look at readme common problems please

brx-ivank, 2024-01-22:
Same here, nothing returned, I had use bondedDevices and for scanning new devices I had to redirect the user to BL screen and do it over there.

chipweinberger, 2024-01-22:
these are common issues that FBP cannot fix.

closed 2024-01-22 by chipweinberger