chipweinberger/flutter_blue_plus #769: [Feature]: Add Support for Web Bluetooth W3C standard ( preliminary )

diegobes, 2024-01-22:
### FlutterBluePlus Version

all

### Flutter Version

all

### What OS?

All

### OS Version

Supported Chrome

### Bluetooth Module

any

### What is your feature request?

With Chrome ( behind flag ) supporting Web Bluetooth standard, This can be a great addition for Flutter Web Development.

If anyone is interested please let me know

### Logs

```shell
Chrome Web BLE support:
https://developer.chrome.com/docs/capabilities/bluetooth?hl=en

Web Bluetooth W3C Spec:
https://webbluetoothcg.github.io/web-bluetooth/
```

chipweinberger, 2024-01-22:
feel free to open a PR!

chipweinberger, 2024-02-15:
Just some thoughts. To keep all the platform code similar, we should simulate the use of a platform channel. Something like this:

in **_invokeMethod**
```
if (Platform.isWeb) {
    out = await FlutterBluePlusWeb.invokeMethod(method, arguments);
} else {
    out = await _methodChannel.invokeMethod(method, arguments);
}
```

in **_initFlutterBluePlus**
```
if (Platform.isWeb) {
    FlutterBluePlusWeb.setMethodCallHandler(_methodCallHandler);
} else {
    _methodChannel.setMethodCallHandler(_methodCallHandler);
}
```

chipweinberger, 2024-04-27:
I've started on web support: https://github.com/boskokg/flutter_blue_plus/compare/master...feature/web-bluetooth-support

I implemented `startScan`. But it should now be easier for someone else to finish it using this as a guide.

tnc1997, 2024-08-12:
> I implemented `startScan`. It should be pretty straightforward to implement the rest, if someone wants to.

Hi @chipweinberger, thank you for your work on this package!

I hope you don't mind, I have implemented the web platform, based on your initial branch.

What is the recommended approach for testing the web implementation?

chipweinberger, 2024-08-12:
thats awesome. No recommended approach. Just try to test everything :)

chipweinberger, 2024-08-12:
your code looks good. I think stopScan should be a no-op, not unsupported.

tnc1997, 2024-08-13:
> your code looks good. I think stopScan should be a no-op, not unsupported.

> also, please update the readme. etc.

Thank you for your feedback, I have refactored `stopScan` to be a no-op, and also updated the readme.

chipweinberger, 2024-08-29:
@tnc1997 recently created a PR for Web: https://github.com/chipweinberger/flutter_blue_plus/pull/964

However, FBP plans to remain Android/iOS/Mac only for the foreseeable future.

If you'd like to add web support, please do it via wrapper (example: https://pub.dev/packages/flutter_blue_plus_windows).

I'll keep this issue open until a wrapper is created for Web.

tnc1997, 2024-11-13:
An in development fork of Flutter Blue Plus that supports web using a [platform interface](https://docs.flutter.dev/packages-and-plugins/developing-packages#plugin) is available [here](https://github.com/tnc1997/flutter-blue-plus).

Please note that this unofficial fork is not endorsed by Chip Weinberger and is supported on a best efforts basis.

I would like to thank Chip Weinberger for their ongoing work on the existing Android/iOS/macOS package.

yaing11, 2024-12-22:
Hello @chipweinberger . Thank you for your work on this plugin. Do you have plans to add [platform_interface](https://docs.flutter.dev/packages-and-plugins/developing-packages#plugin) to your plugin?  If you want to remain Anroid/iOS/Mac support , no problems. Just add platform interface, please, as it did our friend @tnc1997 .

chipweinberger, 2024-12-22:
@yaing11 , this repo will never support other platforms or a platform interface. It's just not a good use of my time. Sorry.

But I encourage you to create & maintain your own `flutter_blue_plus_web` repo, and I'll link to it.

yaing11, 2024-12-22:
Ohhh, @chipweinberger.  It's a pity, but you won't spent your time. Let me create PR where I will add platform interface to your plugin and you still continue support ONLY 3 platform as you want. It's modern solution and Google recommends to use platform interface in a plugin develoment.

yaing11, 2024-12-22:
> You could even create your own `platform_interface`, and use this repo as your android/iOS/macOS implementation.

I've already created but it needs to sync with your repo each time. It's not good. My app supports 6 platforms - android,ios, macos, linux, web, windows,

yaing11, 2024-12-22:
> Why? I don't want to maintain it as each platform evolves.
You will maintain only your platforms , but with platform_interface. There is no need to maintain platform_interface because it's created only once....

tnc1997, 2024-12-22:
Hi @yaing11, it's great to see your interest in a `platform_interface` for `flutter_blue_plus`.

You may be interested in the previous discussion that we had in [this](https://github.com/chipweinberger/flutter_blue_plus/pull/971) Pull Request for context.

chipweinberger, 2024-12-22:
> platform_interface because it's created only once

That's not really true. Every year the BLE APIs change slightly, and the platform API needs to change too. For example, iOS 17 added `autoconnect` support, Android changed how `FINE_LOCATION` permission works. All of these need to be supported by the platform interface.

Which is why I recommend you create your own package.

yaing11, 2024-12-22:
> I've already created but it needs to sync with your repo each time. It's not good. My app supports 6 platforms - android,ios, macos, linux, web, windows,

@chipweinberger as I wrote my app supports 6 platforms, but for android, ios, macos i want to use your solution because it's very good and if we have platform interface we will have more clean solution.

chipweinberger, 2024-12-22:
> @chipweinberger as I wrote my app supports 6 platforms, but for android, ios, macos i want to use your solution because it's very good and if we have platform interface we will have more clean solution.

I understand. I recommend you create a new package, and I'll link to it in my readme. Perhaps @tnc1997 would help too.

yaing11, 2024-12-22:
> .....................and the platform interface needs to change too. For example, iOS

@tnc1997 many thanks that you joined to our discussion. Do you agree with what @chipweinberger wrote ?

tnc1997, 2024-12-22:
> Perhaps @tnc1997 would help too.

You are welcome to use [this](https://github.com/tnc1997/flutter-blue-plus) unofficial fork of Flutter Blue Plus that supports additional platforms using a [platform interface](https://docs.flutter.dev/packages-and-plugins/developing-packages#plugin) and  is supported on a best efforts basis.

chipweinberger, 2024-12-22:
@tnc1997 , wow! your fork has come a long way! You really ought to give it a proper new name!

tnc1997, 2024-12-22:
> Do you agree with what @chipweinberger wrote?

In theory the platform interface wouldn't need to change if a platform specific implementation changes, however it would need to change if an interface method changes (for example if fields are added/removed to/from a request model).

chipweinberger, 2024-12-22:
@tnc1997 , if you give it a new name (so it's less easier to refer to), I'll link to it in the FBP readme.

tnc1997, 2024-12-22:
> @tnc1997 , if you give it a new name (so it's easier to refer to), I'll link to it in the FBP readme.

Thank you very much for your kind words, it currently uses the same name so that developers only need to update the `dependencies` in their `pubspec.yaml` file, without having to modify all the import statements throughout their application.

chipweinberger, 2024-12-22:
Fair point. What about a more distinguished repo & package name? You could still use FlutterBluePlus as your api entrypoint.

tnc1997, 2024-12-22:
> flutter_blue_plus_universal

That would still require developers to update all the import statements throughout their application with the different package name (flutter_blue_plus_universal).

chipweinberger, 2024-12-22:
True. It's your fork so do what you wish. 

But IMO that's rather trivial. Just make it clear in your README. For example:

[image]

I'd love to link to your implementation. I just don't want it to complicate what "flutter_blue_plus" refers to.

chipweinberger, 2024-12-22:
On second thought, I think i'll link to it regardless. But I strongly encourage a proper new name.

tnc1997, 2024-12-22:
Ultimately it would be great one day in the future to merge the fork into this repository and obsolete the fork but I very much appreciate that you would prefer to keep them separate for the time being.

yaing11, 2024-12-23:
@chipweinberger Do you think it's clean solution???
[image]

Obviously it's bad solution and any plugin without platform interface is not scalable. My app supports 6 platforms. If your plugin had platform interface i would add only 1 line to the code of my app. This 
[image]

You have to agree with me it's very clean and scalable solution.

But now i have to add import statement for each new platform. It's not good solution. 

Please, take a look at most popular plugins in flutter https://pub.dev/packages?q=sdk%3Aflutter&sort=popularity.

Each of them has platform interface. For, example, shared_preferences

[image]

or url_launcher

[image]

All of these plugins are at the top with for more than 1M downloads.

chipweinberger, 2024-12-23:
@yaing11 of course, you're right. But it doesn't matter to me. I want to focus on macOS, iOS, Android only.

You should use @tnc1997 's fork!!

chipweinberger, 2024-12-23:
@yaing11 of course, you're right. 

You should use @tnc1997 's [fork](https://github.com/tnc1997/flutter-blue-plus)!!

Eventually the goal is to merge. Once it's stable and well tested by many people.