chipweinberger/flutter_blue_plus #771: [Help]: Exception has occurred. FlutterBluePlusException (FlutterBluePlusException: setNotifyValue: (code: 5) notifications were not updated)

NutdanaiMee, 2024-01-23:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.14.7

### Flutter Version

3.16.7

### What OS?

Android

### OS Version

Android 9

### Bluetooth Module

ADF-B27

### What is your problem?

I have a problem using Flutter Blue plus. I have connected a device. and after that search for service(0000fff0-0000-1000-8000-00805f9b34fb) and after that search characteristic (0000fff4-0000-1000-8000-00805f9b34fb) by after that await characteristic.setNotifyValue(true);

After that there was a problem. await characteristic.setNotifyValue(true);

the following error Exception has occurred. PlatformException (PlatformException(setNotification, gatt.writeDescriptor() returned false, null, null))

My Connect And Discover Notify value function
`  
void connectToDevice() async {
    deviceConnectionSubscription = foundDevice?.state.listen((state) {
      if (state == BluetoothConnectionState.connected) {
        setState(() {
          isConnected = true;
        });
      } else if (state == BluetoothConnectionState.disconnected) {
        setState(() {
          isConnected = false;
        });
      }
    });
    var lock = new Lock();
    await foundDevice?.connect(autoConnect: true);
    List<BluetoothService>? services = await foundDevice?.discoverServices();
    if (services != null) {
      chr = services;
      print(services);
      if (services.isNotEmpty) {
        for (BluetoothService service in services) {
          if (service.uuid.toString() ==
              "0000fff0-0000-1000-8000-00805f9b34fb") {
            List<BluetoothCharacteristic> characteristics =
                service.characteristics;
            for (BluetoothCharacteristic characteristic in characteristics) {
              if (characteristic.uuid.toString() ==
                  "0000fff4-0000-1000-8000-00805f9b34fb") {
                print("Characteristic found");
                await characteristic.setNotifyValue(true , timeout: 15);

                
                

                characteristic.value.listen((value) {
                  hexNumbere(value);
                
              }
            }
          }
        }
      }
}
  }
` 

### Logs

```shell
D/BluetoothLeScanner(25794): onScannerRegistered() - status=0 scannerId=18 mScannerId=0
D/[FBP-Android](25794): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(25794): connect() - device: C6:27:02:00:E8:02, auto: true
D/BluetoothGatt(25794): registerApp()
D/BluetoothGatt(25794): registerApp() - UUID=c4a44c20-cdb3-4a0c-bbdf-7d15e71d61bd
D/BluetoothGatt(25794): onClientRegistered() - status=0 clientIf=19
D/BluetoothGatt(25794): onClientConnectionState() - status=0 clientIf=19 device=C6:27:02:00:E8:02
D/[FBP-Android](25794): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](25794): [FBP-Android] onMethodCall: connect
D/[FBP-Android](25794): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(25794): discoverServices() - device: C6:27:02:00:E8:02
D/[FBP-Android](25794): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(25794): onConnectionUpdated() - Device=C6:27:02:00:E8:02 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(25794): onSearchComplete() = Device=C6:27:02:00:E8:02 Status=0
D/[FBP-Android](25794): [FBP-Android] onServicesDiscovered: count: 5 status: 0
D/BluetoothGatt(25794): onConnectionUpdated() - Device=C6:27:02:00:E8:02 interval=18 latency=4 timeout=300 status=0
I/flutter (25794): [BluetoothService{remoteId: C6:27:02:00:E8:02, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: C6:27:02:00:E8:02, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a02-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: true, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: C6:27:02:00:E8:02, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a03-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: true, notify: false, indicate: false, authenticatedSignedWrites: false, extendedPropertie
I/flutter (25794): Characteristic found
D/[FBP-Android](25794): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt(25794): setCharacteristicNotification() - uuid: 0000fff4-0000-1000-8000-00805f9b34fb enable: true
D/[FBP-Android](25794): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(25794): discoverServices() - device: C6:27:02:00:E8:02
D/BluetoothGatt(25794): onSearchComplete() = Device=C6:27:02:00:E8:02 Status=0
D/[FBP-Android](25794): [FBP-Android] onServicesDiscovered: count: 5 status: 0
D/[FBP-Android](25794): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
I/flutter (25794): [BluetoothService{remoteId: C6:27:02:00:E8:02, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: C6:27:02:00:E8:02, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a02-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: true, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: C6:27:02:00:E8:02, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a03-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: true, notify: false, indicate: false, authenticatedSignedWrites: false, extendedPropertie
I/flutter (25794): Characteristic found
D/[FBP-Android](25794): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt(25794): setCharacteristicNotification() - uuid: 0000fff4-0000-1000-8000-00805f9b34fb enable: true
D/[FBP-Android](25794): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(25794): discoverServices() - device: C6:27:02:00:E8:02
D/BluetoothGatt(25794): onSearchComplete() = Device=C6:27:02:00:E8:02 Status=0
D/[FBP-Android](25794): [FBP-Android] onServicesDiscovered: count: 5 status: 0
D/[FBP-Android](25794): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
```

chipweinberger, 2024-01-23:
this means your device refused to enable notifications 

nothing fbp can do.

you should update fbp

closed 2024-01-23 by chipweinberger

NutdanaiMee, 2024-01-24:
But I have another BLE device that Notify can use.

NutdanaiMee, 2024-01-24:
So what causes this problem?

Exception has occurred.
PlatformException (PlatformException(setNotification, gatt.writeDescriptor() returned false, null, null))

chipweinberger, 2024-01-24:
> So what causes this problem?

your device refused to enable notifications, and fbp is reporting the error.

Its possible your device actually `did` enable notifications, but returned incorrect data.

chipweinberger, 2024-01-24:
im wrong. this was caused by a bug in FBP.

https://github.com/boskokg/flutter_blue_plus/issues/509

you need to update fbp.