chipweinberger/flutter_blue_plus #774: google  play catch  App ANR

coderXiezhiheng, 2024-01-25:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.16.7

### Flutter Version

3.0.0

### What OS?

Android

### OS Version

13

### Bluetooth Module

FlutterBluePlus.turnOn();

### What is your problem?

i want to know the problem is the plugin  crash ?

### Logs

```shell
at jdk.internal.misc.Unsafe.park (Native method)
  at java.util.concurrent.locks.LockSupport.parkNanos (LockSupport.java:252)
  at java.util.concurrent.CompletableFuture$Signaller.block (CompletableFuture.java:1842)
  at java.util.concurrent.ForkJoinPool.unmanagedBlock (ForkJoinPool.java:3466)
  at java.util.concurrent.ForkJoinPool.managedBlock (ForkJoinPool.java:3437)
  at java.util.concurrent.CompletableFuture.timedGet (CompletableFuture.java:1915)
  at java.util.concurrent.CompletableFuture.get (CompletableFuture.java:2071)
  at com.android.modules.utils.SynchronousResultReceiver.awaitResultNoInterrupt (SynchronousResultReceiver.java:210)
  at android.bluetooth.BluetoothAdapter$2.apply (BluetoothAdapter.java:1299)
  at android.bluetooth.BluetoothAdapter$2.apply (BluetoothAdapter.java:1291)
  at android.app.PropertyInvalidatedCache.recompute (PropertyInvalidatedCache.java:714)
  at android.app.PropertyInvalidatedCache.query (PropertyInvalidatedCache.java:931)
  at android.os.IpcDataCache.query (IpcDataCache.java:360)
  at android.bluetooth.BluetoothAdapter.getStateInternal (BluetoothAdapter.java:1331)
  at android.bluetooth.BluetoothAdapter.getState (BluetoothAdapter.java:1359)
  at android.bluetooth.BluetoothAdapter.isEnabled (BluetoothAdapter.java:1161)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$0$com-lib-flutter_blue_plus-FlutterBluePlusPlugin (FlutterBluePlusPlugin.java:329)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda1.op (unavailable:4)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.ensurePermissions (FlutterBluePlusPlugin.java:1211)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall (FlutterBluePlusPlugin.java:327)
  at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage (MethodChannel.java:262)
  at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler (DartMessenger.java:295)
  at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger (DartMessenger.java:319)
  at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run (unavailable:12)
  at android.os.Handler.handleCallback (Handler.java:942)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loopOnce (Looper.java:346)
  at android.os.Looper.loop (Looper.java:475)
  at android.app.ActivityThread.main (ActivityThread.java:7950)
  at java.lang.reflect.Method.invoke (Native method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:548)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:942)
```

chipweinberger, 2024-01-25:
seems like a bug in Android.

chipweinberger, 2024-01-25:
we don't use runblocking afaik, so not sure if relevant

https://stackoverflow.com/questions/75886702/anr-crash-sun-misc-unsafe-parknative-method

chipweinberger, 2024-01-25:
crashing in here

```
/**
     * Disables the current thread for thread scheduling purposes, for up to
     * the specified waiting time, unless the permit is available.
     *
     * <p>If the specified waiting time is zero or negative, the
     * method does nothing. Otherwise, if the permit is available then
     * it is consumed and the call returns immediately; otherwise the
     * current thread becomes disabled for thread scheduling purposes
     * and lies dormant until one of four things happens:
     *
     * <ul>
     * <li>Some other thread invokes {@link [#unpark](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;drc=228276a74bc36e8c0f0fadbb4a15ef013ff6ce24;l=175?q=LockSupport.java) unpark} with the
     * current thread as the target; or
     *
     * <li>Some other thread {@linkplain [Thread#interrupt](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;bpv=1;bpt=1;l=229?q=LockSupport.java&gsn=Thread#interrupt&gs=KYTHE%3A%2F%2Fkythe%3A%2F%2Fandroid.googlesource.com%2Fplatform%2Fsuperproject%2Fmain%2F%2Fmain%3Flang%3Djava%3Fpath%3Djava.lang.Thread%233a950f954e66f809e394e48a06e566e87a5a55ee89ff422f19ba9b64117922e3) interrupts}
     * the current thread; or
     *
     * <li>The specified waiting time elapses; or
     *
     * <li>The call spuriously (that is, for no reason) returns.
     * </ul>
     *
     * <p>This method does <em>not</em> report which of these caused the
     * method to return. Callers should re-check the conditions which caused
     * the thread to park in the first place. Callers may also determine,
     * for example, the interrupt status of the thread, or the elapsed time
     * upon return.
     *
     * @param blocker the synchronization object responsible for this
     *        thread parking
     * @param nanos the maximum number of nanoseconds to wait
     * @since 1.6
     */
    public static [void](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;bpv=1;bpt=1;l=248?q=LockSupport.java&gsn=void&gs=KYTHE%3A%2F%2Fkythe%3A%2F%2Fandroid.googlesource.com%2Fplatform%2Fsuperproject%2Fmain%2F%2Fmain%3Flang%3Djava%23void%2523builtin) [parkNanos](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;bpv=1;bpt=1;l=248?q=LockSupport.java&gsn=parkNanos&gs=KYTHE%3A%2F%2FKvYCCrgBa3l0aGU6Ly9hbmRyb2lkLmdvb2dsZXNvdXJjZS5jb20vcGxhdGZvcm0vc3VwZXJwcm9qZWN0L21haW4vL21haW4_bGFuZz1qYXZhP3BhdGg9amF2YS51dGlsLmNvbmN1cnJlbnQubG9ja3MuTG9ja1N1cHBvcnQjMzA3N2QwYTFjYWFhMDVhYzkzYzA0Y2U0NmEyNTA3MmYzNzM0OGE2MWIwMTNhYmU3NjJmOTU0NjIwYmIxYTg3Mgq4AWt5dGhlOi8vYW5kcm9pZC5nb29nbGVzb3VyY2UuY29tL3BsYXRmb3JtL3N1cGVycHJvamVjdC9tYWluLy9tYWluP2xhbmc9amF2YT9wYXRoPWphdmEudXRpbC5jb25jdXJyZW50LmxvY2tzLkxvY2tTdXBwb3J0IzNlOGIyZGU1NDkwMzE4NDFjZGNmODc4YTVhNzUyZjMwYWQxMzYyNTdkMzBlNmViYjU1OTU1OTM0ZmQxYzQ3Yzc%3D)([Object](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;bpv=1;bpt=1;l=248?q=LockSupport.java&gsn=Object&gs=KYTHE%3A%2F%2Fkythe%3A%2F%2Fandroid.googlesource.com%2Fplatform%2Fsuperproject%2Fmain%2F%2Fmain%3Flang%3Djava%3Fpath%3Djava.lang.Object%23003bede7cf0c65bd777211eba9ca868f918a7093291b3f5afe8d778306030117) [blocker](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;bpv=1;bpt=1;l=248?q=LockSupport.java&gsn=blocker&gs=KYTHE%3A%2F%2FKvYCCrgBa3l0aGU6Ly9hbmRyb2lkLmdvb2dsZXNvdXJjZS5jb20vcGxhdGZvcm0vc3VwZXJwcm9qZWN0L21haW4vL21haW4_bGFuZz1qYXZhP3BhdGg9amF2YS51dGlsLmNvbmN1cnJlbnQubG9ja3MuTG9ja1N1cHBvcnQjM2ViMzIzNDE3NzM2ZTQ3MzdjZjdhMWFjNjYzYTAxYThkOTkxYmMzMGQ5MjljODExMzExYTE2ZWNhZDk3MTdhNAq4AWt5dGhlOi8vYW5kcm9pZC5nb29nbGVzb3VyY2UuY29tL3BsYXRmb3JtL3N1cGVycHJvamVjdC9tYWluLy9tYWluP2xhbmc9amF2YT9wYXRoPWphdmEudXRpbC5jb25jdXJyZW50LmxvY2tzLkxvY2tTdXBwb3J0I2YzNTBlOGVhNjk0NzFkNGEzOWFlOGRhZDgwYjc1YWQ5YmUyNGQzYmUyODJlY2EyMzQyMTkxOTVjZTdlNGQ3ODY%3D), [long](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;bpv=1;bpt=1;l=248?q=LockSupport.java&gsn=long&gs=KYTHE%3A%2F%2Fkythe%3A%2F%2Fandroid.googlesource.com%2Fplatform%2Fsuperproject%2Fmain%2F%2Fmain%3Flang%3Djava%23long%2523builtin) [nanos](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;bpv=1;bpt=1;l=248?q=LockSupport.java&gsn=nanos&gs=KYTHE%3A%2F%2FKvYCCrgBa3l0aGU6Ly9hbmRyb2lkLmdvb2dsZXNvdXJjZS5jb20vcGxhdGZvcm0vc3VwZXJwcm9qZWN0L21haW4vL21haW4_bGFuZz1qYXZhP3BhdGg9amF2YS51dGlsLmNvbmN1cnJlbnQubG9ja3MuTG9ja1N1cHBvcnQjMGVmZDc0NWQ2OGIyOGUzNGY5YjM0Nzg5MWJjYTMzMDk0ZTAyMGE5MWM2YjE2NTMxMjQzNWIwMmYwZDcwYmQyYwq4AWt5dGhlOi8vYW5kcm9pZC5nb29nbGVzb3VyY2UuY29tL3BsYXRmb3JtL3N1cGVycHJvamVjdC9tYWluLy9tYWluP2xhbmc9amF2YT9wYXRoPWphdmEudXRpbC5jb25jdXJyZW50LmxvY2tzLkxvY2tTdXBwb3J0IzgyYWYyYTAxYjIwMDQwMTBjZDRhYmVmYTJhOGQzZTFmY2FmYWRlYTViMzMwZmRjMWU4ZjJlNjBhNzRhMTM4ZGY%3D)) {
        if ([nanos](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;drc=228276a74bc36e8c0f0fadbb4a15ef013ff6ce24;l=248?q=LockSupport.java) > 0) {
[... cut ...] 248?q=LockSupport.java));
            [setBlocker](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;drc=228276a74bc36e8c0f0fadbb4a15ef013ff6ce24;l=142?q=LockSupport.java)([t](https://cs.android.com/android/platform/superproject/main/+/main:libcore/ojluni/src/main/java/java/util/concurrent/locks/LockSupport.java;drc=228276a74bc36e8c0f0fadbb4a15ef013ff6ce24;l=250?q=LockSupport.java), null);
        }
    }
```

coderXiezhiheng, 2024-01-26:
So is that what causes ANR? Is there a way to fix it ？

chipweinberger, 2024-01-26:
not sure

chipweinberger, 2024-02-06:
closing since it is unclear if this is a bug caused by FBP. (probably not)

closed 2024-02-06 by chipweinberger