chipweinberger/flutter_blue_plus #775: [Help]: The connection is automatically disconnected a few seconds after successful connection

roger66, 2024-01-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.12

### Flutter Version

3.13.1

### What OS?

Android

### OS Version

12

### Bluetooth Module

unknown

### What is your problem?

[image]

{disconnect_reason_code: 8, disconnect_reason_string: CONNECTION_TIMEOUT, remote_id: 77:25:34:49:CF:15, connection_state: 0}

The connection was successful. Why did I return a timeout？

### Logs

```shell
I/flutter ( 8058): [FBP] <connect> args: {remote_id: 77:25:34:49:CF:15, auto_connect: 0}
D/[FBP-Android]( 8058): [FBP] onMethodCall: connect
D/BluetoothGatt( 8058): connect() - device: 77:25:34:49:CF:15, auto: false, eattSupport: false
D/BluetoothGatt( 8058): registerApp()
D/BluetoothGatt( 8058): registerApp() - UUID=3b84269a-43d9-49ec-a706-baab4786ea28
D/BluetoothGatt( 8058): onClientRegistered() - status=0 clientIf=8
I/flutter ( 8058): [FBP] <connect> result: true
D/BluetoothGatt( 8058): onClientConnectionState() - status=0 clientIf=8 device=77:25:34:49:CF:15
D/[FBP-Android]( 8058): [FBP] onConnectionStateChange:connected
D/[FBP-Android]( 8058): [FBP]   status: SUCCESS
I/flutter ( 8058): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: 77:25:34:49:CF:15, connection_state: 1}
I/flutter ( 8058): [FBP] <readRssi> args: 77:25:34:49:CF:15
D/[FBP-Android]( 8058): [FBP] onMethodCall: readRssi
D/BluetoothGatt( 8058): readRssi() - device: 77:25:34:49:CF:15
D/[FBP-Android]( 8058): [FBP] onReadRemoteRssi:
D/[FBP-Android]( 8058): [FBP]   rssi: -33
D/[FBP-Android]( 8058): [FBP]   status: GATT_SUCCESS (0)
I/flutter ( 8058): [FBP] <readRssi> result: true
I/flutter ( 8058): [FBP] [[ OnReadRssi ]] result: {error_string: GATT_SUCCESS, rssi: -33, success: 1, remote_id: 77:25:34:49:CF:15, error_code: 0}
D/BluetoothGatt( 8058): onConnectionUpdated() - Device=77:25:34:49:CF:15 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 8058): onConnectionUpdated() - Device=77:25:34:49:CF:15 interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt( 8058): onClientConnectionState() - status=8 clientIf=8 device=77:25:34:49:CF:15
D/[FBP-Android]( 8058): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android]( 8058): [FBP]   status: CONNECTION_TIMEOUT
D/BluetoothGatt( 8058): close()
D/BluetoothGatt( 8058): unregisterApp() - mClientIf=8
I/flutter ( 8058): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 8, disconnect_reason_string: CONNECTION_TIMEOUT, remote_id: 77:25:34:49:CF:15, connection_state: 0}
```

roger66, 2024-01-30:
I tried to connect the device natively using Android and it didn't disconnect automatically

chipweinberger, 2024-01-30:
`case 0x08: return "CONNECTION_TIMEOUT"; // The link supervision timeout has expired for a given connection.`

not sure the reason. unlikely something fbp can fix. device or phone issue

roger66, 2024-01-31:
Thanks for replying, but I tested the same phone and the same device with Android native code, and there was no timeout

roger66, 2024-01-31:
FBP indicates that a timeout message is displayed a few seconds after the connection is successful

chipweinberger, 2024-01-31:
I know. CONNECTION_TIMEOUT. 

This is a bad name. it does not mean your "connection timed out." it means your device did not respond to a periodic link supervision request quickly enough.

I'm not sure why your native code might be working. But this is as much as I can help you.

chipweinberger, 2024-01-31:
I've released 1.31.13 that changes the name `CONNECTION_TIMEOUT` -> `LINK_SUPERVISION_TIMEOUT`

roger66, 2024-01-31:
Ok, Thanks

closed 2024-01-31 by chipweinberger

zfcflower, 2024-06-09:
I found the same problem. The first connection to the device was normal. When I exited the first connection and entered again, it would automatically disconnect.
@roger66 How did you solve it in the end?
`I/flutter (31075): [FBP] <connect> args: {remote_id: E0:F8:98:A3:66:D3, auto_connect: 0}
D/[FBP-Android](31075): [FBP] onMethodCall: connect
D/BluetoothGatt(31075): connect() - device: E0:F8:98:A3:66:D3, auto: false, eattSupport: false
D/BluetoothGatt(31075): registerApp()
D/BluetoothGatt(31075): registerApp() - UUID=431b7140-22de-4426-87ba-f6c132d3b367
I/flutter (31075): [FBP] <connect> result: true
D/BluetoothGatt(31075): onClientRegistered() - status=0 clientIf=20
D/BluetoothGatt(31075): onClientConnectionState() - status=0 clientIf=20 device=E0:F8:98:A3:66:D3
D/[FBP-Android](31075): [FBP] onConnectionStateChange:connected
D/[FBP-Android](31075): [FBP]   status: SUCCESS
I/flutter (31075): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: E0:F8:98:A3:66:D3, connection_state: 1}
I/flutter (31075): [FBP] <requestMtu> args: {remote_id: E0:F8:98:A3:66:D3, mtu: 512}
D/[FBP-Android](31075): [FBP] onMethodCall: requestMtu
D/BluetoothGatt(31075): configureMTU() - device: E0:F8:98:A3:66:D3 mtu: 512
I/flutter (31075): [FBP] <requestMtu> result: true
D/BluetoothGatt(31075): onConnectionUpdated() - Device=E0:F8:98:A3:66:D3 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(31075): onConfigureMTU() - Device=E0:F8:98:A3:66:D3 mtu=23 status=0
D/[FBP-Android](31075): [FBP] onMtuChanged:
D/[FBP-Android](31075): [FBP]   mtu: 23
D/[FBP-Android](31075): [FBP]   status: GATT_SUCCESS (0)
I/flutter (31075): [FBP] [[ OnMtuChanged ]] result: {error_string: GATT_SUCCESS, success: 1, remote_id: E0:F8:98:A3:66:D3, error_code: 0, mtu: 23}
I/flutter (31075): [FBP] <connect> args: {remote_id: E0:F8:98:A3:66:D3, auto_connect: 0}
D/[FBP-Android](31075): [FBP] onMethodCall: connect
D/[FBP-Android](31075): [FBP] already connected
I/flutter (31075): [FBP] <connect> result: false
I/flutter (31075): [FBP] <readRssi> args: E0:F8:98:A3:66:D3
D/[FBP-Android](31075): [FBP] onMethodCall: readRssi
D/BluetoothGatt(31075): readRssi() - device: E0:F8:98:A3:66:D3
I/flutter (31075): [FBP] <readRssi> result: true
D/[FBP-Android](31075): [FBP] onReadRemoteRssi:
D/[FBP-Android](31075): [FBP]   rssi: -41
D/[FBP-Android](31075): [FBP]   status: GATT_SUCCESS (0)
I/flutter (31075): [FBP] [[ OnReadRssi ]] result: {error_string: GATT_SUCCESS, rssi: -41, success: 1, remote_id: E0:F8:98:A3:66:D3, error_code: 0}
D/BluetoothGatt(31075): onConnectionUpdated() - Device=E0:F8:98:A3:66:D3 interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt(31075): onClientConnectionState() - status=8 clientIf=20 device=E0:F8:98:A3:66:D3
D/[FBP-Android](31075): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](31075): [FBP]   status: LINK_SUPERVISION_TIMEOUT
D/BluetoothGatt(31075): close()
D/BluetoothGatt(31075): unregisterApp() - mClientIf=20
I/flutter (31075): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 8, disconnect_reason_string: LINK_SUPERVISION_TIMEOUT, remote_id: E0:F8:98:A3:66:D3, connection_state: 0}
`