chipweinberger/flutter_blue_plus #778: [Help]: could not find callback wrapper

Rizza-Rabe, 2024-01-31:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.12

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 13.

### Bluetooth Module

AT-09 BLE

### What is your problem?

Does not show devices. Always throw: could not find callback wrapper

### Logs

```shell
I/BluetoothAdapter( 4064): STATE_ON
I/BluetoothAdapter( 4064): STATE_ON
D/BluetoothLeScanner( 4064): could not find callback wrapper
```

chipweinberger, 2024-01-31:
show more logs from the example app.

chipweinberger, 2024-01-31:
https://copyprogramming.com/howto/android-btle-cannot-find-callback-wrapper#android-btle-cannot-find-callback-wrapper

check your permissions?

Rizza-Rabe, 2024-01-31:
if (await Permission.bluetoothScan.isDenied) {
      await Permission.bluetoothScan.request();
    }
    if (await Permission.bluetoothConnect.isDenied) {
      await Permission.bluetoothConnect.request();
    }
    if (await Permission.location.isDenied) {
      await Permission.location.request();
    }

    var subscription = FlutterBluePlus.onScanResults.listen((result) {
      Utility().printLog("OnScanResult: $result");
    });

    FlutterBluePlus.cancelWhenScanComplete(subscription);

    await FlutterBluePlus.startScan(
      timeout: const Duration(seconds: 10),
    );

Rizza-Rabe, 2024-01-31:
D/[FBP-Android](14010): [FBP] onMethodCall: startScan
I/ViewRootImpl@1c09fe3[MainActivity](14010): MSG_WINDOW_FOCUS_CHANGED 0 0
I/ViewRootImpl@1c09fe3[MainActivity](14010): stopped(false) old = false
I/DecorView(14010): notifyKeepScreenOnChanged: keepScreenOn=false
I/MSHandlerLifeCycle(14010): removeMultiSplitHandler: no exist. decor=DecorView@866de74[MainActivity]
I/ViewRootImpl@1c09fe3[MainActivity](14010): performTraversals params={(0,0)(fillxfill) sim={adjust=resize forwardNavigation} ty=BASE_APPLICATION fmt=TRANSLUCENT wanim=0x1030001
I/ViewRootImpl@1c09fe3[MainActivity](14010):   fl=81810100
I/ViewRootImpl@1c09fe3[MainActivity](14010):   pfl=12020040
I/ViewRootImpl@1c09fe3[MainActivity](14010):   vsysui=500
I/ViewRootImpl@1c09fe3[MainActivity](14010):   bhv=DEFAULT
I/ViewRootImpl@1c09fe3[MainActivity](14010):   fitSides= naviIconColor=0}
I/ViewRootImpl@1c09fe3[MainActivity](14010): performTraversals mFirst=false windowShouldResize=false viewVisibilityChanged=false mForceNextWindowRelayout=false params={(0,0)(fillxfill) sim={adjust=resize forwardNavigation} ty=BASE_APPLICATION fmt=TRANSLUCENT wanim=0x1030001
I/ViewRootImpl@1c09fe3[MainActivity](14010):   fl=81810100
I/ViewRootImpl@1c09fe3[MainActivity](14010):   pfl=12020040
I/ViewRootImpl@1c09fe3[MainActivity](14010):   vsysui=500
I/ViewRootImpl@1c09fe3[MainActivity](14010):   bhv=DEFAULT
I/ViewRootImpl@1c09fe3[MainActivity](14010):   fitSides= naviIconColor=0}
I/ViewRootImpl@1c09fe3[MainActivity](14010): updateBlastSurfaceIfNeeded mBlastBufferQueue=0xb40000775b660c00 isSameSurfaceControl=true
I/BLASTBufferQueue(14010): update, w= 1080 h= 2408 mName = ViewRootImpl@1c09fe3[MainActivity] mNativeObject= 0xb40000775b660c00 sc.mNativeObject= 0xb4000077c7b187a0 format= -3 caller= android.view.ViewRootImpl.updateBlastSurfaceIfNeeded:2898 android.view.ViewRootImpl.relayoutWindow:9847 android.view.ViewRootImpl.performTraversals:3884 android.view.ViewRootImpl.doTraversal:3116 android.view.ViewRootImpl$TraversalRunnable.run:10885 android.view.Choreographer$CallbackRecord.run:1301 
I/ViewRootImpl@1c09fe3[MainActivity](14010): Relayout returned: old=(0,0,1080,2408) new=(0,0,1080,2408) req=(1080,2408)0 dur=10 res=0x0 s={true 0xb4000077d1226000} ch=false seqId=0
I/ViewRootImpl@1c09fe3[MainActivity](14010): updateBoundsLayer: t = android.view.SurfaceControl$Transaction@5689421 sc = Surface(name=Bounds for - com.zz.arduino.training.pad.training_pad/com.zz.arduino.training.pad.training_pad.MainActivity@1)/@0xdcdc791 frame = 3
I/ViewRootImpl@1c09fe3[MainActivity](14010): mWNT: t=0xb40000775b78c200 mBlastBufferQueue=0xb40000775b660c00 fn= 3 caller= android.view.ViewRootImpl.prepareSurfaces:2985 android.view.ViewRootImpl.performTraversals:4233 android.view.ViewRootImpl.doTraversal:3116 
I/ViewRootImpl@1c09fe3[MainActivity](14010): MSG_WINDOW_FOCUS_CHANGED 1 0
I/ViewRootImpl@1c09fe3[MainActivity](14010): mThreadedRenderer.initializeIfNeeded()#2 mSurface={isValid=true 0xb4000077d1226000}
D/InputMethodManager(14010): startInputInner - Id : 0
I/InputMethodManager(14010): startInputInner - mService.startInputOrWindowGainedFocus
D/[FBP-Android](14010): [FBP] onMethodCall: stopScan
I/BluetoothAdapter(14010): STATE_ON
I/BluetoothAdapter(14010): STATE_ON
D/BluetoothLeScanner(14010): could not find callback wrapper

chipweinberger, 2024-02-03:
use example app.

closed 2024-02-06 by chipweinberger

Xelopakson, 2024-03-29:
did you fix this ?

chipweinberger, 2024-03-29:
@Xelopakson what is your problem? Why do you want this to not be logged?

chipweinberger, 2024-03-29:
Seems to come from this code in Android.

```
    /**
     * Stops an ongoing Bluetooth LE scan.
     * <p>
     * Requires {@link android.Manifest.permission#BLUETOOTH_ADMIN} permission.
     *
     * @param callback
     */
    public void stopScan(ScanCallback callback) {
        checkAdapterState();
        synchronized (mLeScanClients) {
            BleScanCallbackWrapper wrapper = mLeScanClients.remove(callback);
            if (wrapper == null) {
                if (DBG) Log.d(TAG, "could not find callback wrapper");
                return;
            }
            wrapper.stopLeScan();
        }
    }
```

---

chipweinberger, 2024-03-29:
> The log message D/BluetoothLeScanner: could not find callback wrapper appears whenever Android's bluetooth scanning APIs are told top stop scanning for an app when scanning has not started. 
> 
> This is usually safe to ignore

https://stackoverflow.com/questions/40862767/bluetoothlescanner-could-not-find-callback-wrapper

chipweinberger, 2024-03-29:
fixed this issue: https://github.com/boskokg/flutter_blue_plus/commit/69ccb70711180908d5474ab28be1e1019afea134