chipweinberger/flutter_blue_plus #780: [Help]: connecting to a know device.

nissaba, 2024-01-31:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.12

### Flutter Version

3.16.5

### What OS?

iOS

### OS Version

17.2 

### Bluetooth Module

Nordic NRF

### What is your problem?

I am tryng to find out how I can to a device for withch I have saved the information from. Some API lets me call a method like API.connect(name, UUID)

I seen in the doc for flutter_blue_plus that I can scan for a name, but can I ask the API to just find the device and optionally scan for it before trying to connect? 

Such a method would reduce code complexity compare to scanning for a name matcha and then connect to it if found. 

### Logs

```shell
no logs
```

didacroda, 2024-02-01:
```
String remoteId = '05:A4:22:31:F7:ED'; // remoteId example in Android
BluetoothDevice device = BluetoothDevice.fromId(remoteId);
await device.connect();
```

chipweinberger, 2024-02-02:
If you have the uuid from iOS, yes, you can reuse it for connection just like on Android.

But note, it periodically changes. So If connection fails you should probably rescan and get the new uuid.

closed 2024-02-02 by chipweinberger