chipweinberger/flutter_blue_plus #781: [Help]: setNotityValue(true) error

roger66, 2024-02-01:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.12

### Flutter Version

3.16.9

### What OS?

Android

### OS Version

10

### Bluetooth Module

unknown

### What is your problem?

setNotifyValue return GATT_WRITE_NOT_PERMITTED，but FBP1.1.3 is success

### Logs

```shell
D/[FBP-Android]( 6078): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt( 6078): setCharacteristicNotification() - uuid: b044efa0-71ed-11ec-a494-020017000b7b enable: true
E/[FBP-Android]( 6078): [FBP] onDescriptorWrite:
E/[FBP-Android]( 6078): [FBP]   chr: b044efa0-71ed-11ec-a494-020017000b7b
E/[FBP-Android]( 6078): [FBP]   desc: 2902
E/[FBP-Android]( 6078): [FBP]   status: GATT_WRITE_NOT_PERMITTED (3)
E/flutter ( 6078): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException | setNotifyValue | android-code: 3 | GATT_WRITE_NOT_PERMITTED
E/flutter ( 6078): #0      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:280:11)
```

roger66, 2024-02-01:
peripheral's descriptor only have PERMISSION_READ，use FBP1.13 no problem

chipweinberger, 2024-02-02:
If you want the same behavior as 1.1.3, you can just ignore the error.

FBP 1.1.3 did not check for errors. It had no error handling support.

closed 2024-02-02 by chipweinberger

roger66, 2024-02-02:
After ignoring，Does onValueReceived work？

chipweinberger, 2024-02-02:
depends on your device!

roger66, 2024-02-02:
Is it the same if I don't call setNotifyValue?

chipweinberger, 2024-02-02:
not the same. it still sends the request to your device.

roger66, 2024-02-02:
If I ignore the setNofityValue error

roger66, 2024-02-02:
E/flutter (31546): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException | writeCharacteristic | android-code: 133 | UNKNOWN_GATT_ERROR (133)
E/flutter (31546): #0      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:215:9)
E/flutter (31546): <asynchronous suspension>
E/flutter (31546): #1      BleUtils._subpackageSend (package:test_ble/ble_utils.dart:204:5)
E/flutter (31546): <asynchronous suspension>
E/flutter (31546):

roger66, 2024-02-02:
writeCharacteristic not work

roger66, 2024-02-02:
Before I call setNotifyValue，writeCharacteristic can work

chipweinberger, 2024-02-02:
you should be looking at the logs on your device. these are GATT errors. i.e. device errors.