chipweinberger/flutter_blue_plus #782: [Help]: App not always seeing the advertisement of a device

amasuke55, 2024-02-01:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

Issue persists through a lot of versions

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

13

### Bluetooth Module

Nordic nRF52805

### What is your problem?

The App randomly refuses to list a device again when scanning for ble devices.
Usually it works, until the device gets disconnected and then this behaviour can start.

A fix I found is going into the android Bluetooth settings and scanning there. Once the OSs scanning found the device, it will then show up in the App.
Sometimes not even that is fixing it.

This does not produce exceptions and I do not know what could be causing this.
The results are simply being shown with the same logic used in the example code.
Listening to onScanResults and listing the ScanResultTile.

While the App is not showing the advertising device anymore, another mobilephone with the same app on it, can find it. 
So it does not seem to be an issue of the device not advertising correctly.

I already tried different AndroidScanModes. Only on LowPower mode it happens less frequently, but still happens.

### Logs

```shell
Unfortunately no exceptions or log entries from this issue.
```

chipweinberger, 2024-02-02:
Does it happen without `removeIfGone`? does it happen if you remove both `continuousUpdates` & `removeIfGone`?

you can try debugging this yourself. see README. I'd suggest adding additional logs to the scanning code in Dart & Java side.

amasuke55, 2024-02-05:
Hello, thanks for the reply, It does happen without continuousUpdates & removeIfGone too.
I cloned the repo and am trying to figure out what's going on.

chipweinberger, 2024-02-05:
does your device stop advertising after connected to?