chipweinberger/flutter_blue_plus #784: [Help]: Not possible to discover services immediately after the connection function has completed

simon-iversen, 2024-02-04:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.13

### Flutter Version

3.16.9

### What OS?

iOS

### OS Version

iOS 16.1.1

### Bluetooth Module

Nordic nRF5

### What is your problem?

Discovering the services right after the connection is not possible, specifically this code:

```
// dev is of type BluetoothDevice
await dev.connect(mtu: null, autoConnect: true);
List<BluetoothService> services = await dev.discoverServices().timeout(const Duration(seconds: 10));
```
It gives the following error:
> flutter: FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected

When doing this, it works fine:
```
await dev.connect(mtu: null, autoConnect: true);
while (dev.isConnected == false) {
    await Future.delayed(Duration(milliseconds: 10));
}
List<BluetoothService> services = await dev.discoverServices().timeout(const Duration(seconds: 10));
```

This is not a big issue, as I have a workaround, but I'm wondering if this is a bug or intended behaviour? The first snippet worked fine with flutter_blue_plus v1.4.0

### Logs

```shell
Restarted application in 2,000ms.
flutter: unhandled element <inkscape:path-effect/>; Picture key: Svg loader
flutter: unhandled element <sodipodi:namedview/>; Picture key: Svg loader
flutter: unhandled element <metadata/>; Picture key: Svg loader
[FBP-iOS] handleMethodCall: flutterHotRestart
[FBP-iOS] initializing CBCentralManager
[FBP-iOS] initializing checkForMtuChangesTimer
[FBP-iOS] disconnectAllDevices(flutterHotRestart)
[FBP-iOS] connectedPeripherals: 0
[FBP-iOS] handleMethodCall: getAdapterState
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn
[FBP-iOS] handleMethodCall: startScan
[FBP-iOS] handleMethodCall: connect
flutter: ----------------FIREBASE CRASHLYTICS----------------
flutter: FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected
flutter: 
#0      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:228:7)
#1      ConnectFunctions.connect (package:flutter_realtime_chart_example/services/connect_functions.dart:68:21)
<asynchronous suspension>
#2      _ConnectProgressorState._connectFunc (package:flutter_realtime_chart_example/modules/connect_progressor.dart:85:7)
<asynchronous suspension>
#3      _ConnectProgressorState.build.<anonymous closure>.<anonymous closure>.<anonymous closure>.<anonymous closure>.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:flutter_realtime_chart_example/modules/connect_progressor.dart:269:55)
<asynchronous suspension>
flutter: ----------------------------------------------------
10.20.0 - [FirebaseCrashlytics][I-CLS000000] Crashlytics skipped rotating the Install ID during urgent mode because it is run on the main thread, which can't succeed. This can happen if the app crashed the last run and Crashlytics is uploading urgently.
[FBP-iOS] didConnectPeripheral
```

chipweinberger, 2024-02-04:
this is expected

you are using autoconnect

read the docs

chipweinberger, 2024-02-04:
> The first snippet worked fine with flutter_blue_plus v1.4.0

autoconnect was not supported on iOS in v1.4.0

closed 2024-02-04 by chipweinberger