chipweinberger/flutter_blue_plus #788: [Help]: Does not write correct commands to device  every second time.

xyzbilal, 2024-02-07:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.13.8  and 1.31.13

### Flutter Version

3.16.9  and 3.10.5

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

esp 32

### What is your problem?

Hi, firstly thanks for your great effort. 
We have a veird problem on xiomi device android Android 14 installed. on other versions of android everything is fine.

the problem is  this. 
we create rfid inventory request with following  hex commands. and write to esp with 100 milliseconds delays sequentily and we see that device take commands  first time. 

```

Receive Command
8001000000005358002b000000080001008b0000000f00fb536f757263655f30000000000a006a00000006

Receive Command
8001000000005358002b000000080001008b0000000f00fb536f757263655f30000000000a006a00000000

Receive Command
80010000000053580035000000080001008a0000000f00fb536f757263655f30000000000a006a000000000000000a006b00000000

Receive Command
8001000400005358003f00000008000100130000000f00fb536f757263655f3000000000080050000000000006001100000008004e00000000000800670014

```

then   send  AB command and we disconnect device. below is esp serial output.

```
Receive Command
ab

Device Disconnected
```

when we connect second time and try to send commands second time device receives only 0100 command instead of sequential commands mentioned below.
```
Receive Command
0100
```

```
Future<void> startInventory() async {

[... 279 lines cut ...]
D/[FBP-Android](24940): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](24940): [FBP] onCharacteristicWrite:
D/[FBP-Android](24940): [FBP]   chr: 542eec3a-edd1-11ea-adc1-001999bed156
D/[FBP-Android](24940): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](24940): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](24940): [FBP] onCharacteristicWrite:
D/[FBP-Android](24940): [FBP]   chr: 542eec3a-edd1-11ea-adc1-001999bed156
D/[FBP-Android](24940): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](24940): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](24940): [FBP] onCharacteristicWrite:
D/[FBP-Android](24940): [FBP]   chr: 542eec3a-edd1-11ea-adc1-001999bed156
D/[FBP-Android](24940): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](24940): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](24940): [FBP] onCharacteristicRead:
D/[FBP-Android](24940): [FBP]   chr: 2a67
D/[FBP-Android](24940): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](24940): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](24940): [FBP] onCharacteristicRead:
D/[FBP-Android](24940): [FBP]   chr: 2a67
D/[FBP-Android](24940): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](24940): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](24940): [FBP] onCharacteristicRead:
D/[FBP-Android](24940): [FBP]   chr: 2a67
D/[FBP-Android](24940): [FBP]   status: GATT_SUCCESS (0)
```

chipweinberger, 2024-02-07:
this is a problem in your device firmware probably.

try Example App.

chipweinberger, 2024-02-10:
if you have more details to add, please reopen the issue. 

I think this is a device firmware issue.

closed 2024-02-10 by chipweinberger