chipweinberger/flutter_blue_plus #789: [Help]: Android - Dual Human Interface Device Service (0x1812)

PYJTERSTM32, 2024-02-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

flutter_blue_plus: ^1.31.13

### Flutter Version

Flutter 3.16.9

### What OS?

Android

### OS Version

Android 9

### Bluetooth Module

PG-9083s Red Bat Game Controller

### What is your problem?

I have an Ipega-9083S game controller. I wanted to map the keys and joystick to read individual values to the application I am creating, which is intended to control a remote-controlled model.
However, I encountered a double service problem (0x1812)

I scanned with BLE SCAN and I have something like this:

```
     Status: CONNECTED, NOT BONDED

     DEVICE INFORMATION
         PRIMARY SERVICE
         Manufacturer Name String (UUID: 0x2A29)
         Firmware Revision String (UUID: 0x2A26)
         PnP ID (UUID: 0x2A50)

     GENERIC ACCESS
         PRIMARY SERVICE
         Device Name (UUID: 0x2A00)
         Appearance (UUID: 0x2A01)
         Peripheral Preferred Connection Parameters (UUID: 0x2A04)

     GENERIC ATTRIBUTE
         PRIMARY SERVICE
         Service Changed (UUID: 0x2A05)

     HUMAN INTERFACE DEVICE
         PRIMARY SERVICE
         Protocol Mode (UUID: 0x2A4E)
         Report (UUID: 0x2A4D)
         Report Map (UUID: 0x2A4B)
         HID Information (UUID: 0x2A4A)
         HID Control Point (UUID: 0x2A4C)
```

Here are the details for HUMAN INTERFACE DEVICE (0x1812) PRIMARY SERVICE:

```
     Protocol Mode
         UUID: 00002A4E-0000-1000-8000-00805F9B34FB
         Properties: READ, WRITE NO RESPONSE
Write Type: WRITE REQUEST

     Report
         UUID: 00002A4D-0000-1000-8000-00805F9B34FB
         Properties: READ, NOTIFY
         Descriptors:
             Client Characteristic Configuration
                 UUID: 00002902-0000-1000-8000-00805F9B34FB
             UKNOWN
                 UUID: 00002908-0000-1000-8000-00805F9B34FB
      Report
         UUID: 00002A4D-0000-1000-8000-00805F9B34FB
         Properties: READ, NOTIFY
         Descriptors:
             Client Characteristic Configuration
                 UUID: 00002902-0000-1000-8000-00805F9B34FB
             UKNOWN
                 UUID: 00002908-0000-1000-8000-00805F9B34FB

     Report Map
         UUID: 00002A4B-0000-1000-8000-00805F9B34FB
         Properties: READ

     HID Information
         UUID: 00002A4A-0000-1000-8000-00805F9B34FB
         Properties: READ

     HID Control Point
         UUID: 00002A4C-0000-1000-8000-00805F9B34FB
         Properties: WRITE NO RESPONSE
Write Type: WRITE REQUEST
```

As you can see, the 0x2A4D Report has two characteristics and has duplicate descriptors.
I can't manage to subscribe to and read the RAW stream from the service. I have a double characteristic error with the same ID no matter what I do.

[image]

[image]

[image]

### Logs

```shell
Characteristics subscription error: Exception: Multiple matching characteristics found:
QualifiedCharacteristic(characteristicId: 00002a4d-0000-1000-8000-00805f9b34fb, serviceId: 00001812-0000-1000-8000-00805f9b34fb, deviceId: 03:34:08:20:58:A8)
```

PYJTERSTM32, 2024-02-08:
> QualifiedCharacteristic? That is the name for flutter_reactive_ble.

Sorry, I mixed up the messages, but the repository is correct flutter_blue_plus because I check all repositories. flutter_blue_plus I use it by default to connect to esp32 and communication there works properly. But there is a problem with a gamepad that has two the same UUID. And unfortunately I don't know how to get around or solve this problem.

chipweinberger, 2024-02-10:
post the full logs from FBP. From the very begining.

PYJTERSTM32, 2024-02-11:
Thanks for the answer...
Ok, I used an Android 14 A53 phone, I don't have access to iOS, so I don't know. Maybe I'll be able to test it, and in about 2-3 weeks I'll buy a used iOS phone.
If everything will work on Android.

I apologize right away, I'm not a professional programmer. I do this as a hobby.

Attached is the log from running the Android application.
The application's task is to search for an Ipega gamepad, connect to it, and subscribe to the characteristic that is responsible for returning RAW data from pressing buttons or joysticks.
I just want to return the values from the joysticks and buttons to the screen.

[log.txt](https://github.com/boskokg/flutter_blue_plus/files/14232384/log.txt)

[bluetooth_service.dart.txt](https://github.com/boskokg/flutter_blue_plus/files/14232383/bluetooth_service.dart.txt)

[main.dart.txt](https://github.com/boskokg/flutter_blue_plus/files/14232385/main.dart.txt)

PYJTERSTM32, 2024-02-11:
Ok, I don't have access to iOS yet, but if it works on Android, I'll be happy to buy a phone for such tests.
I took an A53 phone with Android 14.
Attached is the log from running the Android application.
The application's task is to search for an Ipega gamepad, connect to it, and subscribe to the characteristic that is responsible for returning RAW data from pressing buttons or joysticks.
I just want to return the values from the joysticks and buttons to the screen.
[bluetooth_service.dart.txt](https://github.com/boskokg/flutter_blue_plus/files/14232552/bluetooth_service.dart.txt)
[log.txt](https://github.com/boskokg/flutter_blue_plus/files/14232554/log.txt)
[main.dart.txt](https://github.com/boskokg/flutter_blue_plus/files/14232555/main.dart.txt)

I would like to emphasize that I am not a professional programmer, I only do it as a hobby to develop my projects.

Mannifest:
`
<uses-feature android:name="android.hardware.bluetooth_le" android:required="false" />

<uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

<uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />

<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" />`

Thank you in advance for your help and time

chipweinberger, 2024-02-11:
your logs say this 

`Neither user 10482 nor current process has android.permission.BLUETOOTH_PRIVILEGED., java.lang.SecurityException: Need BLUETOOTH PRIVILEGED `

chipweinberger, 2024-02-15:
"in the BLE 0x1812 service is it possible to have multiple 0x2a4d characteristics?

Yes, in the Bluetooth Low Energy (BLE) specification, the Human Interface Device (HID) service, identified by the UUID 0x1812, can indeed have multiple instances of the Report characteristic, which is identified by the UUID 0x2A4D. This design allows a single HID service to support multiple reports, such as keyboard input, mouse input, and other control or data reports, each of which can be defined by its own instance of the Report characteristic. 

This capability is crucial for devices that serve multiple functions or support complex interactions, allowing them to report different types of data to a host device under a single HID service. Each 0x2A4D characteristic instance can be configured with different properties (like Read, Write, Notify) and permissions, and it can be associated with different Report Reference descriptors that define the type and ID of the report, enabling the host to distinguish between them."

FlutterBluePlus does not yet support multiple characteristics with the same uuid.

It is something i'd like to support though. 

I think to support it we will add a new class member to BluetoothCharacteristic: `final int index`

for most characteristics, index will be 0. If there are multiple it represents which one it is.

chipweinberger, 2024-02-20:
bump, please let me know if you fix the `BLUETOOTH_PRIVILEGED` and can get better logs. otherwise ill close this issue in favor of the new one.

PYJTERSTM32, 2024-02-24:
> bump, please let me know if you fix the `BLUETOOTH_PRIVILEGED` and can get better logs. otherwise ill close this issue in favor of the new one.

@chipweinberger thx for your answer. And sorry for the long wait for the answer. I received an answer and there is no chance of connecting to the gamepad in the way I thought.
`
Bluetooth HID devices, such as a gamepad, are handled by the operating system itself. You will not have access to this interface.`

That's why I received an access error.

You can use https://pub.dev/documentation/n_gamepad/latest/ or as I used, native listening in Kotlin and forwarding to flutter.
https://developer.android.com/develop/ui/views/touch-and-input/game-controllers

I would also like to thank you for your involvement. If I can help in any way, I will be happy to do so. If you want me to do some tests, that's fine.

To sum up, I can connect to the gamepad and manage the connection, but I can only listen natively from the system. Actually, it makes sense because it would be possible to eavesdrop on passwords entered from the keyboard ;)

closed 2024-02-25 by chipweinberger