chipweinberger/flutter_blue_plus #792: [Help]: Question: How does service discovery work on iOS?

enseitankad0, 2024-02-13:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.13 

### Flutter Version

3.16.8

### What OS?

iOS

### OS Version

iOS 17

### Bluetooth Module

Norfic NRF52840

### What is your problem?

Hello

I wonder, how this package is doing service rediscovery on iOS?

I'm considering whether switching from [flutter reactive ble ](https://pub.dev/packages/flutter_reactive_ble) will resolve my issues. 
Basically I cannot reconnect to the, after the mcumgr update is performed. This bug happens only on iOS and requires manual bonding removal from BT settings. I suspect, that changes of services and characteristics after the update causes the problem. 

I've checked the source code of FlutterBluePlus and we don't perform any service rediscovery on iOS platfrom.

```
 if (subscribeToServicesChanged) {
      if (Platform.isIOS == false && Platform.isMacOS == false) {
        BluetoothCharacteristic? c = _servicesChangedCharacteristic;
        if (c != null && (c.properties.notify || c.properties.indicate) && c.isNotifying == false) {
          await c.setNotifyValue(true);
        }
      }
```

Question 1. How is it guaranteed that iOS device will detect service changes (I'd like to remind that those services are not even visible in nRF Connect app)

### Logs

```shell
-
```

krupikivan, 2024-02-13:
Im getting this error ONLY on IOS on trying to setNotifyValue as true
`flutter: PlatformException(setNotifyValue, neither NOTIFY nor INDICATE properties are supported by this BLE characteristic, null, null)`
Basically Im not able to subscribe

chipweinberger, 2024-02-13:
iOS subscribes to `servicesChangedCharacteristic` automatically. You cant change this. It happens without our control.

just listen for it and it should work.
```
device.onServicesReset.listen(() async {
    print("Services Reset");
    await device.discoverServices();
});
```

But no, I dont think it is your problem. You have a bonding problem. 

You need to disconnect & rescan IMO.

closed 2024-02-13 by chipweinberger