chipweinberger/flutter_blue_plus #795: [Feature]: Support multiple characteristics with same uuid

chipweinberger, 2024-02-15:
### FlutterBluePlus Version

1.31.14

### Flutter Version

3.18.1

### What OS?

All

### OS Version

All

### Bluetooth Module

All

### What is your feature request?

Relevant Issue: https://github.com/boskokg/flutter_blue_plus/issues/789

**FlutterBluePlus does not yet support multiple characteristics with the same uuid**, which is needed for the 0x1812 HID Service.

We need to add a new class member to `BluetoothCharacteristic`: `final int characteristicIndex`

For most characteristics, index will be 0. But If there are multiple characteristics with the same uuid, it represents if it is the first, second, third, etc.  Android & iOS always keep characteristics in the same order, so we can use an index to find the right one.

---

> in the BLE 0x1812 service is it possible to have multiple 0x2a4d characteristics?
> 
> Yes, in the Bluetooth Low Energy (BLE) specification, the Human Interface Device (HID) service, identified by the UUID 0x1812, can indeed have multiple instances of the Report characteristic, which is identified by the UUID 0x2A4D. This design allows a single HID service to support multiple reports, such as keyboard input, mouse input, and other control or data reports, each of which can be defined by its own instance of the Report characteristic. 
> 
> This capability is crucial for devices that serve multiple functions or support complex interactions, allowing them to report different types of data to a host device under a single HID service. Each 0x2A4D characteristic instance can be configured with different properties (like Read, Write, Notify) and permissions, and it can be associated with different Report Reference descriptors that define the type and ID of the report, enabling the host to distinguish between them.

### Logs

```shell
No Logs
```

PYJTERSTM32, 2024-02-24:
@chipweinberger thx for your answer. And sorry for the long wait for the answer. I received an answer and there is no chance of connecting to the gamepad in the way I thought.
`
Bluetooth HID devices, such as a gamepad, are handled by the operating system itself. You will not have access to this interface.`

That's why I received an access error.

You can use https://pub.dev/documentation/n_gamepad/latest/ or as I used, native listening in Kotlin and forwarding to flutter.
https://developer.android.com/develop/ui/views/touch-and-input/game-controllers

I would also like to thank you for your involvement. If I can help in any way, I will be happy to do so. If you want me to do some tests, that's fine.

vishal788, 2024-02-29:
All data is coming properly but after some time have some error and freez application.

i need some help for this error given below :-

D/[FBP-Android](26989): [FBP] onCharacteristicChanged:
D/[FBP-Android](26989): [FBP]   chr: ffe1

MrCsabaToth, 2024-03-12:
Same UUID? That sounds wild, almost defeats the "unique" part of UUID. I didn't know that such a scenario is even possible and valid: I thought that the service UUID + characteristic UUID is unique tuple.

matt-meades, 2024-07-18:
Would be great to see this resolved! I'm facing this issue with my current implementation, and in response to @MrCsabaToth, as per the Bluetooth spec Vol 3, Part G, Section 3.3.1:

"A service may have multiple characteristic definitions with the same Characteristic UUID"

I'm currently implementing something similar to this [wifi scanner service](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42683-ATWINC3400-BLE-WiFi-Scan-and-Connect-Services-Guide_UserGuide.pdf), which populates the scan results as characteristics.

As a workaround for now I'm going to give each of the scan results the same Characteristic UUID up to the first 120 bits to identify them, then increment the last 8 bits giving me 256 possible options.

chipweinberger, 2024-07-19:
@matt-meades 

feel free to open a well tested PR

nebkat, 2024-07-29:
See previous discussions/PR from `flutter_blue`:

https://github.com/pauldemarco/flutter_blue/issues/213
https://github.com/pauldemarco/flutter_blue/pull/215/files

chipweinberger, 2024-11-07:
copying this here for future reference.

I think we should add an index to Guid.

```
class Guid {
  final List<int> bytes;
  final int index = 0;
```

string representation:

```
if (index != 0) {
   // 0000-0000-1000-8000-00805f9b34fb:$index
}
```

normally it will be set to 0. 

and only in cases where there are dupicate services, or duplicate characteristics in a service, will it have a value.

I think we should try to make it work with an index, first. That seems best to me.

mgeilich, 2025-01-14:
I really need a solution to supporting multiple characteristics with the same UUID. This is supported by native Bluetooth on Android and iOS, but not by this plugin.

mgeilich, 2025-02-20:
Something to watch out for when you get around to this...
It looks like the characteristics are getting returned in reverse order under web, which breaks the idea of a reliable index cross-platform.

chipweinberger, 2025-04-26:
it looks like [bluetooth_low_energy](https://github.com/yanshouwang/bluetooth_low_energy) supports this. so we can copy their approach.

chipweinberger, 2025-05-09:
yes it's a tricky problem.

Maybe should use a hash of it's descriptors

In all the BLE APIS that have duplicate CHR uuids, AFAIK they all instead use **descriptors** to identify them. So we should probably use descriptors too.

i.e. we can hash the data within the descriptor and use that as our identifier. Should be good enough for all reasonable purposes.

mgeilich, 2025-05-10:
Isn't the data within descriptors using the same UUID the same?  You need something unique for the hash...

chipweinberger, 2025-05-10:
no, they each have their own unique descriptors

mgeilich, 2025-05-12:
The descriptors are unique but aren't the descriptor property values the same?  You'd have to hash a pointer to a descriptor to pass across the Method Channel.  

Keeping a list of descriptors on the native side and passing the index across the Method Channel works, but relies on Bluetooth returning the descriptors in a consistent order.  I found the return order differs across OSs, but it does seem to be consistent within a given OS.

chipweinberger, 2025-05-12:
ordering is too error-prone and undocumented.  It may change when the user updates their OS, or for other reasons we don't understand.

> The descriptors are unique but aren't the descriptor property values the same?

AFAIK they'll have something like this

* CHR-1234
  * DESC-5678
    * VALUE: 0
* CHR-1234
  * DESC-5678
    * VALUE: 1

chipweinberger, 2025-05-12:
1. use `getInstanceId()` or similar where possible
2. fallback to desc hash for other platforms

chipweinberger, 2025-05-12:
hmm although one big problem, descriptors can change over time.

Moelfarri, 2025-05-13:
What do you mean that descriptors can change over time? I guess the instanceId just needs to be unique for one discovery session at a time? So next time you discover and the characteristics descriptors are not the same it wouldnt really matter because the descriptor hashes still represent uniqueness of the similar characteristics. 

Or is it something i am missing?

chipweinberger, 2025-05-14:
the descriptor values can change at any time.

chipweinberger, 2025-05-14:
New idea. We maintain our own mappings from:

`Map<ObjectReference, uniqueIncrementingInteger>`
`Map<uniqueIncrementingInteger, ObjectReference>`

When we discover characteristics we add them to the map. 

We use the uniqueIncrementingInteger as the instanceId on other platforms.

Moelfarri, 2025-05-14:
uniqueIncrementingInteger needs to be a singleton object of some sort then. I agree that could work better than hashes if the Descriptor hash has a risk of changing / not being unique enough in 1 session.

chipweinberger, 2025-05-14:
> because there is nothing that says you can't have 10 identical characteristics with no discriminating descriptors.

this is not a practical concern. no API is designed this way. And if there is, I don't care to support it!

nebkat, 2025-05-14:
> no API is designed this way

If I'm not mistaken the [WiFi scanner service](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42683-ATWINC3400-BLE-WiFi-Scan-and-Connect-Services-Guide_UserGuide.pdf) linked earlier in this thread is one such example.

We use characteristics to list the partitions on our device. The characteristic value contains all the info we need about the partitions so there is no need to distinguish between them beyond that.

chipweinberger, 2025-05-14:
> And if there is, I don't care to support it!

:)

MrCsabaToth, 2025-05-15:
> If I'm not mistaken the [WiFi scanner service](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42683-ATWINC3400-BLE-WiFi-Scan-and-Connect-Services-Guide_UserGuide.pdf) linked earlier in this [...]

I have to ask this to be sure: you *willingly* created same GUID characteristics just to communicate multiple values?

I understand if the standard would allow that. But if you have twin kids or triplets would you name all of them exactly the same?

In your case you could list multiple values under the same (single) characteristics via some simple array semantics. Why did you go with ambiguous duplicates?

nebkat, 2025-05-15:
The common misconception in GATT is that a UUID is for uniquely identifying a particular instance of an attribute (service, characteristic or descriptor) when in fact it is to uniquely describe the _contents_ of the attribute. Only the _handle_ can be used to uniquely identify specific instances of attributes.

As the handle is just an incrementing integer for every attribute defined in the GATT table (i.e. it changes every time an attribute is added or removed in the middle of the table) most library implementations hide this detail from the user. The correct way to identify attributes is by proxy: observing the descriptors of a characteristic or the characteristics of a service.

> I understand if the standard would allow that.

The standards don't just allow it - it is effectively the recommended way. See for example the [Automation IO Service](https://www.bluetooth.com/specifications/specs/automation-io-service-1-0/):

```
If a device has more than one instance of the Analog characteristic, each characteristic shall
include a Characteristic Presentation Format descriptor that has a namespace / description
value that is unique for that instance of the Analog characteristic. Description values from 0x0001
and upwards shall be used to uniquely identify each Analog characteristic.
```

You could argue it would be simpler to define 10x sequential UUIDs but now my application needs to store in static memory 160 bytes worth of UUIDs when there is only 1 data type. 

> In your case you could list multiple values under the same (single) characteristics via some simple array semantics. Why did you go with ambiguous duplicates?

- Attributes are limited to 512 bytes, my partition info is ~150 bytes.
- The content of the characteristic makes it unambiguous and allows the entries to be sorted.

chipweinberger, 2025-05-15:
@ekuleshov @tnc1997 

just fyi this PR might be coming

mgeilich, 2025-06-17:
I am unable to build with Master:  Getting errors of this form:

../flutter_blue_plus/packages/flutter_blue_plus/lib/src/bluetooth_characteristic.dart:29:24: Error: The getter 'instanceId' isn't defined for the class 'BmBluetoothCharacteristic'.
 - 'BmBluetoothCharacteristic' is from 'package:flutter_blue_plus_platform_interface/src/bluetooth_msgs.dart' ('../../../.pub-cache/hosted/pub.dev/flutter_blue_plus_platform_interface-4.0.2/lib/src/bluetooth_msgs.dart').
Try correcting the name to the name of an existing getter, or defining a getter or field named 'instanceId'.
        instanceId = p.instanceId;

ekuleshov, 2025-06-17:
> I am unable to build with Master: Getting errors of this form:

See how dependencies are declared in the example app. You may need to do that for a local build until `flutter_blue_plus_platform_interface` is released.

@chipweinberger we need to bump version of platform interface...

mgeilich, 2025-06-17:
That worked and most importantly, I can keep 2 characteristics with the same UUID distinct with this version.  Nice job!!

mgeilich, 2025-06-23:
Actually, more testing shows me that the return order differs across machines, so I'm going to have to query these after discovery at runtime to figure out which is which.

mgeilich, 2025-07-02:
Any thoughts as to when this master branch might be published?

chipweinberger, 2025-09-04:
released