chipweinberger/flutter_blue_plus #797: [Help]: Able to auto-connect (bonding) BLE device when app launched again?

mtiendat, 2024-02-23:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.7

### Flutter Version

 3.16.7

### What OS?

Android

### OS Version

android 13

### Bluetooth Module

nordic nRF5340

### What is your problem?

I see method device.connect(autoconnect: true).  is it auto connect to device when app launched next time?

### Logs

```shell
I see method device.connect(autoconnect: true).  is it auto connect to device when app launched next time?
```

chipweinberger, 2024-02-23:
no.

it resets after app launch

closed 2024-02-23 by chipweinberger

mtiendat, 2024-02-24:
> no.
> 
> it resets after app launch
> 
> also you should update

ok, I was mistake with the autoconnect.
So. In my case,  I have to check 
` final bondedDevices = await FlutterBluePlus.bondedDevices;`
whenever re-launched app. if bondedDevices has my BLE device, I don't need to scan again.

chipweinberger, 2024-02-24:
ya. only works on android though

mtiendat, 2024-02-24:
@chipweinberger  if I don't call createBond() after device.connect() I always got bonded state BluetoothBondState.none. Do you know this issue, why it doesn't automatically bond ?

chipweinberger, 2024-02-24:
could be a device issue. not sure.

why do you need to bond?

mtiendat, 2024-02-24:
@chipweinberger because I don't want to re-scan whenever I launch app, only scan first time launch app and then I want to use bonding when connecting to BLE

chipweinberger, 2024-02-24:
you could try saving the device id in a file

mtiendat, 2024-02-24:
@chipweinberger thank you for your help! save  `RemoteId` to local and retrieve it then connect might be better than use `FlutterBluePlus.bondedDevices`