chipweinberger/flutter_blue_plus #798: [Help]: PlatformException on connect (Android only)

JoshuaWell, 2024-02-27:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.15

### Flutter Version

3.19.1

### What OS?

Android

### OS Version

Multiple versions

### Bluetooth Module

ESP32

### What is your problem?

Encountering a PlatformException with the message "_PlatformException(requestMtu, device is disconnected, null, null)_" when attempting to connect using the connect() function on Android. This issue is not observed on iOS.

```
 Future<void> _connectToDevice(BluetoothDevice device) async {
    if (_device != null) {
      return;
    }

    logger.d("Connecting to device");

    _setState(ConnectionStatus.connecting);
    _device = device;

    _deviceSubscription?.cancel();

    var state = await device.connectionState.first;

    if (state == BluetoothConnectionState.connected) {
      return;
    }

    try {
      logger.d("Trying to connect to device...");

      await device.connect(
          timeout: const Duration(seconds: 5), autoConnect: false);

      _deviceSubscription = device.connectionState.listen(_deviceStateChanged);

      await _discoverCharacteristic(device);
    } catch (e) {
      print(e);
      disconnectReader();
    }
  } 
```

### Logs

```shell
D/[FBP-Android]( 8301): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android]( 8301): [FBP] initializing BluetoothAdapter
I/BluetoothAdapter( 8301): getBluetoothLeScanner
D/BluetoothAdapter( 8301): isLeEnabled(): ON
I/BluetoothAdapter( 8301): getBluetoothLeScanner
D/BluetoothAdapter( 8301): isLeEnabled(): ON
D/BluetoothLeScanner( 8301): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
I/flutter (24019): [D] Trying to connect to device...
D/BluetoothGatt( 8301): connect() - device: F4:12:FA:4F:A0:65, auto: false
D/BluetoothGatt( 8301): registerApp()
D/BluetoothGatt( 8301): registerApp() - UUID=e799208c-61ab-42d1-ba16-faca7427fb6e
D/BluetoothGatt( 8301): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt( 8301): onClientConnectionState() - status=0 clientIf=6 device=F4:12:FA:4F:A0:65
D/BluetoothGatt( 8301): cancelOpen() - device: F4:12:FA:4F:A0:65
D/BluetoothGatt( 8301): onConnectionUpdated() - Device=F4:12:FA:4F:A0:65 interval=40 latency=0 timeout=500 status=0
D/BluetoothGatt( 8301): onClientConnectionState() - status=0 clientIf=6 device=F4:12:FA:4F:A0:65
D/BluetoothGatt( 8301): close()
D/BluetoothGatt( 8301): unregisterApp() - mClientIf=6
I/BluetoothAdapter( 8301): getBluetoothLeScanner
D/BluetoothAdapter( 8301): isLeEnabled(): ON
D/BluetoothLeScanner( 8301): could not find callback wrapper
I/BluetoothAdapter( 8301): getBluetoothLeScanner
D/BluetoothAdapter( 8301): isLeEnabled(): ON
D/BluetoothLeScanner( 8301): onScannerRegistered() - status=6 scannerId=-1 mScannerId=0
I/flutter ( 8301): PlatformException(requestMtu, device is disconnected, null, null)
```

chipweinberger, 2024-02-28:
your logs dont make much sense.

where is `D/[FBP-Android]( 8301): [FBP] onMethodCall: connect` ?

chipweinberger, 2024-02-28:
also, you should update

JoshuaWell, 2024-02-28:
There is no D/[FBP-Android]( 8301): [FBP] onMethodCall: connect in the logs. Unfortunately, using the default timeout did not solve the problem.

chipweinberger, 2024-02-28:
you can set mtu to false in the connect arguments.

chipweinberger, 2024-02-28:
you need to set the log level to debug i think

JoshuaWell, 2024-02-28:
Loglevel debug did the trick to get the full logs. 

Setting mtu to null did not solve the problem either.

```
The Flutter DevTools debugger and profiler on EML L29 is available at: http://127.0.0.1:9102?uri=http://127.0.0.1:57499/SGl9iiy0n6g=/
V/AudioManager( 1026): playSoundEffect   effectType: 0
D/[FBP-Android]( 1026): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android]( 1026): [FBP] initializing BluetoothAdapter
I/BluetoothAdapter( 1026): getBluetoothLeScanner
D/[FBP-Android]( 1026): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android]( 1026): [FBP] connectedPeripherals: 0
D/[FBP-Android]( 1026): [FBP] onMethodCall: setLogLevel
D/[FBP-Android]( 1026): [FBP] onMethodCall: isSupported
D/[FBP-Android]( 1026): [FBP] onMethodCall: isSupported
D/[FBP-Android]( 1026): [FBP] onMethodCall: turnOn
D/[FBP-Android]( 1026): [FBP] onMethodCall: turnOn
D/[FBP-Android]( 1026): [FBP] onMethodCall: getAdapterState
D/[FBP-Android]( 1026): [FBP] onMethodCall: getAdapterState
D/[FBP-Android]( 1026): [FBP] onMethodCall: startScan
I/BluetoothAdapter( 1026): getBluetoothLeScanner
D/BluetoothAdapter( 1026): isLeEnabled(): ON
D/BluetoothLeScanner( 1026): onScannerRegistered() - status=0 scannerId=10 mScannerId=0
D/[FBP-Android]( 1026): [FBP] onMethodCall: stopScan
I/BluetoothAdapter( 1026): getBluetoothLeScanner
D/BluetoothAdapter( 1026): isLeEnabled(): ON
D/[FBP-Android]( 1026): [FBP] onMethodCall: startScan
I/BluetoothAdapter( 1026): getBluetoothLeScanner
D/BluetoothAdapter( 1026): isLeEnabled(): ON
D/BluetoothLeScanner( 1026): onScannerRegistered() - status=0 scannerId=10 mScannerId=0
I/zygote64( 1026): Do full code cache collection, code=125KB, data=97KB
I/zygote64( 1026): After code cache collection, code=112KB, data=66KB
I/zygote64( 1026): Do partial code cache collection, code=121KB, data=77KB
I/zygote64( 1026): After code cache collection, code=121KB, data=77KB
I/zygote64( 1026): Increasing code cache capacity to 512KB
I/flutter ( 1026): [D] Connecting to device
I/flutter ( 1026): [D] Trying to connect to device...
I/flutter ( 1026): Exception: Could not connect to reader
D/[FBP-Android]( 1026): [FBP] onMethodCall: connect
D/BluetoothGatt( 1026): connect() - device: F4:12:FA:4F:9F:FD, auto: false
D/BluetoothGatt( 1026): registerApp()
D/BluetoothGatt( 1026): registerApp() - UUID=c6d00094-6c13-452d-aced-4dfa82ecd76a
D/BluetoothGatt( 1026): onClientRegistered() - status=0 clientIf=11
D/[FBP-Android]( 1026): [FBP] onMethodCall: isSupported
D/[FBP-Android]( 1026): [FBP] onMethodCall: turnOn
D/[FBP-Android]( 1026): [FBP] onMethodCall: stopScan
I/BluetoothAdapter( 1026): getBluetoothLeScanner
D/BluetoothAdapter( 1026): isLeEnabled(): ON
D/[FBP-Android]( 1026): [FBP] onMethodCall: startScan
I/BluetoothAdapter( 1026): getBluetoothLeScanner
D/BluetoothAdapter( 1026): isLeEnabled(): ON
D/BluetoothLeScanner( 1026): onScannerRegistered() - status=0 scannerId=10 mScannerId=0
D/BluetoothGatt( 1026): onClientConnectionState() - status=0 clientIf=11 device=F4:12:FA:4F:9F:FD
D/[FBP-Android]( 1026): [FBP] onConnectionStateChange:connected
D/[FBP-Android]( 1026): [FBP]   status: SUCCESS
D/[FBP-Android]( 1026): [FBP] onMethodCall: disconnect
D/BluetoothGatt( 1026): cancelOpen() - device: F4:12:FA:4F:9F:FD
D/BluetoothGatt( 1026): onClientConnectionState() - status=0 clientIf=11 device=F4:12:FA:4F:9F:FD
D/[FBP-Android]( 1026): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android]( 1026): [FBP]   status: SUCCESS
D/BluetoothGatt( 1026): close()
D/BluetoothGatt( 1026): unregisterApp() - mClientIf=11
D/[FBP-Android]( 1026): [FBP] onMethodCall: discoverServices
I/flutter ( 1026): PlatformException(discoverServices, device is disconnected, null, null)
```

chipweinberger, 2024-02-28:
notice you have this in the logs

[FBP] onMethodCall: disconnect

closed 2024-02-28 by chipweinberger