chipweinberger/flutter_blue_plus #801: [Help]: BLE Data Packets Slows Down On iOS

oooluwa, 2024-02-29:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.15

### Flutter Version

3.19.0

### What OS?

iOS

### OS Version

iPadOS 16.6, 17.3

### Bluetooth Module

Nordic nRF52

### What is your problem?

Our app listens to peripheral devices advertising data over BLE. We noticed that the app was missing some data packets and discovered that the BLE data being received from the flutter_blue_plus package slows down for about 1-3 seconds occasionally. The delay can be up to 5 or more seconds in some cases. 

Here are the other things we have tried:
- We tested on the Android version of our app and observed a smooth transmission of data packets with no delays 
- We used the nRF Connect app to listen to the peripheral device and observed an average latency of 130ms. The data packets did not slow down at any time

Has any other person experienced this issue? How can we solve it?

### Logs

```shell
Each timestamp represents a new data packet. Notice the difference in timestamp after *After turning OFF peripheral and turning it ON again*

flutter: 🐛 BLE Service: 2024-02-29 15:41:30.730064
flutter: 🐛 BLE Service: 2024-02-29 15:41:30.730064
flutter: 🐛 BLE Service: 2024-02-29 15:41:30.730064
flutter: 🐛 BLE Service: 2024-02-29 15:41:30.730064
flutter: 🐛 BLE Service: 2024-02-29 15:41:30.783699
flutter: 🐛 BLE Service: 2024-02-29 15:41:30.905540
flutter: 🐛 BLE Service: 2024-02-29 15:41:31.034623
flutter: 🐛 BLE Service: 2024-02-29 15:41:31.152538
flutter: 🐛 BLE Service: 2024-02-29 15:41:31.152538
flutter: 🐛 BLE Service: 2024-02-29 15:41:31.402218
flutter: 🐛 BLE Service: 2024-02-29 15:41:32.502261
flutter: 🐛 BLE Service: 2024-02-29 15:41:32.621619
flutter: 🐛 BLE Service: 2024-02-29 15:41:33.107828
flutter: 🐛 BLE Service: 2024-02-29 15:41:33.234106
flutter: 🐛 BLE Service: 2024-02-29 15:41:33.356785
flutter: 🐛 BLE Service: 2024-02-29 15:41:33.475597
flutter: 🐛 BLE Service: 2024-02-29 15:41:37.018594
flutter: 🐛 BLE Service: 2024-02-29 15:41:37.139212
flutter: 🐛 BLE Service: 2024-02-29 15:41:41.539490
flutter: 🐛 BLE Service: 2024-02-29 15:41:43.618194
flutter: 🐛 BLE Service: 2024-02-29 15:41:43.985213
flutter: 🐛 BLE Service: 2024-02-29 15:41:44.108227
flutter: 🐛 BLE Service: 2024-02-29 15:41:44.235095
flutter: 🐛 BLE Service: 2024-02-29 15:41:44.356477
flutter: 🐛 BLE Service: 2024-02-29 15:41:44.840517
flutter: 🐛 BLE Service: 2024-02-29 15:41:45.450298
flutter: 🐛 BLE Service: 2024-02-29 15:41:45.704151
[... 167 lines cut ...]
flutter: 🐛 BLE Service: 2024-02-29 15:43:05.342111
flutter: 🐛 BLE Service: 2024-02-29 15:43:05.825200
flutter: 🐛 BLE Service: 2024-02-29 15:43:06.073495
flutter: 🐛 BLE Service: 2024-02-29 15:43:08.144822
flutter: 🐛 BLE Service: 2024-02-29 15:43:08.264352
flutter: 🐛 BLE Service: 2024-02-29 15:43:08.391964
flutter: 🐛 BLE Service: 2024-02-29 15:43:08.576204

After turning OFF peripheral and turning it ON again

flutter: 🐛 BLE Service: 2024-02-29 15:43:08.576204
flutter: 🐛 BLE Service: 2024-02-29 15:43:08.576204
flutter: 🐛 BLE Service: 2024-02-29 15:44:06.158156
flutter: 🐛 BLE Service: 2024-02-29 15:44:06.766008
flutter: 🐛 BLE Service: 2024-02-29 15:44:07.379573
flutter: 🐛 BLE Service: 2024-02-29 15:44:10.058008
flutter: 🐛 BLE Service: 2024-02-29 15:44:10.664066
flutter: 🐛 BLE Service: 2024-02-29 15:44:21.168758
flutter: 🐛 BLE Service: 2024-02-29 15:44:26.540021
flutter: 🐛 BLE Service: 2024-02-29 15:44:30.434947
flutter: 🐛 BLE Service: 2024-02-29 15:44:31.664129
flutter: 🐛 BLE Service: 2024-02-29 15:44:34.340325
flutter: 🐛 BLE Service: 2024-02-29 15:44:34.954245
flutter: 🐛 BLE Service: 2024-02-29 15:44:34.954245
```

chipweinberger, 2024-02-29:
are you using `setNotifyValue`? or are you manually calling `read`?

oooluwa, 2024-02-29:
> how are you reading the data? are you using `setNotifyValue`? or are you manually calling `read`?

Not sure I understand that. This is the function that listens to the devices:

```
Stream<BleDeviceModel> get advertisingDevices async* {
    try {
      await for (List<ScanResult> results in FlutterBluePlus.scanResults) {
        for (ScanResult result in results) {
          if (result.advertisementData.manufacturerData.isEmpty) {
            continue;
          }

          var mfgId = result.advertisementData.manufacturerData.entries.first.key;
          if (mfgId == _kOurId) {
            if (_minRssiThreshold == null || result.rssi > _minRssiThreshold!) {
              _logger.d("BLE Service: ${result.timeStamp}");
              yield result;
            }
          }
        }
      }
    } on Exception catch (e) {
      _logger.e(e.toString());
      throw AppException(e.toString());
    }
  }
```

chipweinberger, 2024-02-29:
I see.

scan results come straight from the operating system 

i'm not aware of anything FBP could change to improve your situation

chipweinberger, 2024-03-05:
if you figure it out, please report back

it could be a problem with your device. 

try different ios version also if possible

closed 2024-03-05 by chipweinberger