chipweinberger/flutter_blue_plus #802: [Help]:  Slow Connection new version 1.31.15

miguelangelflores1993, 2024-03-01:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.15

### Flutter Version

3.19.0

### What OS?

Android

### OS Version

ANDROID 14

### Bluetooth Module

android

### What is your problem?

I previously had version 1.31.5
and connects to a device in less than 10 seconds.

Now in the version
1.31.15

It takes more than 25 seconds onwards.

Because? Can the speed be improved?

### Logs

```shell
[√] Flutter (Channel stable, 3.19.1, on Microsoft Windows [Version 10.0.22631.3155], locale en-US)
[√] Windows Version (Installed version of Windows is version 10 or higher)
[√] Android toolchain - develop for Android devices (Android SDK version 34.0.0)
[√] Chrome - develop for the web
[√] Android Studio (version 2023.1)
[√] VS Code (version 1.87.0)
[√] Connected device (2 available)
[√] Network resources
```

chipweinberger, 2024-03-01:
we didn't change anything to affect connection speed

faizannazir, 2024-03-01:
> we didn't change anything to affect connection speed

```

You are applying Flutter's app_plugin_loader Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/go/flutter-gradle-plugin-apply

FAILURE: Build failed with an exception.

* Where:
Build file 'E:\downloads\flutter_blue_plus\example\android\app\build.gradle' line: 3

* What went wrong:
Plugin [id: 'kotlin-android'] was not found in any of the following sources:

- Gradle Core Plugins (not a core plugin, please see https://docs.gradle.org/8.0/userguide/plugin_reference.html for available core plugins)
- Plugin Repositories (plugin dependency must include a version number for this source)

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 5s
Running Gradle task 'assembleDebug'...                              6.2s
Error: Gradle task assembleDebug failed with exit code 1
```

closed 2024-03-05 by chipweinberger