chipweinberger/flutter_blue_plus #803: [Help]: Unable to build Example app

faizannazir, 2024-03-01:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.15

### Flutter Version

3.19.2

### What OS?

Android

### OS Version

10

### Bluetooth Module

esp32

### What is your problem?

Not able to run example app

### Logs

```shell
lib/utils/extra.dart:7:8: Error: Error when reading 'lib/utils/device_screen.dart': The system cannot find the file specified.

import 'device_screen.dart';
       ^
lib/screens/scan_screen.dart:80:12: Error: The method 'connectAndUpdateStream' isn't defined for the class 'BluetoothDevice'.
 - 'BluetoothDevice' is from 'package:flutter_blue_plus/flutter_blue_plus.dart' ('/C:/Users/hp%20840%20G5/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_blue_plus-1.31.15/lib/flutter_blue_plus.dart').        
Try correcting the name to the name of an existing method, or defining a method named 'connectAndUpdateStream'.
    device.connectAndUpdateStream().catchError((e) {
           ^^^^^^^^^^^^^^^^^^^^^^
lib/screens/device_screen.dart:59:47: Error: The getter 'isConnecting' isn't defined for the class 'BluetoothDevice'.
 - 'BluetoothDevice' is from 'package:flutter_blue_plus/flutter_blue_plus.dart' ('/C:/Users/hp%20840%20G5/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_blue_plus-1.31.15/lib/flutter_blue_plus.dart').        
Try correcting the name to the name of an existing getter, or defining a getter or field named 'isConnecting'.
    _isConnectingSubscription = widget.device.isConnecting.listen((value) {
                                              ^^^^^^^^^^^^
lib/screens/device_screen.dart:66:50: Error: The getter 'isDisconnecting' isn't defined for the class 'BluetoothDevice'.
 - 'BluetoothDevice' is from 'package:flutter_blue_plus/flutter_blue_plus.dart' ('/C:/Users/hp%20840%20G5/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_blue_plus-1.31.15/lib/flutter_blue_plus.dart').        
Try correcting the name to the name of an existing getter, or defining a getter or field named 'isDisconnecting'.
    _isDisconnectingSubscription = widget.device.isDisconnecting.listen((value) {
                                                 ^^^^^^^^^^^^^^^
lib/screens/device_screen.dart:89:27: Error: The method 'connectAndUpdateStream' isn't defined for the class 'BluetoothDevice'.
 - 'BluetoothDevice' is from 'package:flutter_blue_plus/flutter_blue_plus.dart' ('/C:/Users/hp%20840%20G5/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_blue_plus-1.31.15/lib/flutter_blue_plus.dart').        
Try correcting the name to the name of an existing method, or defining a method named 'connectAndUpdateStream'.
      await widget.device.connectAndUpdateStream();
                          ^^^^^^^^^^^^^^^^^^^^^^
lib/screens/device_screen.dart:102:27: Error: The method 'disconnectAndUpdateStream' isn't defined for the class 'BluetoothDevice'.
 - 'BluetoothDevice' is from 'package:flutter_blue_plus/flutter_blue_plus.dart' ('/C:/Users/hp%20840%20G5/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_blue_plus-1.31.15/lib/flutter_blue_plus.dart').        
Try correcting the name to the name of an existing method, or defining a method named 'disconnectAndUpdateStream'.
      await widget.device.disconnectAndUpdateStream(queue: false);
                          ^^^^^^^^^^^^^^^^^^^^^^^^^
lib/screens/device_screen.dart:111:27: Error: The method 'disconnectAndUpdateStream' isn't defined for the class 'BluetoothDevice'.
 - 'BluetoothDevice' is from 'package:flutter_blue_plus/flutter_blue_plus.dart' ('/C:/Users/hp%20840%20G5/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_blue_plus-1.31.15/lib/flutter_blue_plus.dart').        
Try correcting the name to the name of an existing method, or defining a method named 'disconnectAndUpdateStream'.
      await widget.device.disconnectAndUpdateStream();
                          ^^^^^^^^^^^^^^^^^^^^^^^^^
lib/utils/extra.dart:80:12: Error: The method 'connectAndUpdateStream' isn't defined for the class 'BluetoothDevice'.
[... 63 lines cut ...]
                          ^^^^^^^^^^^^^^^^^^^^^^^^^
lib/utils/extra.dart:80:12: Error: The method 'connectAndUpdateStream' isn't defined for the class 'BluetoothDevice'.
 - 'BluetoothDevice' is from 'package:flutter_blue_plus/flutter_blue_plus.dart' ('/C:/Users/hp%20840%20G5/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_blue_plus-1.31.15/lib/flutter_blue_plus.dart').
Try correcting the name to the name of an existing method, or defining a method named 'connectAndUpdateStream'.
    device.connectAndUpdateStream().catchError((e) {
           ^^^^^^^^^^^^^^^^^^^^^^
Target kernel_snapshot failed: Exception

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileFlutterBuildDebug'.
> Process 'command 'E:\downloads\flutter_windows_3.19.1-stable\flutter\bin\flutter.bat'' finished with non-zero exit value 1

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 8s
Running Gradle task 'assembleDebug'...                              9.2s
Error: Gradle task assembleDebug failed with exit code 1
```

chipweinberger, 2024-03-01:
never seen this problem before. try `flutter clean`

faizannazir, 2024-03-01:
> never seen this problem before. try `flutter clean`

getting same error

faizannazir, 2024-03-02:
@chipweinberger 
Finally I am able to build it. I want toa add file uplaod ota update  to update firmware can you guide me how to integrate it 

[joshuadutton](https://github.com/joshuadutton)
provided code 

`import 'dart:async';
import 'dart:io';

import 'package:file_picker/file_picker.dart';
import 'package:flutter/foundation.dart';
import 'package:flutter_blue_plus/flutter_blue_plus.dart';
import 'package:http/http.dart' as http;

const int mtuOffsetForChunkSize = 3;
const int otaControlRequest = 0x01;
const int otaControlRequestAck = 0x02;
const int otaControlRequestNak = 0x03;
const int otaControlDone = 0x04;
const int otaControlDoneAck = 0x05;
const int otaControlDoneNak = 0x06;
const int otaControlRestart = 0x07;
const int otaControlRestartAck = 0x08;

class Logger {
  static void debug(Object? object) {
    if (kDebugMode) {
      // ignore: avoid_print
      print("$object");
    }
  }

  static void error(Object? object) {
    // ignore: avoid_print
    print('\x1B[31m$object\x1B[0m');
  }
}

Future<Uint8List> firmwareBinaryFromFile(String filePath) async {
  return File(filePath).readAsBytes();
}

Future<Uint8List> firmwareBinaryFromUrl(String url) async {
  try {
    final response = await http.get(Uri.parse(url)).timeout(const Duration(seconds: 10));
    if (response.statusCode == 200) {
      return response.bodyBytes;
    } else {
      throw 'HTTP Error: ${response.statusCode} - ${response.reasonPhrase}';
    }
  } catch (e) {
    // Handle other errors (e.g., timeout, network connectivity issues)
    throw 'Error fetching firmware from URL: $e';
  }
}

Future<Uint8List> firmwareBinaryFromPicker() async {
  final result = await FilePicker.platform.pickFiles(
    type: FileType.custom,
    allowedExtensions: ['bin'],
  );

  if (result == null || result.files.isEmpty) {
    throw 'Error getting firmware data: No file picked';
  }

  try {
    return firmwareBinaryFromFile(result.files.first.path!);
  } catch (e) {
    throw 'Error getting firmware data: $e';
  }
}

Future<bool> restartEsp32(BluetoothDevice device, BluetoothCharacteristic controlCharacteristic) async {
  await controlCharacteristic.write([otaControlRestart]);
  List<int> value = await controlCharacteristic.read().timeout(const Duration(seconds: 10));
  Logger.debug('value returned is this ------- $value');
  if (value[0] != otaControlRestartAck) {
    throw 'Restart Command Failed';
  }
  return true;
}

Stream<double> updateEsp32Firmware(BluetoothDevice device, BluetoothCharacteristic dataCharacteristic,
    BluetoothCharacteristic controlCharacteristic, Uint8List firmwareBinary) async* {
  bool updateIsDone = false;
  try {
    int mtuSize = await device.mtu.first;
    int chunkSize = mtuSize - mtuOffsetForChunkSize;

    // Prepare a byte list to write MTU size to controlCharacteristic
    Uint8List byteList = Uint8List(2);
    byteList[0] = chunkSize & 0xFF;
    byteList[1] = (chunkSize >> 8) & 0xFF;
    // write mtuSize to dataCharacteristic
    Logger.debug("Sending chunk size $chunkSize $byteList");
    await dataCharacteristic.write(byteList);

    await controlCharacteristic.write([otaControlRequest]);
    List<int> value = await controlCharacteristic.read().timeout(const Duration(seconds: 10));
    Logger.debug('value returned is this ------- ${value[0]}');
    if (value[0] != otaControlRequestAck) {
      throw 'Start Command Failed';
    }

    int packetNumber = 1;
    int totalPackets = (firmwareBinary.length / chunkSize).ceil();
    for (int i = 0; i < firmwareBinary.length; i += chunkSize) {
      int end = i + chunkSize;
      if (end > firmwareBinary.length) {
        end = firmwareBinary.length;
      }
      final chunk = firmwareBinary.sublist(i, end);
      double progress = (packetNumber / totalPackets);
      String roundedProgressPercentage = (progress * 100).toStringAsPrecision(3);
      if (progress >= 1.0) {
        // don't go to 100 until finish command has been acknowledged
        progress = 0.99;
      }
      Logger.debug('Writing packet $packetNumber of $totalPackets ($roundedProgressPercentage%).');

      await dataCharacteristic.write(chunk);

      yield progress;

      packetNumber++;
    }

    updateIsDone = true;

    await controlCharacteristic.write([otaControlDone]);

    // Check if controlCharacteristic reads 0x05, indicating OTA update finished
    value = await controlCharacteristic.read().timeout(const Duration(seconds: 10));
    Logger.debug('value returned is this ------- ${value[0]}');
    if (value[0] == otaControlDoneAck) {
      Logger.debug('OTA update finished');
      yield 1.0;
    } else {
      Logger.debug('OTA update failed');
      throw 'Finish Command Failed';
    }
  } catch (error) {
    Logger.error(error);
    if (updateIsDone && error.toString().contains("disconnected")) {
      yield 1.0;
    } else {
      yield* Stream.error(error);
    }
  }
}`

chipweinberger, 2024-03-03:
glad you're able to build it

closed 2024-03-03 by chipweinberger