chipweinberger/flutter_blue_plus #805: [Help]: iOS writing characteristic sometimes slow, when sending multiple packets

njn-horn, 2024-03-04:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.15

### Flutter Version

3.16.9

### What OS?

iOS

### OS Version

17.3.1

### Bluetooth Module

nordic nRF52805

### What is your problem?

Hello, I am new to flutter and do not understand this behaviour.
Doing it like it is in the example app works, sending one thing every time a button is pressed.
But I need to initialise some data with the device, which means I need to send some packets to the characteristic.
It works well one or two times and then the call of await characteristic.write can take up to 5 seconds.
Since I need to wait for an answer from the BLE device every time something was sent, to know if the data was good or not, this slows down the process of initialising the device a lot.
This works very well on Android, it just flies through sending the packets.
On iOS if I do not wait for a response from the BLE device it goes almost as fast through sending all the packets aswell.
I have checked that it is not the waiting function that slows down the process. Debugging it reveals that await characteristic.write takes the time.

The function that sends the data:
```
Future<int> sendDataFrames(String data, int waitTime, String answerText, ComResCode resultCode, {bool noWait = false}) async {
    rxBuffer = ""; // global buffer in provider class, empty before receiving new data
    resultCode.value = 0;
    try {
      List<int> list = data.codeUnits;
      Uint8List bytes = Uint8List.fromList(list);
      Toolbox.consolePrint("BEFORE SENDING: ${DateTime.now().toString().substring(11,22)}");
      await uartCharacteristicRX.write(bytes, withoutResponse: uartCharacteristicRX.properties.writeWithoutResponse);
      Toolbox.consolePrint("AFTER SENDING:  ${DateTime.now().toString().substring(11,22)}");
      if (uartCharacteristicRX.properties.read) {
        await uartCharacteristicRX.read();
      }
    } catch (e) {
      resultCode.value = 998; // Error Exception
      return resultCode.value;
    }
 
    // wait if necessary
    if (!noWait) {
      await waitForAnswerOrTimeout(answerText, waitTime, resultCode, pollInterval: const Duration(milliseconds: 1000));
    }
    
    return resultCode.value;
  }
  ```

Any pointer or help would be greatly appreciated.

### Logs

```shell
flutter: SERVICE READY: true
flutter: CHECK CHECK: --- doInit end
flutter: CHECK CHECK: --- connectionState changed to BluetoothConnectionState.connected
flutter: Sending packet
flutter: BEFORE SENDING: 11:56:44.18
flutter: AFTER SENDING:  11:56:44.21
flutter: ----------NEW DATA IN rxBuffer !!!! ---------
flutter: CCK -- WAIT FOR ANSWER: true, false, 0
flutter: Sending packet
flutter: BEFORE SENDING: 11:56:45.22
flutter: AFTER SENDING:  11:56:47.37
flutter: ----------NEW DATA IN rxBuffer !!!! ---------
flutter: CCK -- WAIT FOR ANSWER: true, false, 0
flutter: Sending packet
flutter: BEFORE SENDING: 11:56:48.38
flutter: AFTER SENDING:  11:56:49.47
flutter: ----------NEW DATA IN rxBuffer !!!! ---------
flutter: CCK -- WAIT FOR ANSWER: true, false, 0
flutter: Sending packet
flutter: BEFORE SENDING: 11:56:50.48
flutter: AFTER SENDING:  11:56:52.62
flutter: ----------NEW DATA IN rxBuffer !!!! ---------
flutter: CCK -- WAIT FOR ANSWER: true, false, 0
flutter: Sending packet
flutter: BEFORE SENDING: 11:56:53.62
flutter: AFTER SENDING:  11:56:55.79
flutter: ----------NEW DATA IN rxBuffer !!!! ---------
flutter: CCK -- WAIT FOR ANSWER: true, false, 0
flutter: Sending packet
flutter: BEFORE SENDING: 11:56:56.80
flutter: AFTER SENDING:  11:56:58.94
flutter: ----------NEW DATA IN rxBuffer !!!! ---------
flutter: CCK -- WAIT FOR ANSWER: true, false, 0
flutter: Sending packet
flutter: BEFORE SENDING: 11:56:59.94
flutter: AFTER SENDING:  11:57:02.11
flutter: ----------NEW DATA IN rxBuffer !!!! ---------
flutter: CCK -- WAIT FOR ANSWER: true, false, 0
flutter: BEFORE SENDING: 11:57:03.11
flutter: AFTER SENDING:  11:57:05.28
flutter: ----------NEW DATA IN rxBuffer !!!! ---------
```

njn-horn, 2024-03-04:
After a bit more debugging, also inside the flutter blue plus package, I found that the wait time occurs at the await mtx.take() call in the write method of the BluetoothCharacteristic.

I timed the write method aswell and put prints around a few calls.
This is what happened:
```
flutter: Sending packet
flutter: BEFORE SENDING: 16:28:02.28
flutter: START OF CHAR WRITE 16:28:02.28
flutter: START OF AWAIT MUTEX 16:28:02.28
flutter: END OF AWAIT MUTEX 16:28:06.28
flutter: START OF INVOKE METHOD 16:28:06.29
flutter: END OF INVOKE METHOD 16:28:06.29
flutter: AFTER SENDING:  16:28:06.29
```

chipweinberger, 2024-03-05:
ble is very slow 

the mutex is waiting for the previous operation to finish

in theory you can try removing the mutex on iOS

but on android the mutex is required to prevent errors

njn-horn, 2024-03-05:
Hello, thank you for the response. Do you have any idea why the mutex is way faster on android than on iOS?
disabling the mutex part in the characteristic.write method results in iOS being as fast if not a little faster than android.
I made sure to only be able to call the write of that characteristic once at a time. But this seems like a workaround more than a proper fix. Especially since the mutex works fine on android.

I would like to understand this and learn more about it.

chipweinberger, 2024-03-05:
```
import 'dart:async';

// allows awaiting until at least 1 object is in the queue, at which
// point a single consumer is released
class AwaitableQueue<T> {
  final List<T> _list = [];
  final List<Completer<T>> _waiters = [];

  Object? errorObj;

  void add(T item) {
    if (_waiters.isNotEmpty) {
      // pop the oldest waiter
      var waiter = _waiters.removeAt(0);
      waiter.complete(item);
    } else {
      // no waiters? just add to the list
      _list.add(item);
    }
  }

  void cancelWithError(Object error) {
    this.errorObj = errorObj;
    while (_waiters.isNotEmpty) {
      var waiter = _waiters.removeAt(0);
      waiter.completeError(error);
    }
  }

  void clear() {
    _list.clear();
  }

  Future<T> consume() {
    if (errorObj != null) {
      throw Exception(this.errorObj);
    }
    if (_list.isNotEmpty) {
      // can consume immediately
      return Future.value(_list.removeAt(0));
    } else {
      // add ourself to the list of waiters
      var completer = Completer<T>();
      _waiters.add(completer);
      return completer.future;
    }
  }
}

```

njn-horn, 2024-03-05:
Hello, thank you for this answer! I will look into the AwaitableQueue.

chipweinberger, 2024-03-05:
sure. good luck!

closed 2024-03-05 by chipweinberger