chipweinberger/flutter_blue_plus #806: [Help]: MTU not updated on reconnect

rgootjes, 2024-03-05:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.15

### Flutter Version

3.19.0

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

Unknown module

### What is your problem?

I am listening to MTU changes with ```widget.device.mtu.listen((value) {
      _mtuSize = value;
    });```

I'm then connecting to the device with autoConnect enabled ```device.connect(autoConnect: true, mtu: null)``` and I get an MTU of 23 which is expected.

I request a new MTU and the listener updates with 203.

If I disconnect the device (just by turning it off) and then reconnect it, the listener doesn't update and will leave my local ```_mtuSize``` at 203. However the MTU of the device has reverted back to 23, confirmed by ```device.mtuNow```

Not sure if this is the expected behavior or not. It's easy enough for me to fix by re-requesting a new MTU but figured I'd raise the issue anyway

### Logs

```shell
Restarted application in 1,267ms.
D/[FBP-Android](29034): [FBP] onMethodCall: flutterHotRestart
D/BluetoothAdapter(29034): getBleEnabledArray(): ON
D/[FBP-Android](29034): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](29034): [FBP] calling disconnect: DC:0D:30:E9:6F:CE
D/BluetoothGatt(29034): cancelOpen() - device: DC:0D:30:E9:6F:CE
I/flutter (29034): [FBP] <setLogLevel> args: 5
D/[FBP-Android](29034): [FBP] calling close: DC:0D:30:E9:6F:CE
D/BluetoothGatt(29034): close()
D/BluetoothGatt(29034): onClientConnectionState() - status=0 clientIf=5 device=XX:XX:XX:XX:6F:CE
D/BluetoothGatt(29034): unregisterApp() - mClientIf=5
D/[FBP-Android](29034): [FBP] connectedPeripherals: 0
D/[FBP-Android](29034): [FBP] onMethodCall: setLogLevel
I/flutter (29034): [FBP] <setLogLevel> result: true
I/flutter (29034): [FBP] <getAdapterState> args: null
D/[FBP-Android](29034): [FBP] onMethodCall: getAdapterState
I/flutter (29034): [FBP] <getAdapterState> result: {adapter_state: 4}
I/ViewRootImpl@7dd233[MainActivity](29034): ViewPostIme pointer 0
I/ViewRootImpl@7dd233[MainActivity](29034): ViewPostIme pointer 1
I/flutter (29034): [FBP] <getSystemDevices> args: null
D/[FBP-Android](29034): [FBP] onMethodCall: getSystemDevices
I/flutter (29034): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (29034): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android](29034): [FBP] onMethodCall: startScan
2
D/BluetoothAdapter(29034): getBleEnabledArray(): ON
D/BluetoothAdapter(29034): semIsBleEnabled(): ON
D/BluetoothAdapter(29034): getBleEnabledArray(): ON
D/BluetoothLeScanner(29034): Start Scan with callback
D/BluetoothLeScanner(29034): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
I/flutter (29034): [FBP] <startScan> result: true
D/[FBP-Android](29034): [FBP] onScanResult
I/flutter (29034): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -90, manufacturer_data: {6: 010920020941723621ff91d8c90b217b5041f9d21599e900fffd46}, remote_id: 27:78:E7:F1:D8:E0, service_data: {}}]}
4
D/[FBP-Android](29034): [FBP] onScanResult
I/flutter (29034): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -65, manufacturer_data: {}, remote_id: 7F:EF:B0:E4:49:B0, service_data: {fef3: 4a172341314d3911328e3047a120312725b0852df74a4f8071a14e}, service_uuids: [fef3]}]}
I/flutter (29034): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -31, connectable: 1, manufacturer_data: {}, remote_id: DC:0D:30:E9:6F:CE, platform_name: Printer001, tx_power_level: 4, service_data: {}, adv_name: Printer001, service_uuids: [18f0, e7810a71-73ae-499d-8c15-faa9aef0c3f2]}]}
I/flutter (29034): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -49, manufacturer_data: {6: 01092022ca92244e238c84b887e36b085a18abefb96086cf6186ea}, remote_id: 00:CF:08:17:E4:44, service_data: {}}]}
59
D/[FBP-Android](29034): [FBP] onScanResult
I/ViewRootImpl@7dd233[MainActivity](29034): ViewPostIme pointer 0
2
D/[FBP-Android](29034): [FBP] onScanResult
I/ViewRootImpl@7dd233[MainActivity](29034): ViewPostIme pointer 1
D/[FBP-Android](29034): [FBP] onScanResult
```

chipweinberger, 2024-03-05:
ah, yes this is a bug / you should always assume the mtu is 23 after connection.

we should probably re-push to the stream after connection, since users would expect that.

feel free to open a PR.

chipweinberger, 2024-03-06:
fixed on master branch. please try it!

closed 2024-03-06 by chipweinberger