chipweinberger/flutter_blue_plus #807: [Help]: startScan doesn't start with removeIfGone

kent-williams, 2024-03-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.16

### Flutter Version

3.19.3

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

Nordic nRF52840

### What is your problem?

startScan works as expected when not using `removeIfGone`, and doesn't start when including it.

```
FlutterBluePlus.startScan(
      timeout: const Duration(seconds: 2),
      removeIfGone: const Duration(seconds: 1)
    );
```

### Logs

```shell
Working
=====================
                                            
D/[FBP-Android]( 9574): [FBP] onMethodCall: setLogLevel
I/flutter ( 9574): [FBP] <setLogLevel> result: true
I/flutter ( 9574): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android]( 9574): [FBP] onMethodCall: startScan
D/BluetoothAdapter( 9574): isLeEnabled(): ON
D/BluetoothLeScanner( 9574): onScannerRegistered() - status=0 scannerId=3 mScannerId=0
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
I/flutter ( 9574): [FBP] <startScan> result: true
I/flutter ( 9574): [FBP] <stopScan> args: null
I/flutter ( 9574): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -40, connectable: 1, manufacturer_data: {}, remote_id: F9:E4:C6:0F:13:49, platform_name: Device 841B, tx_power_level: 4, service_data: {}, adv_name: Device 841B}]}
I/flutter ( 9574): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -82, connectable: 1, manufacturer_data: {47940: 3b070ce702794c67b7}, remote_id: 44:BB:3B:07:0C:E7, platform_name: Nest Cam, service_data: {}, adv_name: Nest Cam, service_uuids: [d2d3f8ef-9c99-4d9c-a2b3-91c85d44326c]}]}
I/flutter ( 9574): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -61, manufacturer_data: {76: 0908139ac0a801701b581307023678491da8af}, remote_id: 6C:7F:8C:BD:8E:D2, service_data: {}}]}
I/flutter ( 9574): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -43, connectable: 1, manufacturer_data: {}, remote_id: C8:FF:77:9D:6E:0B, platform_name: YR2-US-MEA0044A, tx_power_level: 5, service_data: {}, adv_name: YR2-US-MEA0044A, service_uuids: [2dd10010-1c37-452d-8979-d1b4a787d0a4]}]}       
D/[FBP-Android]( 9574): [FBP] onMethodCall: stopScan
D/BluetoothAdapter( 9574): isLeEnabled(): ON
I/flutter ( 9574): [FBP] <stopScan> result: true
I/flutter ( 9574): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android]( 9574): [FBP] onMethodCall: startScan
D/BluetoothAdapter( 9574): isLeEnabled(): ON
D/BluetoothLeScanner( 9574): onScannerRegistered() - status=0 scannerId=3 mScannerId=0
[... 218 lines cut ...]
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
I/flutter ( 9574): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -79, manufacturer_data: {76: 12020002}, remote_id: CD:72:16:AB:D2:3B, service_data: {}}]}
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
I/flutter ( 9574): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -68, connectable: 1, manufacturer_data: {}, remote_id: DB:48:9B:E5:1F:50, platform_name: N02MP, service_data: {feaf: 100100025a230900cdfcf3000030b41800}, adv_name: N02MP, service_uuids: [feaf]}]}
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
D/[FBP-Android]( 9574): [FBP] onScanResult
I/flutter ( 9574): [FBP] <stopScan> args: null
D/[FBP-Android]( 9574): [FBP] onMethodCall: stopScan
D/BluetoothAdapter( 9574): isLeEnabled(): ON
I/flutter ( 9574): [FBP] <stopScan> result: true
====================
Not Working
I/flutter ( 9574): [FBP] <setLogLevel> args: 5
D/[FBP-Android]( 9574): [FBP] onMethodCall: setLogLevel
Performing hot reload...
Reloaded 0 libraries in 389ms (compile: 16 ms, reload: 0 ms, reassemble: 230 ms).
I/flutter ( 9574): [FBP] <setLogLevel> result: true
D/TrafficStats( 9574): tagSocket(193) with statsTag=0xffffffff, statsUid=-1
D/TrafficStats( 9574): tagSocket(177) with statsTag=0xffffffff, statsUid=-1
```

chipweinberger, 2024-03-08:
what is the first version that broke it?

kent-williams, 2024-03-08:
I searched the releases for `removeIfGone` and found the breaking change in 1.27.0. I have enabled continuousUpdates, and all is working as expected now. 

Thanks!!