chipweinberger/flutter_blue_plus #809: [Help]: Issue with Writing to BLE Device Using FlutterBlue: No Errors but No Write Operations Performed

simplayy, 2024-03-09:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.16 

### Flutter Version

3.16.9

### What OS?

iOS

### OS Version

iOS 17.4

### Bluetooth Module

Nordic nRF5340

### What is your problem?

I am attempting to write a Key (PSK) to a BLE device using the FlutterBluePLus library. 
My code successfully connects to the device, discovers its services and characteristics, and attempts to write to a specific characteristic. 
Despite receiving no errors during the process, the PSK does not seem to be written to the device.

The function executes without throwing any errors, and it reaches the point where it should write the PSK to the characteristic. However, the device does not receive PSK. 
No errors are logged during the process.
```
Future<bool> writePskToDeviceAndDisconnect(
    BluetoothDevice device, String psk) async {
  List<BluetoothService> services = await device.discoverServices();
  bool serviceFound = false;
  for (BluetoothService service in services) {
    if (service.uuid.toString() == "00001553-1212-efde-1523-785feabcd123") {
      var characteristics = service.characteristics;
      for (BluetoothCharacteristic characteristic in characteristics) {
        if (characteristic.uuid.toString() ==
            "00001555-1212-efde-1523-785feabcd123") {
          print("Writing PSK to characteristic");
          print(psk);
          List<int> pskBytes = utf8.encode(psk); // Use the PSK parameter instead of a hardcoded string
          try {
            await characteristic.write(pskBytes, withoutResponse: true);
          } catch (e) {
            print("Error: " + e.toString());
            return false;
          } finally {
            await device.disconnect();
            return true;
          }
        }
      }
    }
  }
  return false;
}
```

### Logs

```shell
flutter: Device connected: BluetoothDevice{remoteId: A26728AF-54FC-19A6-FD4A-38BB6A3EAFC4, platformName: bs02, services: null}
flutter: BS02 device connected: BluetoothDevice{remoteId: A26728AF-54FC-19A6-FD4A-38BB6A3EAFC4, platformName: bs02, services: null}
flutter: Writing PSK to characteristic
flutter: 76JPFCX9LH8NVYC6
```

chipweinberger, 2024-03-11:
doesnt seem like the actual cause imo

chipweinberger, 2024-03-19:
closing since there doesnt seem to be anything we can do. please let me know if you came to some other conclusion.

closed 2024-03-19 by chipweinberger