chipweinberger/flutter_blue_plus #810: [Help]: [IOS] A Bluetooth connection request pops up every second, which is not happend on Android.

longqian1235, 2024-03-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.15

### Flutter Version

3.16.5

### What OS?

iOS

### OS Version

IOS 15.7(19H12)

### Bluetooth Module

unkown

### What is your problem?

When the iPhone is connected to a device, a Bluetooth connection request pops up every second. After one second, it automatically closes, and then it pops up again in a continuous loop. Additionally, each time, the displayed number changes. Both clicking "Cancel" and "Pair" have no response, but the actual device is already connected. This constant pop-up hinders user interaction.

https://github.com/boskokg/flutter_blue_plus/assets/19204095/a220731e-e336-4045-bdb3-e3958ffec3cf

![20240311-113100](

https://github.com/boskokg/flutter_blue_plus/assets/19204095/22240eec-02a8-4e77-916e-60e9ff2a1c7f

https://github.com/boskokg/flutter_blue_plus/assets/19204095/1bc2c0c4-b10c-4494-adc8-df43c38670bf)

### Logs

```shell
[FBP-iOS] initializing CBCentralManager
flutter: [FBP] \^[[1;30m<setLogLevel>\^[[0m args: \^[[1;35<…>
[FBP-iOS] initializing checkForMtuChangesTimer
[FBP-iOS] disconnectAllDevices(flutterHotRestart)
[FBP-iOS] connectedPeripherals: 0
[FBP-iOS] handleMethodCall: setLogLevel
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn
flutter: [FBP] \^[[1;30m<setLogLevel>\^[[0m result: \^[[1;33mtr<…>
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m args: \^[[1;35mnu<…>
[FBP-iOS] handleMethodCall: getAdapterState
flutter: [FBP] \^[[1;30m[[ OnAdapterStateChanged ]]\^[[0m result: \^[[1;33m{adapter_state: <…>
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m result: \^[[1;33m{adapter_state: <…>
flutter: [FBP] \^[[1;30m<getSystemDevices>\^[[0m args: \^[[1;35mnu<…>
[FBP-iOS] handleMethodCall: getSystemDevices
flutter: [FBP] \^[[1;30m<getSystemDevices>\^[[0m result: \^[[1;33m{devices: [<…>
flutter: [FBP] \^[[1;30m<startScan>\^[[0m args: \^[[1;35m{with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_scan_mode: 2, android_uses_fine_location: fals<…>
[FBP-iOS] handleMethodCall: startScan
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m<startScan>\^[[0m result: \^[[1;33mtr<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: FEDB59A0-DACE-A914-2CC8-46BBB3F10B9C, rssi: -69, connectable: 1, tx_power_level: 8}<…>
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -67, remote_id: CFB9756C-232D-47DB-B4AB-9AC94E2D9601, manufacturer_data: {6: 010920025b74425bc74e37559c19d52924c8d1f918d7be9c29bb63}}<…>
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 7194BBF9-0A28-E84B-CE69-C57C265D69C3, rssi: -71, connectable: 1, tx_power_level: 12}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -71, remote_id: FA4586BB-A5FC-F2C0-3B0F-4C576BCE47F8, manufacturer_data: {6: 01092002879593a3e1fe0995d3c5bb105e29e52e008e165010527b}}<…>
[... 237 lines cut ...]
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 7194BBF9-0A28-E84B-CE69-C57C265D69C3, rssi: -72, connectable: 1, tx_power_level: 12}<…>
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]   chr: c422
[FBP-iOS]     desc: 2902
flutter: [FBP] \^[[1;30m[[ OnDiscoveredServices ]]\^[[0m result: \^[[1;33m{remote_id: 20262E95-BC0E-E24D-BF88-7F25B172EFB2, success: 1, services: [{remote_id: 20262E95-BC0E-E24D-BF88-7F25B172EFB2, characteristics: [{remote_id: 20262E95-BC0E-E24D-BF88-7F25B172EFB2, descriptors: [], characteristic_uuid: c421, properties: {notify: 0, indicate_encryption_required: 0, write: 0, read: 0, write_without_response: 1, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: c420, secondary_service_uuid: null}, {remote_id: 20262E95-BC0E-E24D-BF88-7F25B172EFB2, descriptors: [{remote_id: 20262E95-BC0E-E24D-BF88-7F25B172EFB2, characteristic_uuid: c422, descriptor_uuid: 2902, service_uuid: c420, secondary_service_uuid: null}], characteristic_uuid: c422, properties: {notify: 1, indicate_encryption_required: 0, write: 0, read: 0, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: c420, secondary_service_uuid: null}], is_primary: true, service_uuid: c420, included_services: []}], error_string: success, error_code: <…>
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -78, remote_id: 742A6D55-5F73-42CB-A813-C3E5B8D23BD1}<…>
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 979F6B12-C986-023B-04AD-4B605163B36C, rssi: -84, connectable: 1, tx_power_level: 8}<…>
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 4C58B2D8-BE37-41C2-EFB8-D8D4FCE0C9BD, rssi: 127, connectable: 1, tx_power_level: 3}<…>
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -87, remote_id: 014FE1C0-1824-406B-806B-9211625FCBA2}<…>
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 08DB2B02-762A-4514-3D99-AFF1BE2BF7D4, rssi: -89, connectable: 1, tx_power_level: 12}<…>
[FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -77, remote_id: 8626FAB8-6111-DCD8-340E-501E4251E01A}<…>
flutter: [FBP] \^[[1;30m<stopScan>\^[[0m args: \^[[1;35mnu<…>
[FBP-iOS] handleMethodCall: stopScan
flutter: [FBP] \^[[1;30m<stopScan>\^[[0m result: \^[[1;33mtr<…>
[FBP-iOS] didDisconnectPeripheral:
[FBP-iOS]   error: The connection has timed out unexpectedly.
flutter: [FBP] \^[[1;30m[[ OnConnectionStateChanged ]]\^[[0m result: \^[[1;33m{remote_id: 20262E95-BC0E-E24D-BF88-7F25B172EFB2, connection_state: 0, disconnect_reason_code: 6, disconnect_reason_string: The connection has timed out unexpectedly<…>
```

longqian1235, 2024-03-11:
Bluetooth Module：6222M-PUC，The solution is: RTL8822CE-CG

longqian1235, 2024-03-11:
It dont happen on the first time to connect device, but it will happen always after the first connect.

chipweinberger, 2024-03-11:
This paring popup is automatic. FBP cannot control it.

It is a problem with your device firmware.

closed 2024-03-11 by chipweinberger

longqian1235, 2024-03-12:
> This paring popup is automatic. FBP cannot control it.
> 
> It is a problem with your device firmware.

So, what can I do? I dont know how to check the problem of my device firmware.

chipweinberger, 2024-03-12:
You'll need to learn how to debug your firmware