chipweinberger/flutter_blue_plus #812: [Help]: I want ScanResult.getScanRecord().getBytes()，but I couldn't find it

laxian, 2024-03-12:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.16

### Flutter Version

3.19

### What OS?

Android

### OS Version

17.0

### Bluetooth Module

I don't know

### What is your problem?

Taking Android as an example, in native development, I need to obtain `ScanResult. getScanRecord(). getBytes()`, but I cannot find the corresponding data in the parameters encapsulated in `flutter_blue_plus`.
Is there anyway to get it without modify the plugin?

### Logs

```shell
--
```

chipweinberger, 2024-03-12:
why do you want it?

no it's not available - because it is impossible on ios 

we could add it for android only though

laxian, 2024-03-13:
> why do you want it?
> 
> no it's not available - because it is impossible on ios
> 
> we could add it for android only though. But you should not need it.

Yes, you were right. I found my own data in `manufacturerData`. I don't actually need the full byte array.
Thank you.

closed 2024-03-13 by laxian