chipweinberger/flutter_blue_plus #814: [Feature]: Deactivate auto connect on disconnected peripherals

Marika-Fukazawa, 2024-03-18:
### FlutterBluePlus Version

1.31.15

### Flutter Version

3.19.3

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

Mobile integrated

### What is your feature request?

I am creating an application that repeatedly connects, communicates, and disconnects from a peripheral.
The user must be able to disable Auto Connect even after the peripheral disconnects because the user may not be connected to the peripheral at that time.

The README says to do ``await device.disconnect()`` to disable Auto Connect, but even after doing this, when the peripheral sends out an advertisement, it is reconnected.
https://github.com/boskokg/flutter_blue_plus/blob/master/README.md#auto-connect
When debugging according to Debugging in the README, if it has already disconnected, it returns at line 734 of FlutterBluePlusPlugin.java and neither gatt.disconnect() nor gatt.close() is performed.
Retrieved from https://github.com/boskokg/flutter_blue_plus/blob/master/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin. java#L716-L765

Can you please remove the remoteId from mAutoConnected and call gatt.close() if the specified remoteId is in mAutoConnected even if it has already been disconnected?

### Logs

```shell
.
```

chipweinberger, 2024-03-18:
> Can you please remove the remoteId from mAutoConnected and call gatt.close() if the specified remoteId is in mAutoConnected even if it has already been disconnected?

yes. please open a PR!

chipweinberger, 2024-03-18:
>                     if (gatt == null) {
                        log(LogLevel.DEBUG, "already disconnected");

if gatt is null, then we cannot call `gatt.disconnect();`

closed 2024-03-19 by chipweinberger