chipweinberger/flutter_blue_plus #816: [Help]: Repeated connect and disconnected results in erratic behavior

Marika-Fukazawa, 2024-03-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.16

### Flutter Version

3.19.3

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

Mobile integrated

### What is your problem?

My app uses AutoConnect to repeatedly connect and disconnect from peripherals.
When using FlutterBluePlus 1.31.16, as it repeatedly connects and disconnects, the connection relationship logs increase and the same indicate is processed multiple times.
We have confirmed that this does not occur in versions prior to 1.31.15.

I will paste a simple sample code to connect with AutoConnect when disconnected and the logs from running it. (The code for the scanning part is omitted)
After the initial connection, I disconnected from the peripheral, then sent advertisements from the peripheral, waited for the connection, and disconnected from the peripheral, repeating 5 times.

```
            try {
              await peripheral.connect(timeout: const Duration(seconds: 5), autoConnect: false, mtu: 512);
              print("1st connect!!!!!!!");
            } catch (e) {
              print(e);
              return;
            }
 
            _connectionStateSubscription ??= peripheral.connectionState.listen((BluetoothConnectionState state) async {
              if (state == BluetoothConnectionState.disconnected) {
                print("disconnected: ${peripheral.disconnectReason.toString()}");
                _isConnecting = false;
                if (!_isAutoConnected) {
                  try {
                    await peripheral.connect(autoConnect: true, mtu: null);
                    print("2nd connect!!!!!!!!");
                    _isAutoConnected = true;
                  } catch (e) {
                    print(e);
                    return;
                  }
                }
              }
            });
```

### Logs

```shell
D/[FBP-Android](22309): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](22309): [FBP] initializing BluetoothAdapter
D/[FBP-Android](22309): [FBP] disconnectAllDevices(flutterHotRestart)
[... 216 lines cut ...]
D/[FBP-Android](22309): [FBP] autoconnect is true. skipping gatt.close()
I/flutter (22309): disconnected: DisconnectReason{platform: ErrorPlatform.android, code: 19, REMOTE_USER_TERMINATED_CONNECTION}
D/[FBP-Android](22309): [FBP] autoconnect is true. skipping gatt.close()
D/BluetoothGatt(22309): onClientConnectionState() - status=19 clientIf=9 device=AA:AA:AA:AA:AA:AA
D/[FBP-Android](22309): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](22309): [FBP]   status: REMOTE_USER_TERMINATED_CONNECTION
D/[FBP-Android](22309): [FBP] autoconnect is true. skipping gatt.close()
I/flutter (22309): disconnected: DisconnectReason{platform: ErrorPlatform.android, code: 19, REMOTE_USER_TERMINATED_CONNECTION}
D/[FBP-Android](22309): [FBP] onMethodCall: connect
I/flutter (22309): disconnected: DisconnectReason{platform: ErrorPlatform.android, code: 19, REMOTE_USER_TERMINATED_CONNECTION}
D/BluetoothGatt(22309): connect() - device: AA:AA:AA:AA:AA:AA, auto: true, eattSupport: false
D/BluetoothGatt(22309): registerApp()
D/BluetoothGatt(22309): registerApp() - UUID=eec68e48-4968-490d-9977-6b067278e282
I/flutter (22309): disconnected: DisconnectReason{platform: ErrorPlatform.android, code: 19, REMOTE_USER_TERMINATED_CONNECTION}
I/flutter (22309): disconnected: DisconnectReason{platform: ErrorPlatform.android, code: 19, REMOTE_USER_TERMINATED_CONNECTION}
D/BluetoothGatt(22309): onClientRegistered() - status=0 clientIf=13
D/[FBP-Android](22309): [FBP] onMethodCall: connect
D/[FBP-Android](22309): [FBP] already connecting
D/[FBP-Android](22309): [FBP] onMethodCall: connect
D/[FBP-Android](22309): [FBP] already connecting
D/[FBP-Android](22309): [FBP] onMethodCall: connect
D/[FBP-Android](22309): [FBP] already connecting
D/[FBP-Android](22309): [FBP] onMethodCall: connect
D/[FBP-Android](22309): [FBP] already connecting
```

chipweinberger, 2024-03-19:
could be due to this: https://github.com/boskokg/flutter_blue_plus/commit/0fa2610e9fdfc813d9580c3bd132b710861e11fe

perhaps it should be 

```
        if (_autoConnect.contains(r.remoteId)) {
          if (_adapterStateNow == BmAdapterStateEnum.on) {
            if (Platform.isAndroid == false) {
               var d = BluetoothDevice(remoteId: r.remoteId);
               d.connect(autoConnect: true, mtu: null);
             }
          }
        }
```

chipweinberger, 2024-03-19:
should be fixed 1.31.17

closed 2024-03-19 by chipweinberger