chipweinberger/flutter_blue_plus #817: [Help]: scan.scan doesn't call listen on first time called - Android 14

ringonotts, 2024-03-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.37.17

### Flutter Version

3.19.3

### What OS?

Android

### OS Version

14

### Bluetooth Module

STM32

### What is your problem?

I've just updated lots of things on my app due to an upcoming release and as part of trying to tidy up I upgraded flutter versions, package versions etc.
Now, it seems that scan.scan does not work properly the first time I call it. The listen handler doesnt get called. The second time I call scan.scan it works as expected.

### Logs

```shell
Launching lib\main.dart on Pixel 6a in debug mode...
√  Built build\app\outputs\flutter-apk\app-debug.apk.
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
Connecting to VM Service at ws://127.0.0.1:56487/V5p2c9DBts0=/ws
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity]( 7013): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/CompatibilityChangeReporter( 7013): Compat change id reported: 183155436; UID 10310; state: ENABLED
[... 442 lines cut ...]
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
D/[FBP-Android]( 7013): [FBP] onScanResult
Application finished.

Exited (-1).
```

MrCsabaToth, 2024-03-19:
I never knew a scan call would need to call any listens. That would sound like an unwanted side effect to me. In my book there's a separation of concern: scan should only scan, and when I want to listen to anything I'll call. Is this related to any autoconnect or something? Do you have sample code?

ringonotts, 2024-03-19:
@MrCsabaToth Clearly didn't make it clear sorry.

From the upgrade example:

// Example code usage
    FlutterBluePlus.setLogLevel(LogLevel.verbose, color: false);
    final scanDuration = const Duration(milliseconds: 2500);
    final waitDuration = const Duration(milliseconds: 4500);

    debugPrint('BLE start scan...');
    Scan.scan(timeout: scanDuration).listen((scanResult) {
      debugPrint('BLE scanResult: $scanResult');
      debugPrint('BLE scanResult advertisementData: ${scanResult.advertisementData}');
      debugPrint('BLE scanResult advertisementData.advName: ${scanResult.advertisementData.advName}');
    });
    debugPrint('BLE scanning...');
    await Future<void>.delayed(waitDuration);
    debugPrint('BLE scanning DONE');

ringonotts, 2024-03-19:
Specifically, I need to use this so I can have the "onDone()" functionality. I want to find my devices and then display them when the scan has complete in the correct order that the user has put them in....

`FlutterBluePlus.scan(
              withServices: tmp1, 
              timeout: const Duration(seconds: 5),
              androidUsesFineLocation: false)
          .listen((scanResult) {
        // do something with scan result
        _scanListenHandler(scanResult, id);
      }).onDone(() async {
        _isScanning = false;
        setState(() {});
        _displayDevices();                          // This line displays my devices in a known order if they are found. E.g. Device "A" will always appear first if it is found in the scan.
      });`

chipweinberger, 2024-03-19:
See the migration guide about `scan`

https://github.com/boskokg/flutter_blue_plus/blob/master/MIGRATION.md

chipweinberger, 2024-03-19:
you'll need to debug this yourself. I cannot help you! 

try adding print statements to every line in this code

```
extension Scan on FlutterBluePlus {
  static Stream<ScanResult> scan({
    List<Guid> withServices = const [],
    Duration? timeout,
    bool androidUsesFineLocation = false,
  }) {
    if (FlutterBluePlus.isScanningNow) {
        throw Exception("Another scan is already in progress");
    }

    final controller = StreamController<ScanResult>();

    var subscription = FlutterBluePlus.scanResults.listen(
      (r){if(r.isNotEmpty){controller.add(r.first);}},
      onError: (e, stackTrace) => controller.addError(e, stackTrace),
    );

    Future scanComplete = FlutterBluePlus.isScanning.skip(1).where((e) => e == false).first;

    FlutterBluePlus.startScan(
      withServices: withServices,
      timeout: timeout,
      removeIfGone: null,
      oneByOne: true,
      androidUsesFineLocation: androidUsesFineLocation,
    );

    scanComplete.whenComplete(() {
      subscription.cancel();
      controller.close();
    });

    return controller.stream;
  }
}
```

ringonotts, 2024-03-19:
So it is only hitting line 19 on the listen the first time - second time hits all the lines. so "r" is empty for some reason....

```
extension Scan on FlutterBluePlus {
  static Stream<ScanResult> scan({
    List<Guid> withServices = const [],
    Duration? timeout,
    bool androidUsesFineLocation = false,
  }) {
    if (FlutterBluePlus.isScanningNow) {
      throw Exception("Another scan is already in progress");
    }

    final controller = StreamController<ScanResult>();

    var subscription = FlutterBluePlus.scanResults.listen(
      (r) {
        print("..................................................SCAN class line 19");
        if (r.isNotEmpty) {
          print("..................................................SCAN class line 21");
          controller.add(r.first);
          print("..................................................SCAN class line 23");
        }
      },
      onError: (e, stackTrace) => controller.addError(e, stackTrace),
    );

          print("..................................................SCAN class line 29");
    FlutterBluePlus.startScan(
      withServices: withServices,
      timeout: timeout,
      //removeIfGone: null,
      oneByOne: true,
      androidUsesFineLocation: androidUsesFineLocation,
    );

    Future scanComplete =
        FlutterBluePlus.isScanning.where((e) => e == false).first;

    scanComplete.whenComplete(() {
          print("..................................................SCAN class line 42");
      subscription.cancel();
      controller.close();
    });

    return controller.stream;
  }
}
```

ringonotts, 2024-03-19:
```
Launching lib\main.dart on Pixel 7a in debug mode...
√  Built build\app\outputs\flutter-apk\app-debug.apk.
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
Connecting to VM Service at ws://127.0.0.1:54420/b5H0wIorH0Y=/ws
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/CompatibilityChangeReporter(25188): Compat change id reported: 183155436; UID 10308; state: ENABLED
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
I/FirebaseApp(25188): Device unlocked: initializing all Firebase APIs for app [DEFAULT]
D/CompatibilityChangeReporter(25188): Compat change id reported: 3400644; UID 10308; state: ENABLED
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
D/VRI[MainActivity](25188): Cancelling draw. cancelDueToPreDrawListener=true cancelDueToSync=false
[... cut ...]
D/[FBP-Android](25188): [FBP] onScanResult
D/[FBP-Android](25188): [FBP] onScanResult
D/[FBP-Android](25188): [FBP] onScanResult
D/[FBP-Android](25188): [FBP] onScanResult
D/[FBP-Android](25188): [FBP] onScanResult
D/[FBP-Android](25188): [FBP] onScanResult
D/[FBP-Android](25188): [FBP] onScanResult
I/flutter (25188): [FBP] <stopScan> args: null
D/[FBP-Android](25188): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(25188): isLeEnabled(): ON
I/flutter (25188): [FBP] <stopScan> result: true

```

MrCsabaToth, 2024-03-19:
Could you pinpoint which FBP version brought in the unwanted behavior?

MrCsabaToth, 2024-03-20:
Hey @chipweinberger, not sure if this is a far shot, but can it happen that changing `StreamController`'s `sync: true` to `sync: false` may have some effect on this? https://github.com/boskokg/flutter_blue_plus/issues/740

@ringonotts Can you test if FBP `1.31.3` has the problem vs `1.31.2`?

MrCsabaToth, 2024-03-20:
> @MrCsabaToth Yeah I thought the same - got back to 1.37.7 and still had the issue. I swear thats been working for me for a while now. I'm starting to think Flutter is doing something weird. Breakpoints apear as hollow circles now instead of red ones (until they are hit - and then they go red). I'll do more investigating - sorry for the delay, i'm OOO today.
> 
> Ringo

No worries. Flutter may have some effect, but in general Android BT protocol stack can be stochastic as well. There are various scan modes, etc. Test with other phone as well if you have any.

ringonotts, 2024-03-20:
So i checked my repo and I was using 1.28.13 before 1.31.7. The issue still persists.

What seems weird is even though i have set the scanDuration to 2.5 seconds the whenComplete method is called immediately (to the human eye not measured).

chipweinberger, 2024-03-20:
so what was the first version with problem? it should take 5 minutes to binary search

MrCsabaToth, 2024-03-21:
(Somewhat off-topic, but I found that unless I specify at least 6 seconds for scan duration some devices might not show up in the scan.)

ringonotts, 2024-03-21:
@chipweinberger apologies - ignore all comments about whats working and whats not from before. I was changing my pubspec.yaml to different versions and despite doing a flutter clean it was always getting 1.31.17. Anyway, i removed the ^ now (flutter noob still).

It works in 1.31.7 but doesn't work in 1.31.8 or above. I tried 1.31.13, 1.31.10, 1.31.17 ....

I feel so much better now progress is being made :~)

@MrCsabaToth - yeah i tested years ago that 5 seconds seemed optimum but i have noticed recently sometimes my sensors don't get picked up. For the sake of a second i'll take your advice. Thanks!

chipweinberger, 2024-03-21:
1.31.8 had two commits:
https://github.com/boskokg/flutter_blue_plus/commit/d1a9292bb1b3fd44e7df31334b4ba26bcf6cc6d0
https://github.com/boskokg/flutter_blue_plus/commit/c263d6644ed3354b5304440cc545db44e8eef2a3

These both fixed real issues, so they cannot be reverted.

You'll need to add more logs and debug it until you have a full answer.

chipweinberger, 2024-03-21:
just add logs everywhere and try to find out why whenComplete is called immediately.

ringonotts, 2024-03-22:
I'll have to look into it another time and stick on 31.7 for now.

Do you think I should work out a way to stop using the scan extension ? Sounds like i'm the only one using it....

Ringo

chipweinberger, 2024-03-22:
up to you!

egbred, 2024-03-25:
I face the same problem. Flutter 3.19.3, FBP 1.31.15. 
First scan completes immediately for the code from migration guide. 

For me it happens on iOS. Most likely that issue is not platform specific.

ringonotts, 2024-03-25:
Its the mutex. If you remove the check for the mutex - ie remove this:
```
   _Mutex mtx = _MutexFactory.getMutexForKey("scan");
    await mtx.take();
    try {
```
Then it works.

I can't find anything on that MutexFactory - google just shows this issue.

I can't go further without some documentation for that mutex library.

Ringo

ringonotts, 2024-03-25:
As i said before, the "finally" code is strange to me. I don't think its intended for asyncronous behaviour.
The reason being is because the "finally" is called as soon as the try code is executed - not after the scan is completed. So the mutex seems a bit pointless to me really.

In fact, if you just comment out the "mtx.give();" in the finally and the "await mtx.take();" in the stopScan() method then this give the functionality - and it fixes my issues.

However, not sure that fixes the original intended purpose of the mutex used there.... Could you try it ?

Regards,

Richard

chipweinberger, 2024-03-25:
thanks for your debugging

MutexFactory code is part of FBP. look at /utils.dart

the mutex is needed to solve issues /asserts when you call stopScan while startScan is in the middle of the function

ringonotts, 2024-03-25:
This muddled old brain can't work out that code.

Are you sure it's not happening on your example too ? It's just the first time its called that it does this. It's like the stream is released with nothing in it (in the listen method (in the extension code) r is empty) which may not apear to be an issue for you when a stream is returned many times every time a device is found...

ringonotts, 2024-03-25:
So the second and following times you do the scan, the listen method still gets called immediately (not after the duration has expired) - but this time r is not empty - it will have one entry only in the ScanResult - not all the devices it should find like I am expecting.

I've got six devices that i want to connect too and I expect them all to be in the ScanResult...

chipweinberger, 2024-03-25:
are you using the latest version of the extension?

```
extension Scan on FlutterBluePlus {
  static Stream<ScanResult> scan({
    List<Guid> withServices = const [],
    Duration? timeout,
    bool androidUsesFineLocation = false,
  }) {
    if (FlutterBluePlus.isScanningNow) {
        throw Exception("Another scan is already in progress");
    }

    final controller = StreamController<ScanResult>();

    var subscription = FlutterBluePlus.scanResults.listen(
      (r){if(r.isNotEmpty){controller.add(r.first);}},
      onError: (e, stackTrace) => controller.addError(e, stackTrace),
    );

    Future scanComplete = FlutterBluePlus.isScanning.skip(1).where((e) => e == false).first;

    FlutterBluePlus.startScan(
      withServices: withServices,
      timeout: timeout,
      removeIfGone: null,
      oneByOne: true,
      androidUsesFineLocation: androidUsesFineLocation,
    );

    scanComplete.whenComplete(() {
      subscription.cancel();
      controller.close();
    });

    return controller.stream;
  }
}
```

chipweinberger, 2024-03-26:
yes I can see in your above code, your extension is missing the `.skip(1)`

please update your extension

closed 2024-03-26 by chipweinberger

ringonotts, 2024-03-26:
OMG. The perils of using an extension I guesss...

Yeah - that fixed it....

Where is the extension linked too now on the main page ? I can't find it...

Thanks Chip you legend