chipweinberger/flutter_blue_plus #818: [Help]:

Megamil, 2024-03-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.17

### Flutter Version

3.19.2

### What OS?

iOS

### OS Version

iOS 17.4

### Bluetooth Module

Key Bluetooth

### What is your problem?

I'm encountering an error with the blue_plus library when trying to use my module in a native iOS app using method channel.

When I build directly from Flutter on the iPhone, it works perfectly. However, when using it via method channel, with the Flutter project being a module, it always throws an error. It seems like it never manages to access the Bluetooth, even though I've added the necessary permissions in the info.plist.

### Logs

```shell
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method isSupported on channel flutter_blue_plus/methods)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:332)
<asynchronous suspension>
#1      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:571)
<asynchronous suspension>
#2      FlutterBluePlus.isSupported (package:flutter_blue_plus/src/flutter_blue_plus.dart:64)
<asynchronous suspension>
#3      BluetoothManager.checkBluetoothStatus (package:movida_express/hardware/bluetooth_manager.dart:29)
<asynchronous suspension>
#4      openCar (package:movida_express/platform/method_channel.dart:87)
<asynchronous suspension>
#5      _TestScreenState.build.<anonymous closure> (package:movida_express/ui/test_screen.dart:93)
<asynchronous suspension>
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method stopScan on channel flutter_blue_plus/methods)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:332)
<asynchronous suspension>
#1      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:571)
<asynchronous suspension>
#2      FlutterBluePlus._stopScan (package:flutter_blue_plus/src/flutter_blue_plus.dart:357)
<asynchronous suspension>
#3      FlutterBluePlus.stopScan (package:flutter_blue_plus/src/flutter_blue_plus.dart:343)
<asynchronous suspension>
```

chipweinberger, 2024-03-20:
> I've looked at the README 'Common Problems' section

did you lie about this?

closed 2024-03-20 by chipweinberger

Megamil, 2024-03-20:
> > I've looked at the README 'Common Problems' section
> 
> did you lie about this?

No, I didn't lie about it, the issue is that despite trying cleaning, reinstalling, changing devices, restarting the system, the error still persists, and it only happens via method channel. In the native Flutter application, it runs perfectly, as mentioned in the first line of my comment.

chipweinberger, 2024-03-20:
Hmm. Okay it was not clear to me.

This sounds like you have a problem in your build setup. But I'm not sure the solution.

siyengar04, 2024-04-19:
I have the same issue, running on Web

chipweinberger, 2024-04-19:
web is not a supported platform