chipweinberger/flutter_blue_plus #822: [Feature]: Make CBCentralManagerOptionShowPowerAlertKey configurable (iOS)

fesp, 2024-03-26:
### FlutterBluePlus Version

1.31.12

### Flutter Version

3.19.4

### What OS?

iOS

### OS Version

17.3.1

### Bluetooth Module

?

### What is your feature request?

At the moment `CBCentralManagerOptionShowPowerAlertKey` is hardcoded to `YES`. It would be nice if the app could set this to `NO`. Currently my app shows a Flutter dialog if the Bluetooth adapter is turned off. But in order to detect this state I need to use `FlutterBluePlus.adapterState` which causes the native dialog to appear. 

So at this point I have two dialogs. If then dismiss the native dialog, the Flutter dialog is still there (of course). It is not an option to remove the Flutter dialog because (as far as I know) there is no way to re-trigger the native dialog once dismissed.

### Logs

```shell
-
```

chipweinberger, 2024-03-26:
open a PR. no need to file an issue.

chipweinberger, 2024-03-26:
maybe this

`FlutterBluePlus.setOptions({bool iosShowPowerAlert}}`

closed 2024-03-26 by chipweinberger

fesp, 2024-03-28:
> open a PR. no need to file an issue.
> 
> thanks though.

You are right of course. I did have a go at it (with the `setOptions` approach you mentioned above), but found that the very first method call is `flutterHotRestart`, which causes CBCentralManager to initialise. Therefore `setOptions` comes too late. Any thoughts on this?

chipweinberger, 2024-03-28:
yes, don't use FlutterBluePlus._invokeMethod

instead directly call _methodChannel.invokeMethod

chipweinberger, 2024-03-28:
or maybe better, add an option to skip calling _initFlutterBluePlus in this function

static Future<dynamic> _invokeMethod(String method, [dynamic arguments])

fesp, 2024-03-28:
Yes, that solved the issue. Thanks!
Need to finish some work first but then I'll create the PR.