chipweinberger/flutter_blue_plus #828: Why AirPods are not connected

yjlAmor, 2024-03-29:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.17

### Flutter Version

3.16.0

### What OS?

All

### OS Version

iOS 16.1.1

### Bluetooth Module

iOS default module

### What is your problem?

flutter_blue_plus demo displays that AirPods cannot be connected。Why AirPods are not connected？Is there any way to connect AirPods？

### Logs

```shell
No log, the sample has shown that the connection is not possible
```

chipweinberger, 2024-03-29:
nope. audio is not supported.

closed 2024-03-29 by chipweinberger

yjlAmor, 2024-03-29:
> nope. audio is not supported.

Thanks for your answer