chipweinberger/flutter_blue_plus #829: [Help]: On AutoConnect, device ConnectionState does not change to connected.

prachigole, 2024-03-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.0

### Flutter Version

3.19.5

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

Nordic Nrf

### What is your problem?

I have a device connection page, on pressing the connect and disconnect buttons, it updates the connectionState of the device. However, during autoconnect, I observed that the connectionState on switching on my device remains disconnected (even when device has successfully resumed ble operation) and sometimes switches between connected and disconnected. I see that there have been some improvements to the package about this issue, but it was working till yesterday evening for me. This causes an issue to displaynthe disconnect/connect button state and its hard to identify whether the device wok

### Logs

```shell
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
D/BluetoothGatt(22981): onClientConnectionState() - status=0 clientIf=7 device=FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onClientConnectionState() - status=0 clientIf=12 device=FB:34:C1:7E:5E:EE
I/flutter (22981): RESETTING FG: 2024-03-30 14:57:37.385346
D/BluetoothGatt(22981): connect() - device: FB:34:C1:7E:5E:EE, auto: true, eattSupport: false
D/BluetoothGatt(22981): registerApp()
D/BluetoothGatt(22981): registerApp() - UUID=fc7130fa-c4ee-4a83-8bfd-1e90ccb1fd47
D/BluetoothGatt(22981): onClientRegistered() - status=0 clientIf=13
I/flutter (22981): 2024-03-30 14:57:37.403038 | Error: FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected
I/flutter (22981): #0      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:228:7)
I/flutter (22981): #1      ForegroundTaskHandler.connectToDevice (package:breathai_app/features/foreground_task/foreground_task_handler.dart:210:43)
I/flutter (22981): <asynchronous suspension>
I/flutter (22981):
I/flutter (22981): 2024-03-30 14:57:38.392399 | startUploadFiles called
I/flutter (22981): RESETTING FG: 2024-03-30 14:57:38.394603
I/flutter (22981): 2024-03-30 14:57:38.409508 | Error: FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected
I/flutter (22981): #0      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:228:7)
I/flutter (22981): #1      ForegroundTaskHandler.connectToDevice (package:breathai_app/features/foreground_task/foreground_task_handler.dart:210:43)
I/flutter (22981): <asynchronous suspension>
I/flutter (22981):
[... 380 lines cut ...]
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
I/flutter (22981): 2024-03-30 14:59:20.237446 | [FG] Acc is reading
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
I/flutter (22981): 2024-03-30 14:59:21.245482 | [FG] Acc is reading
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
I/flutter (22981): 2024-03-30 14:59:22.256703 | [FG] Acc is reading
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
I/flutter (22981): 2024-03-30 14:59:23.269482 | [FG] Acc is reading
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
I/flutter (22981): 2024-03-30 14:59:24.274156 | [FG] Acc is reading
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
I/flutter (22981): 2024-03-30 14:59:25.277358 | [FG] Acc is reading
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
I/flutter (22981): 2024-03-30 14:59:26.280011 | [FG] Acc is reading
D/BluetoothGatt(22981): discoverServices() - device: FB:34:C1:7E:5E:EE
D/BluetoothGatt(22981): onSearchComplete() = Device=FB:34:C1:7E:5E:EE Status=0
I/flutter (22981): 2024-03-30 14:59:27.302459 | [FG] Acc is reading
```

chipweinberger, 2024-03-30:
what is the first version with this problem?

prachigole, 2024-04-01:
I started facing this from 1.31.17 onwards. It works in 1.31.16.

chipweinberger, 2024-04-01:
thanks. i don't see any cause for your problem 

can you try reverting each commit in 1.31.17 one by one?

chipweinberger, 2024-04-06:
closed due to no response.

closed 2024-04-06 by chipweinberger