chipweinberger/flutter_blue_plus #832: [Help]: ANDROID_SPECIFIC_ERROR  status=133 clientIf=7

Ervin2018, 2024-04-02:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.0

### Flutter Version

3.19.4

### What OS?

Android

### OS Version

/

### Bluetooth Module

/

### What is your problem?

I encounter this issue when establishing a connection with my device

### Logs

```shell
V/AudioManager(24475): querySoundEffectsEnabled...
D/[FBP-Android](24475): [FBP] onMethodCall: stopScan
I/BluetoothAdapter(24475): isLeEnabled(): ON
D/[FBP-Android](24475): [FBP] onMethodCall: connect
D/BluetoothGatt(24475): connect() - device: 53:5A:0A:**:**:**, auto: false
D/BluetoothGatt(24475): registerApp()
D/BluetoothGatt(24475): registerApp() - UUID=52c9dda1-fab7-41bc-9e2d-a9f0d5f91565
D/BluetoothGatt(24475): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(24475): onClientConnectionState() - status=133 clientIf=7 device=53:5A:0A:**:**:**
D/[FBP-Android](24475): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](24475): [FBP]   status: ANDROID_SPECIFIC_ERROR
D/BluetoothGatt(24475): close()
D/BluetoothGatt(24475): unregisterApp() - mClientIf=7
```

chipweinberger, 2024-04-02:
sorry but we can't fix this.

it is a problem with your phone or your ble device.

closed 2024-04-02 by chipweinberger

Jbz797, 2024-05-10:
I had no problems when I was in version 1.20.8. This error appeared in subsequent versions, so I had to freeze the version in flutter_blue_ plus.