chipweinberger/flutter_blue_plus #834: [Help]: I need to print on the Phomemo D520BT printer

Arhidev, 2024-04-03:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.1

### Flutter Version

3.13.2

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

Phomemo D520BT

### What is your problem?

Hello dear everyone!
Can I use package flutter_blue_plus: ^1.32.1 to print with Phomemo D520BT?
reference: [](https://phomemo.com/fr/products/entreprises-detiquettes-dexpedition-bluetooth-d520-bt)

### Logs

```shell
flutter_bluetooth_serial  package to print 

Failed to connect: PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1
I/flutter (14205): 	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:776)
I/flutter (14205): 	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:790)
I/flutter (14205): 	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:415)
I/flutter (14205): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
I/flutter (14205): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
I/flutter (14205): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.lambda$onMethodCall$4$io-github-edufolly-flutterbluetoothserial-FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler(FlutterBluetoothSerialPlugin.java:1007)
I/flutter (14205): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler$$Externa
```

chipweinberger, 2024-04-03:
you should open this issue in the flutterbluetoothserial package

Arhidev, 2024-04-03:
ok buut Can I use package flutter_blue_plus: ^1.32.1 to print with Phomemo D520BT?

chipweinberger, 2024-04-03:
no

Arhidev, 2024-04-03:
ok, thank you !