chipweinberger/flutter_blue_plus #835: [Help]: randomly not getting any scan results

cestaxelle, 2024-04-03:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.17

### Flutter Version

3.16.5

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

Qualcomm® FastConnect™ 6200

### What is your problem?

Hi,
I use FlutterBluePlus in my app and usually everything works perfectly, but sometimes, randomly, no device is detected. `onScanResult` is seemingly never called and my list of devices remains empty, even after waiting a few moments. Stopping and starting the scan again does the trick, but do you know what causes this behaviour and how to avoid it? 
The example app behaves the same way after a few scans, I've copied my logs below (where the empty scan is the last one). (I've removed a few "D/[FBP-Android](15784): [FBP] onScanResult" lines, otherwise the logs were too long.)
Many thanks!

### Logs

```shell
Launching lib/main.dart on SM P613 in debug mode...
✓  Built build/app/outputs/flutter-apk/app-debug.apk.
Connecting to VM Service at ws://127.0.0.1:34327/WZ2NXEz6Zw0=/ws
D/[FBP-Android](15784): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](15784): [FBP] initializing BluetoothAdapter
I/flutter (15784): [FBP] <setLogLevel> args: 5
I/BluetoothAdapter(15784): BluetoothAdapter() : com.lib.flutter_blue_plus_example
I/BluetoothAdapter(15784): STATE_ON
D/[FBP-Android](15784): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](15784): [FBP] connectedPeripherals: 0
D/[FBP-Android](15784): [FBP] onMethodCall: setLogLevel
I/flutter (15784): [FBP] <setLogLevel> result: true
I/flutter (15784): [FBP] <getAdapterState> args: null
D/[FBP-Android](15784): [FBP] onMethodCall: getAdapterState
I/flutter (15784): [FBP] <getAdapterState> result: {adapter_state: 4}
I/ViewRootImpl@d1aa13b[MainActivity](15784): handleAppVisibility mAppVisible = false visible = true
I/SurfaceView@5e54be6(15784): onWindowVisibilityChanged(4) false io.flutter.embedding.android.FlutterSurfaceView{5e54be6 G.E...... ......I. 0,0-0,0} of ViewRootImpl@d1aa13b[MainActivity]
D/SurfaceView@5e54be6(15784): updateSurface: has no frame
D/OpenGLRenderer(15784): setSurface called with nullptr
I/ViewRootImpl@d1aa13b[MainActivity](15784): performTraversals mFirst=false windowShouldResize=false viewVisibilityChanged=true mForceNextWindowRelayout=false params=null
D/InsetsController(15784): onStateChanged: InsetsState: {mDisplayFrame=Rect(0, 0 - 1200, 2000), mDisplayCutout=DisplayCutout{insets=Rect(0, 0 - 0, 0) waterfall=Insets{left=0, top=0, right=0, bottom=0} boundingRect={Bounds=[Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0)]} cutoutPathParserInfo={CutoutPathParserInfo{displayWidth=0 displayHeight=0 physicalDisplayWidth=0 physicalDisplayHeight=0 density={0.0} cutoutSpec={} rotation={0} scale={0.0} physicalPixelDisplaySizeRatio={0.0}}}}, mRoundedCorners=RoundedCorners{[RoundedCorner{position=TopLeft, radius=20, center=Point(20, 20)}, RoundedCorner{position=TopRight, radius=20, center=Point(1180, 20)}, RoundedCorner{position=BottomRight, radius=20, center=Point(1180, 1980)}, RoundedCorner{position=BottomLeft, radius=20, center=Point(20, 1980)}]}  mRoundedCornerFrame=Rect(0, 0 - 1200, 2000), mPrivacyIndicatorBounds=PrivacyIndicatorBounds {static bounds=Rect(1134, 0 - 1200, 36) rotation=0}, mSources= { InsetsSource: {mType=ITYPE_STATUS_BAR, mFrame=[0,0][1200,36], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_LEFT_GESTURES, mFrame=[0,0][0,2000], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_RIGHT_GESTURES, mFrame=[1200,0][1200,2000], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_MANDATORY_GESTURES, mFrame=[0,0][1200,36], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_MANDATORY_GESTURES, mFrame=[0,1928][1200,2000], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_TAPPABLE_ELEMENT, mFrame=[0,0][1200,36], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_TAPPABLE_ELEMENT, mFrame=[0,1928][1200,2000], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_IME, mFrame=[0,0][0,0], mVisibleFrame=[0,1382][1200,2000], mVisible=false, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_EXTRA_NAVIGATION_BAR, mFrame=[0,1928][1200,2000], mVisible=false, mInsetsRoundedCornerFrame=true} } host=com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity from=android.view.ViewRootImpl.relayoutWindow:9945
I/ViewRootImpl@d1aa13b[MainActivity](15784): Relayout returned: old=(0,0,1200,2000) new=(0,0,1200,2000) req=(1200,2000)4 dur=7 res=0x0 s={false 0x0} ch=false seqId=0
D/SurfaceView@5e54be6(15784): updateSurface: has no frame
I/ViewRootImpl@d1aa13b[MainActivity](15784): stopped(false) old = true
I/ViewRootImpl@d1aa13b[MainActivity](15784): WindowStopped on com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity set to false
D/SurfaceView@5e54be6(15784): updateSurface: has no frame
D/DecorView(15784): isDeviceLocked = false
I/MSHandlerLifeCycle(15784): check: return. pkg=com.lib.flutter_blue_plus_example mode=fullscreen op=true locked=false callers=com.android.internal.policy.DecorView.setVisibility:4487 android.app.Activity.makeVisible:6819 android.app.ActivityThread.updateVisibility:5876 android.app.ActivityThread.handleStartActivity:4255 android.app.servertransaction.TransactionExecutor.performLifecycleSequence:221
I/MSHandlerLifeCycle(15784): removeMultiSplitHandler: no exist. decor=DecorView@40e021f[MainActivity]
I/ViewRootImpl@d1aa13b[MainActivity](15784): stopped(false) old = false
D/DecorView(15784): isDeviceLocked = false
I/MSHandlerLifeCycle(15784): check: return. pkg=com.lib.flutter_blue_plus_example mode=fullscreen op=true locked=false callers=com.android.internal.policy.DecorView.setVisibility:4487 android.app.Activity.makeVisible:6819 android.app.ActivityThread.handleResumeActivity:5556 android.app.servertransaction.ResumeActivityItem.execute:54 android.app.servertransaction.ActivityTransactionItem.execute:45
[... 555 lines cut ...]
D/BluetoothLeScanner(15784): Stop Scan with callback
I/flutter (15784): [FBP] <stopScan> result: true
I/ViewRootImpl@d1aa13b[MainActivity](15784): ViewPostIme pointer 0
I/ViewRootImpl@d1aa13b[MainActivity](15784): ViewPostIme pointer 1
I/flutter (15784): [FBP] <getSystemDevices> args: null
D/[FBP-Android](15784): [FBP] onMethodCall: getSystemDevices
I/flutter (15784): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (15784): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android](15784): [FBP] onMethodCall: startScan
I/BluetoothAdapter(15784): STATE_ON
I/BluetoothAdapter(15784): STATE_ON
I/BluetoothAdapter(15784): STATE_ON
I/BluetoothAdapter(15784): STATE_ON
D/BluetoothLeScanner(15784): Start Scan with callback
D/BluetoothLeScanner(15784): onScannerRegistered() - status=6 scannerId=-1 mScannerId=0
I/flutter (15784): [FBP] <startScan> result: true
I/ViewRootImpl@d1aa13b[MainActivity](15784): ViewPostIme pointer 0
I/ViewRootImpl@d1aa13b[MainActivity](15784): ViewPostIme pointer 1
I/flutter (15784): [FBP] <stopScan> args: null
D/[FBP-Android](15784): [FBP] onMethodCall: stopScan
I/BluetoothAdapter(15784): STATE_ON
I/BluetoothAdapter(15784): STATE_ON
D/BluetoothLeScanner(15784): could not find callback wrapper
I/flutter (15784): [FBP] <stopScan> result: true
```

chipweinberger, 2024-04-03:
update to latest FBP and make sure you handle `onError`

Android only allows 6 scans per minute

cestaxelle, 2024-04-04:
Thanks, I am handling `onError` but no error ever occurs, which is consistent with what I have just read about Android blocking the number of scans. I didn't know about that, but that explains the app's behaviour, thanks a lot!