chipweinberger/flutter_blue_plus #837: [Help]: Exception handling

enseitankad0, 2024-04-05:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.19

### Flutter Version

3.16.0

### What OS?

macOS

### OS Version

14.1.1

### Bluetooth Module

Norfic NRF52840

### What is your problem?

I want to catch all exception and map them into a well defined classes. E.g

If I'm connecting to device id which does not exist I can check for:

```
e is PlatformException
e.code == 'connect'
e.message?.contains('is not a valid Bluetooth address')
```

and based on those rules I can return `DeviceNotFoundError`

I can also check platform and code of FlutterBluePlusException
```
e is FlutterBluePlusException
e.platform == ErrorPlatform.apple
e.code == 15
```
and base on it determine InsufficientEncryptionError

Question:
1. How to map similar exceptions into one common class to easily recognize them on UI? I know that all Exceptions thrown by FBP are very helpful in logs and in debug, but what if I want to present more specific error to the user?

### Logs

```shell
no logs
```

chipweinberger, 2024-04-05:
I recommend you use your app like a normal user to determine which error occur

then catch them and add nicer messages

most exceptions should never occur in normal usage

chipweinberger, 2024-04-05:
I dont have a comprehensive list of errors because iOS and Android do not provide a full list of errors

chipweinberger, 2024-04-05:
for example, here is the documentation for `connect` on iOS

https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/connect(_:options:)

notice: they do not list the errors that can be thrown

enseitankad0, 2024-04-05:
Thank you for quick reply. Basically I'm migrating my project from Flutter Reactive Ble to FBP. Previously I could just read excpection's message and based on it I could determine e.g., wrong pairing (bonding) issue, if exception contains  `"Writing is not permitted"`. 

Based on that, I could tell user to go to settings and to remove pairing manually. It was necessary because sometimes w/o this step, I couldn't reconnect to the device after the update on Nordic chip was performed. 

The problem is, if I would receive the same effect I should check all possibilities (for Android and iOS) and based on that, return appropriate error.

chipweinberger, 2024-04-05:
yes. 

you do do the same thing with FBP

i dont understand the problem.

enseitankad0, 2024-04-05:
So for this specific error I've mentioned,  should I check whether 
`e is FlutterBluePlusException` +
`apple-code: 3  | Writing is not permitted.` or `android-code: 3  | GATT_WRITE_NOT_PERMITTED`, and based on that, create my custom error?

Or the check should be done based on: 
`e is PlatformException && e.code == 'writeCharacteristic` ?

chipweinberger, 2024-04-05:
up to you.

id probably check for `e.function == 'writeCharacteristic` and `e.code == 3`

closed 2024-04-05 by chipweinberger