chipweinberger/flutter_blue_plus #840: [Help]: On the OPPO Reno 11 phone, when obtaining a large amount of data, only the latter half of the data is received in the listener.

guduqiucai, 2024-04-07:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.1

### Flutter Version

3.19.5

### What OS?

Android

### OS Version

14

### Bluetooth Module

unknown

### What is your problem?

my Code:
```
void connectToBlueDevice(BluetoothDevice bluetoothDevice) async {
    // 监听蓝牙设备的连接状态
    bluetoothDevice.connectionState.listen((event) {
      // 如果蓝牙断开，且当前页面不是蓝牙列表页，就跳转到蓝牙列表页
      if (event == BluetoothConnectionState.disconnected) {
        if (JiChenRoutes.route.location != JiChenRoutes.bluetoothPath) {
          print("蓝牙断开了！！！");
          JiChenRoutes.route.go(JiChenRoutes.bluetoothPath);
        }
      }
    });

    /// 已连接的服务, 这里判断如果蓝牙连接失败就提示重试
    List<BluetoothService> services = await bluetoothDevice.discoverServices();
    if (services.length == 0) {
      EasyLoading.showError(S.current.bluetoothConnectFail);
      return;
    } else{
      EasyLoading.dismiss();
    }

    /// 下面的代码是蓝牙连接成功后的代码
    currentService = services.singleWhere((v) => v.uuid
        .toString()
        .toLowerCase()
        .contains(BlueToothCharacter.serviceUUID));

    // 已连接服务的读写特征码
    writeChar = currentService.characteristics.singleWhere((v) =>
        v.toString().toLowerCase().contains(BlueToothCharacter.writeUUID));

    GlobalData.curDeviceMaxMtu = 512;
    if (Platform.isAndroid) {
      await bluetoothDevice.mtu.first.then( (value) => {
        GlobalData.curDeviceMaxMtu = value,
        bluetoothDevice.requestMtu(value)
      });
    }
[... 65 lines cut ...]
D/[FBP-Android]( 9667): [FBP]   chr: c305
I/flutter ( 9667): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c305, value: 55aa1100000000000100040002f84f5866}
I/flutter ( 9667): 报文中的crc值 66584ff8
I/flutter ( 9667): 读到的值：Instance of 'Msg'
I/flutter ( 9667): wifi的连接状态结束时间：2024-04-07 14:46:25.812904
I/flutter ( 9667): 附近的wifi开始时间：2024-04-07 14:46:25.815323
I/flutter ( 9667): [FBP] <writeCharacteristic> args: {remote_id: 08:B6:1F:FD:CB:3E, service_uuid: a002, secondary_service_uuid: null, characteristic_uuid: c304, write_type: 0, allow_long_write: 0, value: 55aa110000000000010005000071ad2222}
D/[FBP-Android]( 9667): [FBP] onMethodCall: writeCharacteristic
I/flutter ( 9667): [FBP] <writeCharacteristic> result: true
D/[FBP-Android]( 9667): [FBP] onCharacteristicWrite:
D/[FBP-Android]( 9667): [FBP]   chr: c304
D/[FBP-Android]( 9667): [FBP]   status: GATT_SUCCESS (0)
I/flutter ( 9667): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c304, value: 55aa110000000000010005000071ad2222}
D/[FBP-Android]( 9667): [FBP] onCharacteristicChanged:
D/[FBP-Android]( 9667): [FBP]   chr: c305
D/[FBP-Android]( 9667): [FBP] onCharacteristicChanged:
D/[FBP-Android]( 9667): [FBP]   chr: c305
I/flutter ( 9667): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c305, value: 55aae30200000000010005002b43574c41503a28332c224c4e48432d4755455354222c2d35392c2231343a38343a37373a31633a33663a3438222c31312c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c224c4e48432d574f524b222c2d35392c2231343a38343a37373a31633a33663a3439222c31312c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c224c4e48432d4755455354222c2d36352c2231343a38343a37373a31633a34363a3638222c312c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c224c4e48432d574f524b222c2d36392c2231343a38343a37373a31633a34363a3639222c312c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c224c4e48432d4755455354222c2d37322c2231343a38343a37373a31633a34363a3438222c362c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c224c4e48432d574f524b222c2d37322c2231343a38343a37373a31633a34363a3439222c362c2d312c2d312c342c342c372c30290d0a2b43574c41503a28302c2254502d4c494e4b
I/flutter ( 9667): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c305, value: 2c362c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c22e78e8be4beafe5b086e79bb8222c2d38332c2231613a66353a37363a63333a38323a3035222c31312c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c224c4e48432d574f524b222c2d38362c2231343a38343a37373a31633a34343a3039222c312c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c22e5b08fe9b9bfe79a844d6174652035302050726f222c2d38392c2230653a33623a33323a66363a62653a3030222c382c2d312c2d312c342c342c372c30290d0af316089e}
I/flutter ( 9667): 报文中的crc值 9e0816f3
I/flutter ( 9667): 读到的值：Instance of 'Msg'
I/flutter ( 9667): 保存搜索的周围wifi:"+CWLAP:(3,\"LNHC-GUEST\",-59,\"14:84:77:1c:3f:48\",11,-1,-1,4,4,7,0)\r\n+CWLAP:(3,\"LNHC-WORK\",-59,\"14:84:77:1c:3f:49\",11,-1,-1,4,4,7,0)\r\n+CWLAP:(3,\"LNHC-GUEST\",-65,\"14:84:77:1c:46:68\",1,-1,-1,4,4,7,0)\r\n+CWLAP:(3,\"LNHC-WORK\",-69,\"14:84:77:1c:46:69\",1,-1,-1,4,4,7,0)\r\n+CWLAP:(3,\"LNHC-GUEST\",-72,\"14:84:77:1c:46:48\",6,-1,-1,4,4,7,0)\r\n+CWLAP:(3,\"LNHC-WORK\",-72,\"14:84:77:1c:46:49\",6,-1,-1,4,4,7,0)\r\n+CWLAP:(0,\"TP-LINK\",-80,\"18:f2:2c:26:ab:62\",1,-1,-1,0,0,7,0)\r\n+CWLAP:(3,\"LNHC-GUEST\",-80,\"14:84:77:1c:42:68\",6,-1,-1,4,4,7,0)\r\n+CWLAP:(3,\"çä¾¯å°ç¸\",-83,\"1a:f5:76:c3:82:05\",11,-1,-1,4,4,7,0)\r\n+CWLAP:(3,\"LNHC-WORK\",-86,\"14:84:77:1c:44:09\",1,-1,-1,4,4,7,0)\r\n+CWLAP:(3,\"å°é¹¿çMate 50 Pro\",-89,\"0e:3b:32:f6:be:00\",8,-1,-1,4,4,7,0)\r\n"
I/resgying.eurOps( 9667): This is sticky GC, maxfree is 8388608 minfree is 524288
I/flutter ( 9667): 附近的wifi结束时间：2024-04-07 14:46:29.820910
```

chipweinberger, 2024-04-07:
seems like a problem with your device, not FBP

guduqiucai, 2024-04-07:
> FBP

Device side packet capture, the returned data is complete, and the data obtained using other Bluetooth tools is also normal：BLE调试宝

chipweinberger, 2024-04-07:
Okay. you should try to debug this problem and open a PR.

I will not have time to investigate.

guduqiucai, 2024-04-07:
error log:
Launching lib/main.dart on PJH110 in debug mode...
Running Gradle task 'assembleDebug'...
✓  Built build/app/outputs/flutter-apk/app-debug.apk.
Installing build/app/outputs/flutter-apk/app-debug.apk...
Uninstalling old version...
Error: ADB exited with exit code 1
Performing Streamed Install

adb: failed to install /Users/coco/Documents/Code/eurapp/build/app/outputs/flutter-apk/app-debug.apk: Failure [INSTALL_FAILED_UPDATE_INCOMPATIBLE: Existing package com.gresgying.eurOps signatures do not match newer version; ignoring!]
Installing build/app/outputs/flutter-apk/app-debug.apk...
Debug service listening on ws://127.0.0.1:57744/HgUqRaC4_mQ=/ws
Syncing files to device PJH110...
D/VRI[MainActivity](25247):  debugCancelDraw some OnPreDrawListener onPreDraw return false,cancelDraw=true,count=50,android.view.ViewRootImpl@301ad8c
I/flutter (25247): MyNavObserver:didPush: _CustomTransitionPageRoute<dynamic>(NoTransitionPage<dynamic>("null", null, null), animation: AnimationController#f9b74(⏭ 1.000; paused; for _CustomTransitionPageRoute<dynamic>)), previousRoute= null
I/flutter (25247): PermissionStatus.denied
D/MAGT_SYNC_FRAME(25247): MAGT Sync: MAGT is not supported. Disabling Sync.
W/WindowOnBackDispatcher(25247): OnBackInvokedCallback is not enabled for the application.
W/WindowOnBackDispatcher(25247): Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
D/BLASTBufferQueue(25247): [SurfaceView[com.gresgying.eurOps/com.example.eurOps.MainActivity]#1](f:0,a:1) acquireNextBufferLocked size=1080x2412 mFrameNumber=1 applyTransaction=true mTimestamp=569828642849425(auto) mPendingTransactions.size=0 graphicBufferId=108435039322112 transform=0
D/SurfaceView(25247): 181463488 handleSyncNoBuffer
D/SurfaceView(25247): 181463488 finishedDrawing
D/SurfaceView(25247): 181463488 performDrawFinished mAttachedToWindow true
D/VRI[MainActivity](25247):  debugCancelDraw  cancelDraw=false,count = 199,android.view.ViewRootImpl@301ad8c
I/SurfaceControl(25247):  setExtendedRangeBrightness sc=Surface(name=com.gresgying.eurOps/com.example.eurOps.MainActivity)/@0xb4e5f33,currentBufferRatio=1.0,desiredRatio=1.0
D/SurfaceView(25247): 181463488 dispatchDraw mDrawFinished true isAboveParent false (mPrivateFlags & PFLAG_SKIP_DRAW) 128
D/SurfaceView(25247): 181463488 clearSurfaceViewPort mCornerRadius 0.0
D/BLASTBufferQueue(25247): [VRI[MainActivity]#0](f:0,a:1) acquireNextBufferLocked size=1080x2412 mFrameNumber=1 applyTransaction=true mTimestamp=569828691817425(auto) mPendingTransactions.size=0 graphicBufferId=108435039322114 transform=0
D/VRI[MainActivity](25247): Received frameCommittedCallback lastAttemptedDrawFrameNum=1 didProduceBuffer=true syncBuffer=false
D/VRI[MainActivity](25247): draw finished.
D/VRI[MainActivity](25247): reportDrawFinished
I/Quality (25247): Skipped: false 1 cost 32.342995 refreshRate 16774116 bit true processName com.gresgying.eurOps
D/VRI[MainActivity](25247): onFocusEvent true
D/OplusScrollToTopManager(25247): com.gresgying.eurOps/com.example.eurOps.MainActivity,This DecorView@e76cd92[MainActivity] change focus to true
D/OplusSystemUINavigationGestureExtImpl(25247): regionString = null
D/OplusInputMethodManagerInternal(25247): get inputMethodManager extension: com.android.internal.view.IInputMethodManager$Stub$Proxy@6d7b23
D/ViewRootImplExtImpl(25247): MotionEvent MotionEvent { action=ACTION_UP, actionButton=0, id[0]=0, x[0]=320.125, y[0]=1354.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x0, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=569830467, downTime=569830385, deviceId=2, source=0x1002, displayId=0, eventId=53334543 } handled by client, just return
D/ScrollOptim [SceneManager](25247): updateCurrentActivity: mCurrentActivityName=null, isOptEnable=true, isAnimAheadEnable=true, isFrameInsertEnable=true, InsertNum=1, isEnabledForScrollChanged=false
D/VRI[MainActivity](25247): onFocusEvent false
D/OplusScrollToTopManager(25247): com.gresgying.eurOps/com.example.eurOps.MainActivity,This DecorView@e76cd92[MainActivity] change focus to false
D/ColorModeChangeItem(25247): preExecute mColorMode=0,token=android.os.BinderProxy@e1e2104
D/CompatibilityChangeReporter(25247): Compat change id reported: 78294732; UID 10354; state: ENABLED
D/ScrollOptim [SceneManager](25247): updateCurrentActivity: mCurrentActivityName=com.example.eurOps.MainActivity, isOptEnable=true, isAnimAheadEnable=true, isFrameInsertEnable=true, InsertNum=1, isEnabledForScrollChanged=false
D/ActivityThread(25247): ComponentInfo{com.gresgying.eurOps/com.example.eurOps.MainActivity} checkFinished=false 4
D/ResourcesManagerExtImpl(25247): applyConfigurationToAppResourcesLocked app.getDisplayId() return callback.displayId:-1
D/VRI[MainActivity](25247): onFocusEvent true
D/permissions_handler(25247): Bluetooth permission missing in manifest
I/flutter (25247): 蓝牙权限请求失败
D/[FBP-Android](25247): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](25247): [FBP] initializing BluetoothAdapter
D/CompatibilityChangeReporter(25247): Compat change id reported: 263076149; UID 10354; state: DISABLED
D/[FBP-Android](25247): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](25247): [FBP] connectedPeripherals: 0
D/[FBP-Android](25247): [FBP] onMethodCall: isSupported
I/flutter (25247): 蓝牙是否可用,true
D/[FBP-Android](25247): [FBP] onMethodCall: getAdapterState
I/flutter (25247): 蓝牙状态为开启
I/flutter (25247): [FBP] stopScan: already stopped
[... cut ...]
I/flutter (25247): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c305, value: 55455354222c2d38342c2231343a38343a37373a31633a34323a3638222c362c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c224c4e48432d574f524b222c2d38352c2231343a38343a37373a31633a34323a3639222c362c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c22e5b08fe9b9bfe79a844d6174652035302050726f222c2d38352c2230653a33623a33323a66363a62653a3030222c382c2d312c2d312c342c342c372c30290d0aee66dc46}

guduqiucai, 2024-04-07:
[840](https://github.com/boskokg/flutter_blue_plus/issues/841)

closed 2024-04-07 by guduqiucai