chipweinberger/flutter_blue_plus #841: [Feature]: On the OPPO Reno 11 phone, when obtaining a large amount of data, only the latter half of the data is received in the listener.

guduqiucai, 2024-04-07:
### FlutterBluePlus Version

1.32.1

### Flutter Version

3.19.5

### What OS?

Android

### OS Version

14

### Bluetooth Module

unknown

### What is your feature request?

D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onMethodCall: writeCharacteristic
W/WindowOnBackDispatcher(25247): sendCancelIfRunning: isInProgress=falsecallback=ImeCallback=ImeOnBackInvokedCallback@223065331 Callback=android.window.IOnBackInvokedCallback$Stub$Proxy@fa87598
W/VRI[MainActivity](https://github.com/boskokg/flutter_blue_plus/issues/25247): handleResized abandoned!
I/flutter (25247): [FBP] result: true
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onCharacteristicWrite:
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] chr: c304
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] status: GATT_SUCCESS (0)
I/flutter (25247): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c304, value: 55aa2c0000000000020003000f000000340000000b000000e807000004000000070000000700000016d88216}
I/ImeTracker(25247): com.gresgying.eurOps:cbb511fc: onRequestHide at ORIGIN_CLIENT_HIDE_SOFT_INPUT reason HIDE_SOFT_INPUT_BY_INSETS_API
I/ImeTracker(25247): com.gresgying.eurOps:bc7a5e57: onHidden
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onCharacteristicChanged:
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] chr: c305
I/flutter (25247): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c305, value: 55aa110000000000020003000092eba0bb}
I/flutter (25247): 报文中的crc值 bba0eb92
I/flutter (25247): 读到的值：Instance of 'Msg'
W/VRI[MainActivity](https://github.com/boskokg/flutter_blue_plus/issues/25247): handleResized abandoned!
I/flutter (25247): [FBP] args: {remote_id: 08:B6:1F:FD:CB:3E, service_uuid: a002, secondary_service_uuid: null, characteristic_uuid: c304, write_type: 0, allow_long_write: 0, value: 55aa120000000000010007001600c10a2e39}
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onMethodCall: writeCharacteristic
I/flutter (25247): [FBP] result: true
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onCharacteristicWrite:
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] chr: c304
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] status: GATT_SUCCESS (0)
I/flutter (25247): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c304, value: 55aa120000000000010007001600c10a2e39}
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onCharacteristicChanged:
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] chr: c305
I/flutter (25247): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c305, value: 55aa82000000000001000700700016007b0a092252554c45223a09302c0a0922504f574552223a09302c0a0922434f554e54223a09302c0a092252455352415445223a0931302c0a0922475249445048415345223a09302c0a09224d45544552454e41424c45223a09302c0a09224d45544552434845434b223a09300a7d20a5dc37}
I/flutter (25247): 报文中的crc值 37dca520
I/flutter (25247): 读到的值：Instance of 'Msg'
D/ViewRootImplExtImpl(25247): MotionEvent MotionEvent { action=ACTION_UP, actionButton=0, id[0]=0, x[0]=118.375, y[0]=511.125, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x0, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=569854996, downTime=569854840, deviceId=2, source=0x1002, displayId=0, eventId=759820596 } handled by client, just return
I/flutter (25247): MyNavObserver:didPush: _PageBasedMaterialPageRoute(MaterialPage("/wifiConnect", [<'/wifiConnect-p0'>], {}), animation: AnimationController#a1c88(▶ 0.000; for _PageBasedMaterialPageRoute(/wifiConnect))), previousRoute= _CustomTransitionPageRoute(NoTransitionPage("null", null, null), animation: AnimationController#09538(⏭ 1.000; paused; for _CustomTransitionPageRoute))
I/flutter (25247): 已连接的wifi开始时间：2024-04-07 15:52:20.265128
W/WindowOnBackDispatcher(25247): OnBackInvokedCallback is not enabled for the application.
W/WindowOnBackDispatcher(25247): Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
I/flutter (25247): [FBP] args: {remote_id: 08:B6:1F:FD:CB:3E, service_uuid: a002, secondary_service_uuid: null, characteristic_uuid: c304, write_type: 0, allow_long_write: 0, value: 55aa1100000000000100030000665b93f3}
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onMethodCall: writeCharacteristic
I/flutter (25247): [FBP] result: true
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onCharacteristicWrite:
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] chr: c304
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] status: GATT_SUCCESS (0)
I/flutter (25247): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c304, value: 55aa1100000000000100030000665b93f3}
I/flutter (25247): wifi的连接状态开始时间：2024-04-07 15:52:31.462923
I/flutter (25247): [FBP] args: {remote_id: 08:B6:1F:FD:CB:3E, service_uuid: a002, secondary_service_uuid: null, characteristic_uuid: c304, write_type: 0, allow_long_write: 0, value: 55aa11000000000001000400000f3f6387}
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onMethodCall: writeCharacteristic
I/flutter (25247): [FBP] result: true
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] onCharacteristicWrite:
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] chr: c304
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/25247): [FBP] status: GATT_SUCCESS (0)
I/flutter (25247): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c304, value: 55aa11000000000001000400000f3f6387}
[... 45 lines cut ...]
D/FBP-Android: [FBP] status: GATT_SUCCESS (0)
I/flutter (25247): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c304, value: 55aa11000000000001000400000f3f6387}
D/FBP-Android: [FBP] onCharacteristicChanged:
D/FBP-Android: [FBP] chr: c305
I/flutter (25247): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c305, value: 55aa5a00000000000100030056312e302e3134000000000000000000000000000000000056312e3056312e300000000000000000000000000000000016000000094c4e48432d574f524b0b4c6e4032303230303831355415d12d}
I/flutter (25247): 报文中的crc值 2dd11554
I/flutter (25247): 读到的值：Instance of 'Msg'
D/FBP-Android: [FBP] onCharacteristicChanged:
D/FBP-Android: [FBP] chr: c305
I/flutter (25247): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c305, value: 55aa1100000000000100040002f84f5866}
I/flutter (25247): 报文中的crc值 66584ff8
I/flutter (25247): 读到的值：Instance of 'Msg'
I/flutter (25247): wifi的连接状态结束时间：2024-04-07 15:52:34.098971
I/flutter (25247): 附近的wifi开始时间：2024-04-07 15:52:34.108823
I/flutter (25247): [FBP] args: {remote_id: 08:B6:1F:FD:CB:3E, service_uuid: a002, secondary_service_uuid: null, characteristic_uuid: c304, write_type: 0, allow_long_write: 0, value: 55aa110000000000010005000071ad2222}
D/FBP-Android: [FBP] onMethodCall: writeCharacteristic
I/flutter (25247): [FBP] result: true
D/FBP-Android: [FBP] onCharacteristicWrite:
D/FBP-Android: [FBP] chr: c304
D/FBP-Android: [FBP] status: GATT_SUCCESS (0)
[... cut ...]
I/flutter (25247): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: a002, success: 1, remote_id: 08:B6:1F:FD:CB:3E, error_code: 0, characteristic_uuid: c305, value: 55455354222c2d38342c2231343a38343a37373a31633a34323a3638222c362c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c224c4e48432d574f524b222c2d38352c2231343a38343a37373a31633a34323a3639222c362c2d312c2d312c342c342c372c30290d0a2b43574c41503a28332c22e5b08fe9b9bfe79a844d6174652035302050726f222c2d38352c2230653a33623a33323a66363a62653a3030222c382c2d312c2d312c342c342c372c30290d0aee66dc46}
```

guduqiucai, 2024-04-07:
my code:
```
void connectToBlueDevice(BluetoothDevice bluetoothDevice) async {
    // 监听蓝牙设备的连接状态
    bluetoothDevice.connectionState.listen((event) {
      // 如果蓝牙断开，且当前页面不是蓝牙列表页，就跳转到蓝牙列表页
      if (event == BluetoothConnectionState.disconnected) {
        if (JiChenRoutes.route.location != JiChenRoutes.bluetoothPath) {
          print("蓝牙断开了！！！");
          JiChenRoutes.route.go(JiChenRoutes.bluetoothPath);
        }
      }
    });

    /// 已连接的服务, 这里判断如果蓝牙连接失败就提示重试
    List<BluetoothService> services = await bluetoothDevice.discoverServices();
    if (services.length == 0) {
      EasyLoading.showError(S.current.bluetoothConnectFail);
      return;
    } else{
      EasyLoading.dismiss();
    }

    /// 下面的代码是蓝牙连接成功后的代码
    currentService = services.singleWhere((v) => v.uuid
        .toString()
        .toLowerCase()
        .contains(BlueToothCharacter.serviceUUID));

    // 已连接服务的读写特征码
    writeChar = currentService.characteristics.singleWhere((v) =>
        v.toString().toLowerCase().contains(BlueToothCharacter.writeUUID));

    GlobalData.curDeviceMaxMtu = 512;
    if (Platform.isAndroid) {
      await bluetoothDevice.mtu.first.then( (value) => {
        GlobalData.curDeviceMaxMtu = value,
        bluetoothDevice.requestMtu(value)
      });
    }
    FlutterBluePlus.setLogLevel(LogLevel.verbose, color:false);
    readChar = currentService.characteristics.singleWhere((v) =>
        v.toString().toLowerCase().contains(BlueToothCharacter.readUUID));

    // 全局保存当前服务，当前读写，当前设备
    GlobalData.currentService = currentService;
    GlobalData.writeChar = writeChar;
    GlobalData.readChar = readChar;
    GlobalData.curBluetoothDevice = bluetoothDevice;

    //此变量作为存储蓝牙读取到的原始字节的容器，数据一来就放到该容器中，解析时（定时或者每次收到新数据），将解析过的字节从容器中删除

    if (readChar.properties.notify) {
      print("开始读取消息");

      await readChar.setNotifyValue(true);
      readChar.lastValueStream.listen((val) {
        //每次来新数据，都将新字节放入comingBytes容器
        if(GlobalData.comingBytes.length >= val.length && listEquals(val, GlobalData.comingBytes.sublist(GlobalData.comingBytes.length - val.length))){

        } else {
          GlobalData.comingBytes.addAll(val);
          //同时触发解析（此处是每次来新的数据流，就尝试解析，看是否是完整的包，不是完整的包则不做处理，是完整的包则开始解析，也可以定时1秒中解析一次）
          parseDataBytes(GlobalData.comingBytes);
        }
      });
    }

    afterBluetoothConnected();
  }
```

chipweinberger, 2024-04-10:
nothing I can do about this. I don't have this problem. If you find the problem, please report back.

closed 2024-04-10 by chipweinberger