chipweinberger/flutter_blue_plus #842: [Help]: Why Android MethodChannel cost so much time than iOS

laxian, 2024-04-07:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.16

### Flutter Version

3.19

### What OS?

Android

### OS Version

SAMSUN S9, android 28

### Bluetooth Module

I don't know

### What is your problem?

I did a stress test on Bluetooth write data, **20Hz**, the data size is 14Byte, iOS is fast, about 2ms. 
It takes about 20ms for Android to go from dart to native to dart, and time fluctuates a lot on Android. 
The maximum value exceeded 4000ms. 

```Dart
// call this in timer 20Hz（duration 50ms）
Future<void> sendMessage(
    Message msg,
  ) async {
    var bytes = msg.toBytes();
    log("BEFORE ENC: ${bytes.toHexString()}");
    await lock.synchronized(() async {
      Uint8List cr = encrypt(bytes);
      var tik = DateTime.now().millisecondsSinceEpoch;
      print("WRITE BEFORE:");
      await _writeToCharacteristic(cr);
      var tok = DateTime.now().millisecondsSinceEpoch;
      var t = tok - tik;
      print("WRITE cost($t ms): ${cr.toHexString()}");
    });
  }

  Future<void> _writeToCharacteristic(
    Uint8List data,
  ) async {
    log(data.toHexString());
    // log("WRITE TO BLE: ${data.toHexString()}");
    await writeCharacteristic.write(data,
        withoutResponse: writeCharacteristic.properties.writeWithoutResponse);
  }
```

```Bash
# part of log
04-07 14:07:45.491  4729  4763 I flutter : WRITE BEFORE:

# from dart to Android native, 2ms
04-07 14:07:45.493  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic

# here cost 4000+ms
04-07 14:07:50.023  4729  4841 D [FBP-Android]: [FBP] onCharacteristicWrite:

# here always cost over 10ms
04-07 14:07:50.033  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
04-07 14:07:50.033  4729  4763 I flutter : ZWX DART COST: _methodCallHandler ADD TO STREAM OnCharacteristicWritten
04-07 14:07:50.035  4729  4763 I flutter : WRITE cost(4544 ms): A01501ADE2DA4E4ACE9FA9B20021
```
The following is a sampling statistics chart using Android S9 at a certain time, where dart2native fluctuates greatly
[image]

This is the result of testing using iPhone 13, without counting the intermediate time, but the total time is very short
[image]

Is there any way to improve the performance of Android？

### Logs

```shell
I did a stress test on Bluetooth write data, 50Hz, the data size is 14Byte, iOS is fast, about 2ms. 
It takes about 20ms for Android to go from dart to native to dart, and time fluctuates a lot on Android. 
The maximum value exceeded 4000ms. 

04-07 14:07:45.094  4729  4763 I flutter : WRITE BEFORE:
04-07 14:07:45.096  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic
04-07 14:07:45.100  4729  4751 D [FBP-Android]: [FBP] onCharacteristicWrite:
04-07 14:07:45.141  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
04-07 14:07:45.142  4729  4763 I flutter : ZWX DART COST: _methodCallHandler ADD TO STREAM OnCharacteristicWritten
04-07 14:07:45.143  4729  4763 I flutter : WRITE cost(49 ms): A015010DBA556F01A355E707001A
04-07 14:07:45.147  4729  4763 I flutter : WRITE BEFORE:
04-07 14:07:45.149  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic
04-07 14:07:45.152  4729  4751 D [FBP-Android]: [FBP] onCharacteristicWrite:
04-07 14:07:45.153  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
04-07 14:07:45.153  4729  4763 I flutter : ZWX DART COST: _methodCallHandler ADD TO STREAM OnCharacteristicWritten
04-07 14:07:45.155  4729  4763 I flutter : WRITE cost(7 ms): A0150173163B9B74D7F76266001B
04-07 14:07:45.231  4729  4763 I flutter : WRITE BEFORE:
04-07 14:07:45.234  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic
04-07 14:07:45.273  4729  4751 D [FBP-Android]: [FBP] onCharacteristicWrite:
04-07 14:07:45.286  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
04-07 14:07:45.286  4729  4763 I flutter : ZWX DART COST: _methodCallHandler ADD TO STREAM OnCharacteristicWritten
04-07 14:07:45.287  4729  4763 I flutter : WRITE cost(56 ms): A01501E571782DAB00CB57AF001C
04-07 14:07:45.289  4729  4763 I flutter : WRITE BEFORE:
04-07 14:07:45.291  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic
04-07 14:07:45.293  4729  4751 D [FBP-Android]: [FBP] onCharacteristicWrite:
04-07 14:07:45.293  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
04-07 14:07:45.294  4729  4763 I flutter : ZWX DART COST: _methodCallHandler ADD TO STREAM OnCharacteristicWritten
04-07 14:07:45.295  4729  4763 I flutter : WRITE cost(6 ms): A01501848BADD97A0C6BC064001D
04-07 14:07:45.343  4729  4763 I flutter : WRITE BEFORE:
04-07 14:07:45.345  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic
04-07 14:07:45.352  4729  4751 D [FBP-Android]: [FBP] onCharacteristicWrite:
04-07 14:07:45.377  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
04-07 14:07:45.377  4729  4763 I flutter : ZWX DART COST: _methodCallHandler ADD TO STREAM OnCharacteristicWritten
04-07 14:07:45.379  4729  4763 I flutter : WRITE cost(35 ms): A0150198D8AA2E65E6B868B7001E
04-07 14:07:45.382  4729  4763 I flutter : WRITE BEFORE:
04-07 14:07:45.384  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic
04-07 14:07:45.389  4729  4751 D [FBP-Android]: [FBP] onCharacteristicWrite:
04-07 14:07:45.390  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
04-07 14:07:45.390  4729  4763 I flutter : ZWX DART COST: _methodCallHandler ADD TO STREAM OnCharacteristicWritten
04-07 14:07:45.392  4729  4763 I flutter : WRITE cost(9 ms): A015017D8C93958A2BE6E74C001F
04-07 14:07:45.428  4729  4763 I flutter : WRITE BEFORE:
04-07 14:07:45.430  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic
04-07 14:07:45.439  4729  4751 D [FBP-Android]: [FBP] onCharacteristicWrite:
04-07 14:07:45.461  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
04-07 14:07:45.462  4729  4763 I flutter : ZWX DART COST: _methodCallHandler ADD TO STREAM OnCharacteristicWritten
04-07 14:07:45.463  4729  4763 I flutter : WRITE cost(35 ms): A01501EC9EEFDDB312D90CAA0020

04-07 14:07:45.491  4729  4763 I flutter : WRITE BEFORE:

# from dart to Android native, 2ms
04-07 14:07:45.493  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic

# here cost 4000+ms
04-07 14:07:50.023  4729  4841 D [FBP-Android]: [FBP] onCharacteristicWrite:

# here always cost over 10ms
04-07 14:07:50.033  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
04-07 14:07:50.033  4729  4763 I flutter : ZWX DART COST: _methodCallHandler ADD TO STREAM OnCharacteristicWritten
04-07 14:07:50.035  4729  4763 I flutter : WRITE cost(4544 ms): A01501ADE2DA4E4ACE9FA9B20021
```

chipweinberger, 2024-04-08:
Yes you are right. Android is really slow. sucks.

AFAIK only the flutter team could fix this.

chipweinberger, 2024-04-08:
how are you measuring the dart-to-native time?

chipweinberger, 2024-04-08:
```
# from dart to Android native, 2ms
04-07 14:07:45.493  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic

# here cost 4000+ms
04-07 14:07:50.023  4729  4841 D [FBP-Android]: [FBP] onCharacteristicWrite:
```

the 4000ms includes the bluetooth communication time. This includes the android BLE stack & the time it takes your device to respond.

chipweinberger, 2024-04-08:
so really it seems like the MethodChannel is not slow, but your BLE device is slow / android BLE stack is slow.

chipweinberger, 2024-04-08:
also know that we can only do 1 write at a time, generally.

laxian, 2024-04-08:
> how are you measuring the dart-to-native time?
> 
> in your log it says 2ms

_dart 2 native_: The time difference between the two rows of logs.  Yeah, it's fast.
```
04-07 14:07:45.094  4729  4763 I flutter : WRITE BEFORE:
04-07 14:07:45.096  4729  4729 D [FBP-Android]: [FBP] onMethodCall: writeCharacteristic
```

_native 2 dart_, these two rows below, cost more milliseconds. 
```
04-07 14:07:50.023  4729  4841 D [FBP-Android]: [FBP] onCharacteristicWrite:
04-07 14:07:50.033  4729  4763 I flutter : ZWX DART COST: _methodCallHandler OnCharacteristicWritten
```

Overall, iOS is a bit faster than Android. I'm not sure if there might be some issues with my usage. Thanks for your reply.

chipweinberger, 2024-04-10:
theres nothing FBP can do about this. We just call the native android BLE functions.

You can try opening this issue on android. but bluetooth is very complicated on android. depends on the phone & bluetooth processor.