chipweinberger/flutter_blue_plus #846: [Feature]: iOS Core Bluetooth Background Processing support

William-Geng, 2024-04-09:
### FlutterBluePlus Version

1.31.17

### Flutter Version

3.19.3

### What OS?

iOS

### OS Version

17

### Bluetooth Module

nordic nRF52832

### What is your feature request?

It would be nice to support iOS core bluetooth background execution mode as described here: https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html.

I believe there is a PR for flutter_blue that was never merged:  https://github.com/pauldemarco/flutter_blue/pull/191 and https://github.com/pauldemarco/flutter_blue/pull/210.

### Logs

```shell
none
```

chipweinberger, 2024-04-22:
feel free to create new PRs and test them

MrCsabaToth, 2024-05-07:
FYI there was a https://github.com/espresso3389/flutter_blue fork of the original pauldemarco `flutter_blue` back in the day which dealt with foreground task issue as well, along with some emphasis on iOS and also permissions. But seems like that fork died too some times after `flutter_blue`, but it might be useful to see what PRs they merged and what they did for this.

I'll need some foreground support as well in my app at some point - highly requested -, however this seems to be a major undertaking in my case. More specifically at my app level this looks to be a significant architectural change. Even though there are plugins such as https://pub.dev/packages/flutter_foreground_task/ you can see it's unfortunately not a simple plop in of some code, it's 5+ stage step procedure. I also don't see a way once I convert my app, to have that functionality be optional. That's important because if such function would be at the FBP plugin level, it could happen not everyone would want it (because for example it needs extra manifest declarations, you'd need to select type of service, extra permission shenanigans, declaring callbacks, and the list goes on and on).

But ultimately I'd need to tackle this as well for user request. I'm not sure if it's even doable at FBP level, it'll be hard enough at app level. I'm curious what is your use-case and plans.

Codel1417, 2024-05-14:
My initial testing with the app in the background, using a 15 second keep awake ping, kept my app open.

h4nkyn, 2024-06-13:
What type of keep awake ping are you talking about? Like a characteristic value change?

Codel1417, 2024-06-13:
> What type of keep awake ping are you talking about? Like a characteristic value change?

Yeah. Having a characteristic value change wakes the app for 10-15 seconds. If you do this enough the app stays open

drolpi, 2024-07-02:
> feel free to create new PRs and test them

@chipweinberger are there any updates on this? I absolutely need this feature! Otherwise, I have to completely rebuild my app in a different language with a different framework. Can someone please implement IOS BLE State Restoration.

chipweinberger, 2024-07-02:
@drolpi , just open a PR!

chipweinberger, 2024-07-02:
@drolpi 

> Can someone please implement IOS BLE State Restoration.

I don't need this feature, but I will implement it for a fee. email me.

 Otherwise you should implement it yourself.

MrCsabaToth, 2024-07-03:
> @chipweinberger are there any updates on this? I absolutely need this feature! Otherwise, I have to completely rebuild my app in a different language with a different framework. Can someone please implement IOS BLE State Restoration.

BLE State Restoration and background services are two different things, although may not be completely unrelated (in case the the background service gets yeeted (this part of Android is not standardized https://dontkillmyapp.com/) then restoration may come into play?)

MrCsabaToth, 2024-07-03:
I think Chip did some work related to reconnection if I'm not mistaken, however BLE State Restoration sounds more formal.

chipweinberger, 2024-08-27:
I've pushed this commit (https://github.com/chipweinberger/flutter_blue_plus/commit/12c17a6f49c3d6223f7dffaa75f845ccc01c8701) , which *should* enable this feature. But I have not been able to get it to work in actual testing.

1. when the app is killed, it is not restored
2. when the phone is rebooted, it is not restored

chipweinberger, 2024-08-31:
released version 1.32.13 with State Preservation

Scottie-Fischer, 2024-09-06:
> * when the app is killed, the app is not restored ❌
> * when the phone is rebooted, the app is not restored ❌

I made some minor tweaks using your latest update and was able to get it to restore the CBCentralManager and peripherals. 
I did see some issues where when I restore the peripheral and save it in the `connectedDevices` list that it times out when trying to discover services or subscribe to characteristic using `setNotifyingValue`. Have not pinned down the issues, but will make a PR. 

Additionally I created a restoration state that tells the app whether or not it was restored or the initial start. This may be useful for diverging behaviors, but not technically required.

chipweinberger, 2024-09-06:
awesome!

i look forward to the PR!

gpminsuk, 2024-10-15:
@chipweinberger Thanks for this update. I was able to verify that the app reconnect, wake up, and runs the code.

> when the app is killed, the app is not restored ❌

When you say this, do you mean when the app is killed by user? Or do you mean when the app is OS as well?
The problem I am having now is that
1. My device is connected with state restoration and autoconnect
2. The device gets disconnected due to distance from the central device (app).
3. Time passes and OS decides to kill the app.
4. The device come back close to the central device (app)
5. Reconnect / wake up does not happen.

I believe iOS should support this scenario. If you could shed some light, I can try to prepare a PR.

RohitMishra1402, 2024-12-24:
Is there any working solution for background fetching BLE for iOS devices?

chipweinberger, 2024-12-24:
not sure. can someone else chime in?

RohitMishra1402, 2024-12-24:
@all Is there any documentation stating that background BLE fetching on iOS devices is impossible?

tayz98, 2025-01-30:
@Codel1417 

> > What type of keep awake ping are you talking about? Like a characteristic value change?
> 
> Yeah. Having a characteristic value change wakes the app for 10-15 seconds. If you do this enough the app stays open. The device I connect to uses 2 characteristics for a read/write command setup so I just sent ping which gets a response.

Sorry for digging up this thread. Basically this is what i am doing but I have the feeling that even when the app is alive it doesn't execute any code that is not related to bluetooth. I need to create objects and store the collected BLE data, but the app is just skipping/ignoring these commands only executing "write commands", which is basically worthless if i can't do anything with the received data....

gpminsuk, 2025-03-05:
I am looking for someone who can help resolve this issue as a paid task.

iyzg, 2025-03-11:
Me as well. Can background BLE write to characteristics or scan for devices?

cyonix, 2025-07-31:
> I made some minor tweaks using your latest update and was able to get it to restore the CBCentralManager and peripherals.
> I did see some issues where when I restore the peripheral and save it in the [...]

@Scottie-Fischer Did you end up creating the PR? I'm interested in seeing what changes you made here.

chipweinberger, 2025-08-21:
It seems like the piece needed to get things running in Dart is this: "B. Start a headless Flutter engine for quick background processing"